gazebosim/gz-rendering #84: Create lidar visualization

iche033, 2020-05-14:
Create visualizations for lidar data. This can be divided into smaller subtasks:

1. Design public APIs for a lidar visualization class, e.g. `LidarVisual`. The abstract interface will go in `ign-rendering/include/ignition/rendering/` and a base implementation will live in `ign-rendering/include/ignition/rendering/base` 
1. Port the [LaserVisual]( https://github.com/osrf/gazebo/blob/gazebo11/gazebo/rendering/LaserVisual.cc) class from gazebo to `ign-rendering/ogre`. This would be the ogre 1.x implementation. Note that in gazebo's `LaserVisual` class, the data come from ign-transport but since ign-rendering does not use ign-transport, we should design the APIs in 1. to support accepting and modifying lidar data, e.g. either an API that takes a vector of floating point values or individual values. We will call these APIs from ign-gazebo to feed it the lidar data to be visualized.
1. Add ogre 2.x implementation

chapulina, 2020-06-18:
Done, thanks!

mihirk284, 2020-07-08:
Additional Task:-
1. Add option to change the visual types to Lines and Points
2. Adding colour to visualised point

chapulina, 2020-08-05:
Sorry, I think this was closed because I linked it to #116, but that was probably not the intention. Reopening just in case.

closed 2020-09-28 by iche033