gazebosim/gz-rendering #85: Missing AxisVisual and ArrowVisual implementation

iche033, 2020-05-19:
Implementation of AxisVisual and ArrowVisual is missing for ogre ([empty classes](https://github.com/ignitionrobotics/ign-rendering/blob/master/ogre/src/OgreArrowVisual.cc)) and ogre2 ([no implementation](https://github.com/ignitionrobotics/ign-rendering/blob/master/ogre2/src/Ogre2Scene.cc#L282)).

Levi-Armstrong, 2020-05-19:
These methods currently work for ogre1 and ogre2 on melodic so I looked into the issue. The class is empty because it is implemented in the base class (BaseAxisVisual and BaseArrowVisual) using calls to createConeVisual and createCylinderVisual.

iche033, 2020-05-20:
oh I missed that. Thanks for looking into this. I think I just tested ogre2 at the time and it was not working. It's mainly missing a few hooks to return a valid pointer. Done in pull requests #87

Levi-Armstrong, 2020-05-27:
Thank you for the support.

iche033, 2020-05-27:
pull request #87

closed 2020-05-27 by iche033