gazebosim/gz-rendering #86: Copying Visuals in a Scene

peter-lehner, 2020-05-19:
I would like to create a (shallow) copy of a Visual in a Scene, along the lines of:

```cpp
ScenePtr scene = CreateSomeScene();
VisualPtr source_visual = scene->CreateVisual("source");

// Make source_visual nice and shiny ...

// Create a visual named target 
// target_visual should receive the geometries of source_visual, 
// but not copy the child nodes, which would open up problems about naming etc.
VisualPtr target_visual = scene->CopyVisualByName("source", "target");

```

I have looked for alternatives in the newest reference, but have found none. I'm willing to contribute if someone points me to where to begin.

iche033, 2020-05-26:
How about adding a `Clone` function to the Visual class? The function can take an optional `recursive` arg. If true, it does a deep copy instead of a shallow copy, e.g.

```
// a couple of overloaded Clone functions
VisualPtr Visual::Clone(bool _recursive = false);
VisualPtr Visual::Clone(const std::string &_name, bool _recursive = false);
```

```
VisualPtr clone = visual->Clone();  // shallow copy
VisualPtr clone2 = visual->Clone("new_name", true); // deep copy with user-specified name
```

peter-lehner, 2020-05-27:
That would definitely work for my use case, yes. My thoughts originally were that the scene seems like the factory for visuals, so I would have left the construction there.

What pattern would you suggest to construct unique names to the cloned visuals if none is specified? 
```
new_name = old_name + "_copy_" + index? 
```

This would the also require a similar `Clone` function for the Geometry class tree, correct?

iche033, 2020-05-27:
in the `Clone` function, you should be able to create a visual by calling [Scene::CreateVisual()](https://github.com/ignitionrobotics/ign-rendering/blob/master/include/ignition/rendering/Scene.hh#L691) which will generate a unique name for you. 

> This would the also require a similar Clone function for the Geometry class tree, correct?

yeah this may be a little more involved with all the different classes that derive from Geometry. You can probably start with Mesh as that's the most used geometry.

peter-lehner, 2020-05-28:
Alright, thanks for your input. I will try to implement a first version and then get back to you. This will take me a while.