gazebosim/gz-rendering #89: Meshes not visible when using dae files that include the transparency tag.

Levi-Armstrong, 2020-05-25:


chapulina, 2020-05-25:
@Levi-Armstrong , are you using Ogre 1 or 2? Or is the issue on both?

Levi-Armstrong, 2020-05-26:
I was using Ogre1, Ubuntu 18.04 and ignition-rendering3.

iche033, 2020-05-26:
The collada loader [transparency fix](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-rendering/pull-requests/2616) should be ported to ign-common. Could you post the `<transparency>` and `<transparent`> tags from your collada file?

Levi-Armstrong, 2020-05-27:
In my case it only had the transparent tag and did not have the transparency tag. I removed the lines below and it showed up, but the comments in the link also suggested adding the transparency  tag setting the alpha to one also works.
```
<transparent opaque="A_ONE">
     <color>1 1 1 1</color>
 </transparent>
```

iche033, 2020-05-27:
made a fix in ignitionrobotics/ign-common#68

chapulina, 2020-06-05:
`ign-common` PR merged, the fix will be available on the next release.

closed 2020-06-05 by chapulina