gazebosim/gz-rendering #90: Failed to load plugin [ignition-rendering3-ogre2]

agayev169, 2020-05-26:
I am trying to build ign-rendering from the source. Although, building and installing are done successfully (I use Ogre and Ogre 2, without Optix), when I run ign-gazebo I get the following errors:
```
[Err] [RenderEngineManager.cc:394] Failed to load plugin [ignition-rendering3-ogre2] : couldn't load library on path [/usr/local/lib/libignition-rendering3-ogre2.so].
[Err] [RenderUtil.cc:1024] Engine [ogre2] is not supported
```
I have already checked the file, it is a symlink to a file which is a symlink to the another ".so" file. ```ldd``` of all of three files shows that there is no problem with the links (at least I cannot see it). Any ideas why it happens?

iche033, 2020-05-26:
can you see if you can build and run any one of ign-rendering the [examples](https://github.com/ignitionrobotics/ign-rendering/tree/master/examples)?

do you have other versions of ign-rendering installed in the system?

agayev169, 2020-05-27:
> can you see if you can build and run any one of ign-rendering the examples?
Yes, I tried several examples and they execute successfully.
> do you have other versions of ign-rendering installed in the system?
No, only ign-rendering3 built from source.

agayev169, 2020-05-28:
I tried to recode everything and in the end, found what caused the error. After I removed the folowing line everything works as expected:
```
#include <ignition/plugin/Register.hh>
```

iche033, 2020-05-28:
did you remove that from the ign-rendering side or ign-gazebo?

agayev169, 2020-05-28:
I have a plugin for the camera sensor. I removed the line from the code of the plugin.

iche033, 2020-05-28:
ok I see. To confirm, no changes are required here in ign-rendering right? Can we close this issue?

agayev169, 2020-05-28:
I am not sure. I do not have enough time to look at the source code of several packages, but it seems that ign-rendering (at least ign-rednering3-ogre2) and ign-plugin are not compatible with each other.
So, that may mean that there are issues either in ign-rendering or ign-plugin or in both.

acxz, 2021-01-01:
I'm also facing this issue (I believe it is the same) when trying to run the package on Arch Linux. It is important to note that I have built `ign-rendering` with `ogre` and not `ogre2`.

Running `ign gazebo` results in a seg fault:

<details>
<summary> Log </summary>

  ```
[1;31m[GUI] [Err] [RenderEngineManager.cc:461] [0m[1;31mFailed to load plugin [[0m[1;31mogre2[0m[1;31m] : couldn't find shared library.[0m[1;31m[0m
[1;31m[GUI] [Err] [RenderUtil.cc:1223] [0m[1;31mEngine [[0m[1;31mogre2[0m[1;31m] is not supported. [0m[1;31mLoading OGRE2 instead.[0m[1;31m[0m
[1;31m[GUI] [Err] [RenderEngineManager.cc:461] [0m[1;31mFailed to load plugin [[0m[1;31mogre2[0m[1;31m] : couldn't find shared library.[0m[1;31m[0m
/usr/lib/ruby/ignition/cmdgazebo4.rb:426: [BUG] Segmentation fault at 0x0000000000000000
ruby 2.7.2p137 (2020-10-01 revision 5445e04352) [x86_64-linux]

-- Control frame information -----------------------------------------------
c:0007 p:---- s:0058 e:000057 CFUNC  :call
c:0006 p:0016 s:0053 e:000052 METHOD /usr/lib/ruby/ignition/cmdgazebo4.rb:426
c:0005 p:0060 s:0047 e:000046 BLOCK  /usr/lib/ruby/ignition/cmdgazebo4.rb:456 [FINISH]
c:0004 p:---- s:0044 e:000043 CFUNC  :fork
c:0003 p:0654 s:0040 e:000039 METHOD /usr/lib/ruby/ignition/cmdgazebo4.rb:452
c:0002 p:0615 s:0019 E:002038 EVAL   /usr/bin/ign:267 [FINISH]
c:0001 p:0000 s:0003 E:000c50 (none) [FINISH]

-- Ruby level backtrace information ----------------------------------------
/usr/bin/ign:267:in `<main>'
/usr/lib/ruby/ignition/cmdgazebo4.rb:452:in `execute'
/usr/lib/ruby/ignition/cmdgazebo4.rb:452:in `fork'
/usr/lib/ruby/ignition/cmdgazebo4.rb:456:in `block in execute'
/usr/lib/ruby/ignition/cmdgazebo4.rb:426:in `runGui'
/usr/lib/ruby/ignition/cmdgazebo4.rb:426:in `call'

-- Machine register context ------------------------------------------------
 RIP: 0x00007f1f5def1c5a RBP: 0x00007f1f0bffe800 RSP: 0x00007f1f0bffe7e0
 RAX: 0x00007f1f0bffe870 RBX: 0x000055f06b5d9848 RCX: 0x00007f1f000008d0
 RDX: 0x000055f06aa1eb40 RDI: 0x00007f1f0bffe890 RSI: 0x0000000000000000
  R8: 0x0000000000000000  R9: 0x0000000000000008 R10: 0x0000000000000000
 R11: 0x0000000000000293 R12: 0x00007f1f0bffe880 R13: 0x00007f1f0bffe830
 R14: 0x00007f1f0bffe890 R15: 0x00007f1f0bffe860 EFL: 0x0000000000010246

-- C level backtrace information -------------------------------------------
/usr/lib/libruby.so.2.7(0x7f1f7d937c04) [0x7f1f7d937c04]
/usr/lib/libruby.so.2.7(0x7f1f7d7a2019) [0x7f1f7d7a2019]
/usr/lib/libruby.so.2.7(0x7f1f7d8b2b3a) [0x7f1f7d8b2b3a]
/usr/lib/libc.so.6(__restore_rt+0x0) [0x7f1f7d5726a0]
/usr/lib/libignition-gazebo4-rendering.so.4(_ZN8ignition6gazebo2v410RenderUtil4InitEv+0x26a) [0x7f1f5def1c5a]
/usr/lib/ign-gazebo-4/plugins/gui/libGzScene3D.so(_ZN8ignition6gazebo2v411IgnRenderer10InitializeEv+0x80) [0x7f1f5dfafeb0]
/usr/lib/ign-gazebo-4/plugins/gui/libGzScene3D.so(_ZN8ignition6gazebo2v412RenderThread10RenderNextEv+0x89) [0x7f1f5dfb61e9]
/usr/lib/libQt5Core.so.5(_ZN7QObject5eventEP6QEvent+0x292) [0x7f1f790e1582]
/usr/lib/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x82) [0x7f1f78641752]
/usr/lib/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x12a) [0x7f1f790b4a7a]
/usr/lib/libQt5Core.so.5(_ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData+0x173) [0x7f1f790b7573]
/usr/lib/libQt5Core.so.5(0x7f1f7910e0a4) [0x7f1f7910e0a4]
/usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x184) [0x7f1f7681fa84]
/usr/lib/libglib-2.0.so.0(0x7f1f768739b1) [0x7f1f768739b1]
/usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f1f7681e2b1]
/usr/lib/libQt5Core.so.5(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x61) [0x7f1f7910d6e1]
/usr/lib/libQt5Core.so.5(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x12c) [0x7f1f790b33fc]
/usr/lib/libQt5Core.so.5(_ZN7QThread4execEv+0x62) [0x7f1f78ecbd22]
/usr/lib/libQt5Core.so.5(0x7f1f78eccf0f) [0x7f1f78eccf0f]
/usr/lib/libpthread.so.0(start_thread+0xe9) [0x7f1f7d5023e9]
/usr/lib/libc.so.6(clone+0x43) [0x7f1f7d635293]

-- Other runtime information -----------------------------------------------

* Loaded script: /usr/bin/ign

* Loaded features:

    0 enumerator.so
    1 thread.rb
[... 1878 lines cut ...]
7f1f7d6fe000-7f1f7d72b000 r--p 00000000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f1f7d72b000-7f1f7d942000 r-xp 0002d000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f1f7d942000-7f1f7da3a000 r--p 00244000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f1f7da3a000-7f1f7da43000 r--p 0033b000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f1f7da43000-7f1f7da44000 rw-p 00344000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f1f7da44000-7f1f7da60000 rw-p 00000000 00:00 0 
7f1f7da60000-7f1f7da62000 r--p 00000000 103:04 6163799                   /usr/lib/ld-2.32.so
7f1f7da62000-7f1f7da83000 r-xp 00002000 103:04 6163799                   /usr/lib/ld-2.32.so
7f1f7da83000-7f1f7da8c000 r--p 00023000 103:04 6163799                   /usr/lib/ld-2.32.so
7f1f7da8c000-7f1f7da8d000 r--p 0002b000 103:04 6163799                   /usr/lib/ld-2.32.so
7f1f7da8d000-7f1f7da8f000 rw-p 0002c000 103:04 6163799                   /usr/lib/ld-2.32.so
7ffd7d111000-7ffd7d910000 rw-p 00000000 00:00 0                          [stack]
7ffd7d99c000-7ffd7d9a0000 r--p 00000000 00:00 0                          [vvar]
7ffd7d9a0000-7ffd7d9a2000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]
  ```

</details>

Sys info:
OS: Arch Linux
GCC: 10.2.0
Qt: 5.15.2
ruby: 2.7.2
protobuf: 3.12.4