gazebosim/gz-rendering #91: How to visualize both a Point Cloud and Octomap?

Levi-Armstrong, 2020-05-30:
I am interested to learn how to visualize a Point Cloud and Octomap in Ignition Rendering. I am currently using a depth sensor in ignition to generating a point cloud and I currentlypublish it for visualization in Rviz. I have an application built using Ignition GUI and if possible I would like to visualize the information in the ignition scene.

Levi-Armstrong, 2020-05-30:
It looks like I may be able to leverage the scene marker which supports points, but need to test. Still looking into how best to visualize an Octomap.

iche033, 2020-06-01:
for now you may be able to visualize point clouds by publishing to the marker topic like in the [marker example](https://github.com/ignitionrobotics/ign-gazebo/tree/master/examples/standalone/marker) in ign-gazebo.

We just started a GOSC project to create a lidar visual (issue #84) in ign-rendering and may extend to point cloud visualization if there is time in the end, cc @mihirk284

Levi-Armstrong, 2020-06-04:
I will hold of on the point cloud visual and track issue #84 to avoid duplicating work. 

Though I could look at the Octomap. Would your preferred approach be similar to issue #84 for the octomap? I could begin working on an OctomapVisual.

iche033, 2020-06-05:
> Though I could look at the Octomap. Would your preferred approach be similar to issue #84 for the octomap? I could begin working on an OctomapVisual.

yeah that'd would be great, thanks! I have not looked at rviz's implementation to see how it's done but it looks the visualization just consists of colored cubes. Hopefully it'll be straightforward.

Levi-Armstrong, 2020-06-05:
> yeah that'd would be great, thanks! I have not looked at rviz's implementation to see how it's done but it looks the visualization just consists of colored cubes. Hopefully it'll be straightforward.

That is what I am thinking. I hope I can use what is in rviz and integrating into the ogre1 and ogre2.

Levi-Armstrong, 2023-07-03:
I just now getting back to revisiting this. @iche033 Has any progress been made on visualizing point clouds or octomaps?

iche033, 2023-07-12:
We now have a [LidarVisual](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/include/gz/rendering/LidarVisual.hh) class for visualizing lidar point clouds. The ogre2 implementation is here: https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2LidarVisual.cc.

There is a gui plugin in gz-sim that essentially subscribes to lidar data to populates the LidarVisual:
https://github.com/gazebosim/gz-sim/tree/gz-sim7/src/gui/plugins/visualize_lidar