gazebosim/gz-rendering #92: Collada file rendering with a blue tint when exported from Blender 2.82.

Levi-Armstrong, 2020-05-31:
I recently upgraded from Blender 2.79 to 2.82 and when exporting collada file from 2.82 they all have a blue tint to them in ignition rendering. I have attached a link from blender 2.79 and 2.82 along with an image.

[geom.zip](https://github.com/ignitionrobotics/ign-rendering/files/4707052/geom.zip)

[image]

Levi-Armstrong, 2023-01-25:
@iche033 Do you have any thoughts on this? This can also be seen in my recent issue #810. I see different results when using ogre and ogre2.

Levi-Armstrong, 2023-01-25:
Look at the content of the files version 2.79 outputs material using phong and version 2.82 uses lambert.

Version 2.79
``` xml
    <effect id="gray-effect">
      <profile_COMMON>
        <technique sid="common">
          <phong>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <ambient>
              <color sid="ambient">0 0 0 1</color>
            </ambient>
            <diffuse>
              <color sid="diffuse">0.64 0.64 0.64 1</color>
            </diffuse>
            <specular>
              <color sid="specular">0.0125 0.0125 0.0125 1</color>
            </specular>
            <shininess>
              <float sid="shininess">205</float>
            </shininess>
            <index_of_refraction>
              <float sid="index_of_refraction">1</float>
            </index_of_refraction>
          </phong>
        </technique>
      </profile_COMMON>
    </effect>
```

Version 2.82
``` xml
    <effect id="gray-effect">
      <profile_COMMON>
        <technique sid="common">
          <lambert>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <diffuse>
              <color sid="diffuse">0.64 0.64 0.64 1</color>
            </diffuse>
            <reflectivity>
              <float sid="specular">0.5</float>
            </reflectivity>
            <index_of_refraction>
              <float sid="ior">1</float>
            </index_of_refraction>
          </lambert>
        </technique>
      </profile_COMMON>
    </effect>
```

Levi-Armstrong, 2023-01-25:
I noticed that all of the dae in gz-common do contain lambert materials but do not contain emission, reflectivity or index_of_reflectivity.

Levi-Armstrong, 2023-01-25:
It looks like the current collada parser for lambert material does not process the following tags. Is this a issue?

- specular
- shininess
- index_of_refraction

iche033, 2023-01-25:
ah I was wondering why your model appears blue in https://github.com/gazebosim/gz-rendering/issues/810 but I don't see that in mine.

I tested `link_3.dae` and `link_4.dae` and I don't get the blue tint though. 

I created a [`dae_blue_tint`](https://github.com/gazebosim/gz-rendering/compare/ign-rendering6...dae_blue_tint) branch to test this. I updated the `mesh_viewer` example to load your `link_3.dae` and `link_4.dae` in the geom.zip` file.

To build:

1. check out `dae_blue_tint` branch 
1. Go to `examples/mesh_viewer` dir and

    ```
    mkdir build
    cd build
    cmake ..
    make

    # run with ogre render engine
    ./mesh_viewer ogre

    # run with ogre2 render engine
    ./mesh_viewer ogre
    ```

ogre:

[image]

ogre2:

[image]

Levi-Armstrong, 2023-01-26:
@iche033 I have tracked down the cause of the blue tint. It is a result of the ambient tag not be provided. I have not tracked down in the code that is causing this but maybe this is something that has already been fixed on the latest branch. 

The lambert material causing blue tent.
``` xml
        <technique sid="common">
          <lambert>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <diffuse>
              <color sid="diffuse">1 1 1 1</color>
            </diffuse>
            <index_of_refraction>
              <float sid="ior">1.45</float>
            </index_of_refraction>
          </lambert>
        </technique>
```

Fixed by adding the ambient tag
``` xml
        <technique sid="common">
          <lambert>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <ambient>
              <color sid="ambient">0 0 0 1</color>
            </ambient>
            <diffuse>
              <color sid="diffuse">1 1 1 1</color>
            </diffuse>
            <index_of_refraction>
              <float sid="ior">1.45</float>
            </index_of_refraction>
          </lambert>
        </technique>
```

Levi-Armstrong, 2023-01-26:
Well the change above only fixes it ogre and not ogre2.

Levi-Armstrong, 2023-01-26:
If ambient is not provided for both ogre and ogre2 it defaults to [0.4 0.4 0.4 1]

Levi-Armstrong, 2023-01-26:
@iche033, I was able to finally track down the root cause. In my case I was setting the scene ambient light to [0, 0, 1, 1] where gazebo sim sets it to [0.4, 0.4, 0.4, 1.0]. If I us the same values as gazebo the blue tint is gone in both ogre and ogre2 regardless of whether ambient is set for the mesh material, and if you set gazebo sim to use mine you will see the blue tint. Not sure if this is an indication of an issue or not.

Levi-Armstrong, 2023-01-26:
More results with setting scene ambient light to Blue, Green and Red. I guess it makes sense now, but don't understand why only some models had the tent and other did not. 

**Scene Ambient Light [0, 0, 1, 1]**
[image]

**Scene Ambient Light [0, 1, 0, 1]**
[image]

**Scene Ambient Light [1, 0, 0, 1]**
[image]

iche033, 2023-01-26:
> @iche033, I was able to finally track down the root cause. In my case I was setting the scene ambient light to [0, 0, 1, 1]

ok that makes sense! The blue tint on the objects as the result of blue ambient light is expected. The scene ambient light color will provide the "base color" for all objects in the scene.  The final color of an object is the combined result of the color of scene ambient light, color of other light sources (e.g. directional light), and color of the visual object itself.

> Well the change above only fixes it ogre and not ogre2.

Some differences between ogre 1.x and 2.x: 

* ogre 1.x uses Blinn-Phong shading where the color of visual object is determined by ambient, diffuse, and specular components. So when the `<color sid="ambient">0 0 0 1</color>` tag is present in the blender dae mesh, the combined (multiplied) ambient value is now zero ([0, 0, 1] *  [0, 0, 0]) so you don't see the blue tint.
* ogre 2.x uses PBR, which ignores the ambient component of the object, e.g. `visual->SetAmbient(...)` has no effect. The scene ambient lighting will still be applied so you see the blue tint.

Levi-Armstrong, 2023-01-26:
Thanks for the explanation; it was very helpful. I think it is good to close this now.

closed 2023-01-26 by Levi-Armstrong