gazebosim/gz-rendering #93: Dummy ignition::rendering::BaseCamera::SaveFrame

agayev169, 2020-06-02:
```ignition::rendering::BaseCamera::SaveFrame``` is a dummy object. Please, either mention this in the documentation or write the function.

chapulina, 2020-06-08:
All functions on [BaseCamera](https://github.com/ignitionrobotics/ign-rendering/blob/master/include/ignition/rendering/base/BaseCamera.hh) are virtual functions that can be reimplemented by each rendering engine.

So I imagine what you're asking for is for that function to be implemented on either Ogre or Ogre2, right?

agayev169, 2020-06-08:
I use Ogre2 in my project, that is right. And as far as I understand, I cannot save an image from the camera itself. The way I did it is by capturing the image and then saving it with the help of ign-common.

I think it may be better to implement the function or override it for Ogre2Camera. Otherwise, a lot of users will use the function without understanding why it does not do anything (which happened to me).

chapulina, 2020-06-08:
Understood. I don't think we'll have time to work on this anytime in the future, but pull requests are welcome :wink: