gazebosim/gz-rendering #96: Unable to build against official Ogre2.1

stefanbuettner, 2020-06-04:
Hey there,

I'm trying to package ign-rendering 3.1.0 with a self packaged Ogre 2.1 version, both using the conan package manager. Packaging other ignition projects worked like a charm so far as did packaging Ogre 2.1. However, when building the rendering module, IgnOGRE2 cannot be found. I checked the ignition-cmake's [FindIgnOGRE2.cmake](https://github.com/ignitionrobotics/ign-cmake/blob/ignition-cmake2_2.2.0/cmake/FindIgnOGRE2.cmake) of version 2.2.0 which I'm using and found that for non-WIN32 architectures some [pkg_config approach is used](https://github.com/ignitionrobotics/ign-cmake/blob/ignition-cmake2_2.2.0/cmake/FindIgnOGRE2.cmake#L96) whereas on WIN32 cmake's [find_package is used](https://github.com/ignitionrobotics/ign-cmake/blob/ignition-cmake2_2.2.0/cmake/FindIgnOGRE2.cmake#L272). I was able to fix this by setting the `PKG_CONFIG_PATH` but then I get the following compilation error:
```
[ 36%] Building CXX object ogre2/src/CMakeFiles/ignition-rendering3-ogre2.dir/Ogre2DynamicRenderable.cc.o
~/.conan/data/ignition-rendering/3.1.0/user/channel/source/ogre2/src/Ogre2DepthCamera.cc: In member function ‘virtual void ignition::rendering::v3::Ogre2DepthCamera::PostRender()’:
~/.conan/data/ignition-rendering/3.1.0/user/channel/source/ogre2/src/Ogre2DepthCamera.cc:487:64: error: no matching function for call to ‘Ogre::RenderTexture::copyContentsToMemory(Ogre::PixelBox&, Ogre::RenderTarget::FrameBuffer)’
   rt->copyContentsToMemory(dstBox, Ogre::RenderTarget::FB_FRONT);
                                                                ^
In file included from ~/.conan/data/ogre3d/2.1.0/user/channel/package/9bfdcfa2bb925892ecf42e2a018a3f3529826676/include/OGRE/Ogre.h:91:0,
                 from ~/.conan/data/ignition-rendering/3.1.0/user/channel/source/ogre2/include/ignition/rendering/ogre2/Ogre2Includes.hh:26,
                 from ~/.conan/data/ignition-rendering/3.1.0/user/channel/source/ogre2/include/ignition/rendering/ogre2/Ogre2Conversions.hh:27,
                 from ~/.conan/data/ignition-rendering/3.1.0/user/channel/source/ogre2/src/Ogre2DepthCamera.cc:28:
~/.conan/data/ogre3d/2.1.0/user/channel/package/9bfdcfa2bb925892ecf42e2a018a3f3529826676/include/OGRE/OgreRenderTexture.h:54:22: note: candidate: virtual void Ogre::RenderTexture::copyContentsToMemory(const Ogre::Box&, const Ogre::PixelBox&, Ogre::RenderTarget::FrameBuffer)
         virtual void copyContentsToMemory(const Box& src, const PixelBox &dst, FrameBuffer buffer = FB_AUTO);
                      ^~~~~~~~~~~~~~~~~~~~
~/.conan/data/ogre3d/2.1.0/user/channel/package/9bfdcfa2bb925892ecf42e2a018a3f3529826676/include/OGRE/OgreRenderTexture.h:54:22: note:   no known conversion for argument 2 from ‘Ogre::RenderTarget::FrameBuffer’ to ‘const Ogre::PixelBox&’
ogre2/src/CMakeFiles/ignition-rendering3-ogre2.dir/build.make:110: recipe for target 'ogre2/src/CMakeFiles/ignition-rendering3-ogre2.dir/Ogre2DepthCamera.cc.o' failed
make[2]: *** [ogre2/src/CMakeFiles/ignition-rendering3-ogre2.dir/Ogre2DepthCamera.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

The function changed in ogre commit [548255e70](https://github.com/OGRECave/ogre-next/commit/548255e706e0e74ff651f51443e441afd0810a14) which seems to have been merged a while ago into the v2-1 branch. The 2.1 tag is from 6 weeks ago.

Now my questions are:
1. Which Ogre-next commit is the Ogre2 build based on which you provide as a debian package?
2. Why not using cmake's find_package for non-WIN32 systems as well?
3. Am I missing something?

Thanks for your help and cheers,
Stefan

iche033, 2020-06-04:
1. we're using this fork: https://github.com/osrf/ogre-2.1-release. I think we had to make some changes in order to package it for ubuntu so that it is side-by-side installable with ogre 1.x . Is that correct @j-rivero ?

2. You mean to find ogre? We use this: https://github.com/ignitionrobotics/ign-cmake/blob/master/cmake/FindIgnOGRE2.cmake
So we can find distinguish ogre 2 from ogre 1 when both versions (source or debs) are installed.

3. There was some effort to update ign-rendering so that it's compatible with new ogre2 commits:
https://bitbucket.org/ignitionrobotics/ign-rendering/pull-requests/179
but we never got around to rolling out new ogre2.1 debs and we were also busy migrating to github

j-rivero, 2020-06-05:
>     1. Is that correct @j-rivero ?

That is correct, code must be relocated to be able to install side by side both versions and build mechanisms (cmake, pkg-config) updated to find the new locations.

stefanbuettner, 2020-07-09:
Thanks for the fast reply. I tested the forked repo but still got this error when compiling the rendering component
```
make[2]: *** No rule to make target 'OgreMain', needed by 'lib/libignition-rendering3-ogre.so.3.1.0'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```
even though IgnOgre2 was successfully found
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- ignition-rendering3 version 3.1.0
-- Operating system is Linux
-- Found CPack generators: DEB
-- Looking for ignition-math6 -- found version 6.4.0
-- Searching for dependencies of ignition-math6
-- Looking for ignition-math6 - found

-- Looking for ignition-common3 -- found version 3.5.0
-- Searching for dependencies of ignition-common3
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- Found DL: TRUE  
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Found UUID: TRUE  
-- Searching for <ignition-common3> component [graphics]
-- Looking for ignition-common3-graphics -- found version 3.5.0
-- Searching for dependencies of ignition-common3-graphics
-- Looking for ignition-math6 -- found version 6.4.0
-- Searching for <ignition-common3> component [events]
-- Looking for ignition-common3-events -- found version 3.5.0
-- Searching for dependencies of ignition-common3-events
-- Looking for ignition-math6 -- found version 6.4.0
-- Looking for ignition-common3 - found

-- Looking for ignition-plugin1 -- found version 1.1.0
-- Searching for dependencies of ignition-plugin1
-- Searching for <ignition-plugin1> component [all]
-- Looking for all libraries of ignition-plugin1 -- found version 1.1.0
-- Looking for ignition-plugin1 -- found version 1.1.0
-- Looking for ignition-plugin1-loader -- found version 1.1.0
-- Searching for dependencies of ignition-plugin1-loader
-- Looking for ignition-plugin1-register -- found version 1.1.0
-- Searching for dependencies of ignition-plugin1-register
-- Looking for ignition-plugin1 - found

-- Found FreeImage: TRUE (Required is at least version "3.9") 
-- Looking for FreeImage - found

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Looking for OpenGL - found

-- Checking for module 'OGRE-RTShaderSystem >= 1.8'
--   Found OGRE-RTShaderSystem , version 1.9.0
-- Checking for module 'OGRE-Terrain >= 1.8'
--   Found OGRE-Terrain , version 1.9.0
-- Checking for module 'OGRE-Overlay >= 1.8'
--   Found OGRE-Overlay , version 1.9.0
-- Looking for IgnOGRE - found

-- Boost version: 1.65.1
-- Looking for IgnOGRE2 - found

CUDA_TOOLKIT_ROOT_DIR not found or specified
-- Could NOT find CUDA (missing: CUDA_TOOLKIT_ROOT_DIR CUDA_NVCC_EXECUTABLE CUDA_INCLUDE_DIRS CUDA_CUDART_LIBRARY) 
-- optix library not found.  Please locate before proceeding.
-- OptiX headers (optix.h and friends) not found.  Please locate before proceeding.
-- optix Prime library not found.  Please locate before proceeding.
-- Could NOT find OptiX (missing: OptiX_FOUND) (Required is at least version "3.8.0")
-- Looking for OptiX - not found

-- BUILD WARNINGS
-- 	Cannot build component [optix] - Missing: OptiX
-- END BUILD WARNINGS

-- 
-- Searching for host SSE information
-- SSE2 found
-- SSE3 found
-- SSE4.1 found
-- SSE4.2 found
-- Configuring library: ignition-rendering3
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Adding 23 UNIT tests
-- Adding 7 INTEGRATION tests
-- Adding 1 PERFORMANCE tests
-- No tests have been specified for REGRESSION
-- Configuring library: ignition-rendering3-ogre
-- No tests have been specified for UNIT
-- Configuring library: ignition-rendering3-ogre2
-- No tests have been specified for UNIT
-- The program [cppcheck] was not found! Skipping codecheck setup
-- Build configuration successful
-- Build type: Release
-- Install prefix: /home/sb/.conan/data/ignition-rendering/3.1.0/ar/stable/package/4896a57efa54b1c69bbc2319124583bdb72cfdfb
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_EXPORT_NO_PACKAGE_REGISTRY
    CONAN_COMPILER
    CONAN_COMPILER_VERSION
    CONAN_CXX_FLAGS
    CONAN_C_FLAGS
    CONAN_EXPORTED
    CONAN_IN_LOCAL_CACHE
    CONAN_LIBCXX
    CONAN_SHARED_LINKER_FLAGS
```

I set the `PKG_CONFIG_PATH` to `<ignition-orge2-root>/lib/pkgconfig`.

stefanbuettner, 2020-07-14:
So finally I think I found the problem. The changes to the OGRE2 repo were not done on a cmake level but on a debian package level (specifically in debian/rules). Thus, in order to work with my setup, I would need to fix the cmake and the project's pkgconfig templates. I'm not gonna do that but rather find a way to use the debian packages.

Thanks for the pointers, though.

closed 2020-07-14 by stefanbuettner