gazebosim/gz-rendering #100: Support custom rendering engine plugins

chapulina, 2020-06-12:
Ignition Rendering provides a plugin interface so that different rendering engines can be loaded at runtime. All the APIs needed for a downstream project to implement their custom engine plugin should already be installed and visible from downstream projects.

However, at the moment, Ignition Rendering is only able to load engines that are provided within this package itself. These default engines are registered here:

https://github.com/ignitionrobotics/ign-rendering/blob/dfcf7c4a574b6b0e94419b7ba1ab3ebce64055f3/src/RenderEngineManager.cc#L337-L367

I'm not sure if there's any code path exposed to users which will allow them to load an engine other than these.

One suggestion for the TODO above is to install the default engines in a path where Ignition Rendering always knows to look for them, so that they don't need to be hardcoded.

---

The ultimate goal is to allow `ign-gazebo` to load custom rendering engines using the `--rendering-engine` option in combination with the `IGN_GAZEBO_RENDERING_ENGINE_PATH` environment variable, like how it's done for physics, see [this tutorial](https://github.com/ignitionrobotics/ign-gazebo/blob/master/tutorials/physics.md).

The [Finding Resources](https://github.com/ignitionrobotics/design/blob/2a2c22e6366ff3d6cf1feece03986d67199d645b/finding_resources.md) design document also offers some guidelines on how to let the users find custom engines.

Levi-Armstrong, 2020-06-21:
I believe this may be related to the current issue I am having with releasing a snap package levering Ignition GUI which fails to find the rendering engines even thought they are located in the snap package. I was having a similar issues with GUI plugins but was able to solve this by setting the `IGN_GUI_PLUGIN_PATH`.

Levi-Armstrong, 2020-06-21:
I created a PR #104 to support environment variable similar to ignition gui.

chapulina, 2020-09-09:
I looked a bit into this and got a proof of concept working:

<details><summary>See diff</summary>
<p>

```diff
diff --git a/examples/simple_demo/Main.cc b/examples/simple_demo/Main.cc
index 09358186..ad01220f 100644
--- a/examples/simple_demo/Main.cc
+++ b/examples/simple_demo/Main.cc
@@ -172,7 +172,7 @@ int main(int _argc, char** _argv)
   std::vector<CameraPtr> cameras;
 
   engineNames.push_back("ogre");
-  engineNames.push_back("optix");
+  engineNames.push_back("ignition-rendering3-ogre");
   for (auto engineName : engineNames)
   {
     try
diff --git a/src/RenderEngineManager.cc b/src/RenderEngineManager.cc
index 8d934ba8..d2d8f2c1 100644
--- a/src/RenderEngineManager.cc
+++ b/src/RenderEngineManager.cc
@@ -28,6 +28,13 @@
 #include "ignition/rendering/RenderEngineManager.hh"
 #include "ignition/rendering/RenderEnginePlugin.hh"
 
+/// \brief Holds information about an engine
+struct EngineInfo
+{
+  std::string name;
+  ignition::rendering::RenderEngine *engine;
+};
+
 /// \brief Private implementation of the RenderEngineManager class
 class ignition::rendering::RenderEngineManagerPrivate
 {
@@ -40,7 +47,7 @@ class ignition::rendering::RenderEngineManagerPrivate
   /// \brief Get a pointer to the render engine from an EngineMap iterator
   /// \param[in] _iter EngineMap iterator
   /// \param[in] _path Another search path for rendering engine plugin.
-  public: RenderEngine *Engine(EngineIter _iter,
+  public: RenderEngine *Engine(EngineInfo _info,
       const std::map<std::string, std::string> &_params,
       const std::string &_path);
 
@@ -151,17 +158,19 @@ RenderEngine *RenderEngineManager::Engine(const std::string &_name,
     const std::map<std::string, std::string> &_params,
     const std::string &_path)
 {
+  EngineInfo info{_name, nullptr};
+
   std::lock_guard<std::recursive_mutex> lock(this->dataPtr->enginesMutex);
-  // check in the list of available engines
-  auto iter = this->dataPtr->engines.find(_name);
 
-  if (iter == this->dataPtr->engines.end())
+  // check in the list of registered engines
+  auto iter = this->dataPtr->engines.find(_name);
+  if (iter != this->dataPtr->engines.end())
   {
-    ignerr << "No render-engine registered with name: " << _name << std::endl;
-    return nullptr;
+    info.name = iter->first;
+    info.engine = iter->second;
   }
 
-  return this->dataPtr->Engine(iter, _params, _path);
+  return this->dataPtr->Engine(info, _params, _path);
 }
 
 //////////////////////////////////////////////////
@@ -178,7 +187,7 @@ RenderEngine *RenderEngineManager::EngineAt(unsigned int _index,
   std::lock_guard<std::recursive_mutex> lock(this->dataPtr->enginesMutex);
   auto iter = this->dataPtr->engines.begin();
   std::advance(iter, _index);
-  return this->dataPtr->Engine(iter, _params, _path);
+  return this->dataPtr->Engine({iter->first, iter->second}, _params, _path);
 }
 
 //////////////////////////////////////////////////
@@ -284,28 +293,37 @@ void RenderEngineManager::UnregisterEngineAt(unsigned int _index)
 //////////////////////////////////////////////////
 // RenderEngineManagerPrivate
 //////////////////////////////////////////////////
-RenderEngine *RenderEngineManagerPrivate::Engine(EngineIter _iter,
+RenderEngine *RenderEngineManagerPrivate::Engine(EngineInfo _info,
     const std::map<std::string, std::string> &_params,
     const std::string &_path)
 {
-  RenderEngine *engine = _iter->second;
+  RenderEngine *engine = _info.engine;
 
   if (!engine)
   {
     // check if it's an engine in the list of default engines provided by
     // ign-rendering. If so, load it
-    auto defaultIt = this->defaultEngines.find(_iter->first);
+    auto defaultIt = this->defaultEngines.find(_info.name);
     if (defaultIt != this->defaultEngines.end())
     {
       std::string libName = defaultIt->second;
       if (this->LoadEnginePlugin(libName, _path))
       {
         std::lock_guard<std::recursive_mutex> lock(this->enginesMutex);
-        auto engineIt = this->engines.find(_iter->first);
+        auto engineIt = this->engines.find(_info.name);
         if (engineIt != this->engines.end())
           engine = engineIt->second;
       }
     }
+    // Otherwise, try to load as a shared library
+    else
+    {
+      std::string libName = _info.name;
+      if (this->LoadEnginePlugin(libName, _path))
+      {
+        // Register engine?
+      }
+    }
   }
 
   if (!engine)
```

</p>
</details>

I was trying to change as little as possible, so it needs to be cleaned up. I think it may be worth it revisiting the need for registering the "default" engines at compile time. I think it should be enough to be able to find their shared libraries at runtime.

Another thing that I think would be good to do is installing version-agnostic libraries like we do on `ign-gazebo` and `ign-physics`, see [this PR](https://osrf-migration.github.io/ignition-gh-pages/#!/ignitionrobotics/ign-physics/pull-requests/121/page/1). This means installing `libignition-rendering-ogre.so` as a symlink to `libignition-rendering3-ogre.so`. This way users can refer to the plugin without needing to specify some specific `ign-rendering` version.

Finally, I think we should only call this issue done when we have an example plugin in the example folder that can be loaded successfully. I think it doesn't need to render anything, but it should compile and run successfully. That would be the equivalent of [this ign-physics example](https://github.com/ignitionrobotics/ign-physics/tree/master/examples/hello_world_plugin).

JShep1, 2020-10-29:
All subtasks have been completed and this functionality is completed so I'm marking it as resolved.

closed 2020-10-29 by JShep1