gazebosim/gz-rendering #101: Object visible with a camera but invisible in the 3D World

ahcorde, 2020-06-16:
Porting the [ros_ign_gazebo_demos](https://github.com/ignitionrobotics/ros_ign/pull/82) I found this issue. The cone is visible with the camera but it's invisible in the 3D world.

[image]

I'm compiling from source `dome`.

@iche033 or @chapulina any thoughts?

iche033, 2020-06-16:
Not sure. The world is `sensors_demo.sdf` from ign-gazebo right? I just tested that world in ign-gazebo2, 3, 4 but the cone is shown in both 3d window and sensors. Is it visible for you if you just launch the world using ign instead of ros2?

mcres, 2020-06-29:
I am facing the same problem in Ubuntu 18.04 after installing `dome` from source and launching the world with `ign gazebo`.
I've also tried to launch some more models and it happens exactly the same. It seems that meshes in `.obj` like [this one](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/SurgicalTrolleyMed) will work, but `.dae` or `.stl` like [this one](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Black%20and%20Decker%20Cordless%20Drill) won't.

chapulina, 2020-07-01:
I can't reproduce this issue using `ros_ign_gazebo_demos` with:

* Ubuntu Focal (Docker)
* Either Citadel from debs or Dome from source
* Foxy from debs

Do you see any errors / warnings when running verbose?

Also, when you expand the "Construction" entity on the tree, does it show that it has a visual?

mcres, 2020-07-02:
I don't see any errors or warnings related.
The Construction entity does have a visual in the tree, but I can't see any difference between visuals of the different models when using the Component Inspector.

ahcorde, 2020-09-01:
Using this env variable fix the issue

```bash
export LC_NUMERIC="C"
```

can you confirm this @mcres ?

chapulina, 2020-09-01:
> export LC_NUMERIC="C"

Ohhhh locales strike back! Nice catch!

ahcorde, 2020-09-02:
should we need to add any reference to this issue in the installation guide or the troubleshooting sections?

chapulina, 2020-09-02:
I think we should be able to fix this programmatically. Were you able to pin-point the place where the locale made a difference? Is is when parsing the COLLADA files, when loading them into rendering...?