gazebosim/gz-rendering #105: Clean linter warnings

ahcorde, 2020-06-22:
Running the following script generates many linter warnings

```bash
bash tools/code_check.sh
```

The idea of this issue is clean all these warnings

chapulina, 2020-06-22:
That is running on CI and it's currently clear for all supported branches:

* https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering2-bionic-amd64/
* https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/
* https://build.osrfoundation.org/job/ignition_rendering-ci-master-bionic-amd64/

What happens often is that a different version of cppcheck will show different errors. So maybe this issue can be more explicit about which version we're trying to satisfy?

chapulina, 2021-05-06:
`tools/code_check.sh` is deprecated in favour of `make codecheck`, which is being tested on CI and is currently clear for all supported branches.

closed 2021-05-06 by chapulina