gazebosim/gz-rendering #128: Ignition Dome crash when trying to run Ogre2 on Nvidia card running nouveau drivers

clalancette, 2020-08-18:
My computer has an Nvidia GeForce GTX 1070 (NV130) card in it.  I'm currently running the nouveau drivers on kernel 5.7.8.  The glxinfo from that card is here: https://gist.github.com/clalancette/03a21867327fdcd13dd32bd320e21a4e

I've built ignition-dome from sources.  Here are the versions I'm currently using:
```
repositories:
  ign-cmake:
    type: git
    url: https://github.com/ignitionrobotics/ign-cmake
    version: 220e990588895e1a1773f2bea68f21c4b288e8f4
  ign-common:
    type: git
    url: https://github.com/ignitionrobotics/ign-common
    version: 48a670fc949760de8416760ce5e03b0151edd332
  ign-fuel-tools:
    type: git
    url: https://github.com/ignitionrobotics/ign-fuel-tools
    version: 9e5bec0df2703626df6a77d33b40d42e65ed01cf
  ign-gazebo:
    type: git
    url: https://github.com/ignitionrobotics/ign-gazebo
    version: 8ec8d66cc8a6b661a2db6ccae153a785287fa315
  ign-gui:
    type: git
    url: https://github.com/ignitionrobotics/ign-gui
    version: 58cedf4c96b4baf46a0c5c79234faf9e3dc4e924
  ign-launch:
    type: git
    url: https://github.com/ignitionrobotics/ign-launch
    version: 10f21dfe0bf4ffa2c4f9f4b0548054d33487159a
  ign-math:
    type: git
    url: https://github.com/ignitionrobotics/ign-math
    version: 6e80a49a17172bd24a39bf7316166f86413374cc
  ign-msgs:
    type: git
    url: https://github.com/ignitionrobotics/ign-msgs
    version: 424db1eeabd2561682173100daaa98932e299b7c
  ign-physics:
    type: git
    url: https://github.com/ignitionrobotics/ign-physics
    version: 5f987c5df0eb71237e2329a7ef61f73983ec41cb
  ign-plugin:
    type: git
    url: https://github.com/ignitionrobotics/ign-plugin
    version: 0f75e41d9c59b0f37562a25e16eb124dff783d2e
  ign-rendering:
    type: git
    url: https://github.com/ignitionrobotics/ign-rendering
    version: 42e20eda915fb6fc30dcbb4faffc359dceab26ac
  ign-sensors:
    type: git
    url: https://github.com/ignitionrobotics/ign-sensors
    version: 03e7f51295624be5d043f82b817a375bc127eefc
  ign-tools:
    type: git
    url: https://github.com/ignitionrobotics/ign-tools
    version: 6c6ed6a1308503a5a4865c610f6d99d607153b3a
  ign-transport:
    type: git
    url: https://github.com/ignitionrobotics/ign-transport
    version: d72d63ecf9c782c80ba6813a499111e3b9b13dea
  sdformat:
    type: git
    url: https://github.com/osrf/sdformat
    version: 5b53e44c161bc3ce9d31c30793e2b89c52494311
```

I can successfully build the entire workspace.  If I try to run one of the demos: `ign gazebo shapes.sdf`, I get a crash with a backtrace: https://gist.github.com/clalancette/193fd47c04c9a0ee1df5d736f47ad1ae

After some debugging with @mjcarroll and @azeey , we determined that the problem was that this demo was trying to use Ogre2.  If I switch to using Ogre1, everything works fine.

There are 2 separate problems here:
* It would be nice if this driver/ignition combination could work with Ogre2.  I'm not sure what is involved with that.
* It would be very helpful for the user if the failure wasn't a crash, and instead said something like "it doesn't look like Ogre2 is supported, try Ogre1 instead".

I'm happy to try out any fixes or give more information here.

mjcarroll, 2020-08-18:
Running `glxinfo` may help with debugging, for reference here is a working configuration on 18.04 + nvidia 1060 + nvidia drivers:

https://gist.github.com/mjcarroll/2cbb74400b1e85ac49a07aa80806c6a4

azeey, 2020-08-18:
It would be nice to know exactly what feature is missing that cause Ogre2 to fail. I have an Intel card that I would think is modern enough for Ogre2, but I can't run ign-gazebo on it.
https://gist.github.com/azeey/c677a8f675e4d6a1b989409ba9f384c8

chapulina, 2020-08-18:
Did you try the troubleshooting steps described here?

https://github.com/ignitionrobotics/docs/pull/48/files#diff-43525d0faa713d80ee1af06a9a0ce1d4R261

Also, one thing I think we should add to that troubleshooting section is setting this environment variable to see if it fixes OGRE2:

`MESA_GL_VERSION_OVERRIDE=3.3`

In any case, we shouldn't crash when the user is using an unsupported OpenGL version.

iche033, 2020-08-18:
could you share your `~/.ignition/rendering/oger2.log`? It should give more hints on why ogre2 is not happy

clalancette, 2020-08-18:
> Did you try the troubleshooting steps described here?
>
> https://github.com/ignitionrobotics/docs/pull/48/files#diff-43525d0faa713d80ee1af06a9a0ce1d4R261

Oops, no, I missed that.  If I run the debugging there, I do indeed see `OpenGL 3.3 is not supported. Please update your graphics card drivers.`.

> 
> Also, one thing I think we should add to that troubleshooting section is setting this environment variable to see if it fixes OGRE2:
> 
> `MESA_GL_VERSION_OVERRIDE=3.3`

And yep, that seems to fix it for me; it launches fine that way.

> 
> In any case, we shouldn't crash when the user is using an unsupported OpenGL version.

:+1:

clalancette, 2020-08-18:
> could you share your `~/.ignition/rendering/oger2.log`? It should give more hints on why ogre2 is not happy

Attached here: [ogre2.log](https://github.com/ignitionrobotics/ign-rendering/files/5091944/ogre2.log)

iche033, 2020-08-18:
I'll add a check for open gl version in the code and suggest overriding the opengl version if it fails to create the render window.

peci1, 2020-09-08:
I've noticed that on Bionic+Blueprint+xvfb (or TurboVNC server), the rendering thread correctly gets OpenGL Core Profile 3.3, but the GUI gets OpenGL 3.1.

Sensor RenderThread: (log generated by `xvfb-run -a ign launch -v4 cloudsim_sim.ign worldName:=simple_cave_01 robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1 headless:=true`).

```
16:59:44: ******************************
*** Starting GLX Subsystem ***
******************************
16:59:44: GL3PlusRenderSystem::_createRenderWindow "OgreWindow(0)_2097153", 1x1 windowed  miscParams: FSAA=0 border=none contentScalingFactor=1.000000 gamma=true parentWindowHandle=2097153 stereoMode=Frame Sequential
16:59:44: Created GL 3.3 context
16:59:44: GLXWindow::create used FBConfigID = 215
16:59:44: GL_VERSION = 3.3.0.0
16:59:44: GL_VENDOR = VMware, Inc.
16:59:44: GL_RENDERER = llvmpipe (LLVM 10.0.0, 256 bits)
...
17:03:29: **************************************
17:03:29: ***   OpenGL 3+ Renderer Started   ***
17:03:29: **************************************
```
And the headless Gazebo doesn't crash.

GUI thread: (log generated by `xvfb-run -a ign launch -v4 cloudsim_sim.ign worldName:=simple_cave_01 robotName1:=X1 robotConfig1:=X1_SENSOR_CONFIG_1`).
```
17:01:29: ******************************
*** Starting GLX Subsystem ***
******************************
17:01:29: GL3PlusRenderSystem::_createRenderWindow "OgreWindow(0)_0", 1x1 windowed  miscParams: FSAA=0 border=none contentScalingFactor=1,000000 currentGLContext=true externalGLControl=true gamma=true stereoMode=Frame Sequential
17:01:29: GLXWindow::create used FBConfigID = 206
17:01:29: GL_VERSION = 3.1.0.0
17:01:29: GL_VENDOR = VMware, Inc.
17:01:29: GL_RENDERER = llvmpipe (LLVM 10.0.0, 256 bits)
...
17:01:29: OGRE EXCEPTION(3:RenderingAPIException): OpenGL 3.3 is not supported. Please update your graphics card drivers. in GL3PlusRenderSystem::initialiseContext at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusRenderSystem.cpp (line 3415)
```

And the non-headless Gazebo crashes.

XVFB supports OpenGL 3.3 in core profile and 3.1 in compatibility profile:

```
$ xvfb-run -a glxinfo -B
name of display: :99
display: :99  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: VMware, Inc. (0xffffffff)
    Device: llvmpipe (LLVM 10.0.0, 256 bits) (0xffffffff)
    Version: 20.0.8
    Accelerated: no
    Video memory: 15772MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 10.0.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 3.1 Mesa 20.0.8
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
```

Running GUI with `MESA_GL_VERSION_OVERRIDE=3.3` works. So I'd guess this error would stem from the place where GUI/rendering threads ask for rendering context. They have to do something different.

It is weird that the GUI thread gets a different OpenGL version, because I thing OGRE should always be asking for the core profile:

https://github.com/OGRECave/ogre-next/blob/c8ec333fb85f0ccbd791b79aa53b69faac6573ca/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXGLSupport.cpp#L842

But maybe some settings/required extensions/whatever tell OpenGL that the core profile is not a good match?

Using `MESA_GL_VERSION_OVERRIDE=3.3` automatically seems more like a nasty workaround to me - it'd be much better to force GUI to acquire the OpenGL context in the same way the sensor thread does that.

iche033, 2020-09-08:
The opengl context is not created by OGRE on the GUI side so that could be the issue:
https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/gui/plugins/scene3d/Scene3D.cc#L1917

peci1, 2020-09-08:
According to https://stackoverflow.com/questions/55685351/how-to-choose-the-opengl-context-for-qtquick-applications , adding

```
QSurfaceFormat surfaceFormat;
surfaceFormat.setMajorVersion(3);
surfaceFormat.setMinorVersion(3);
surfaceFormat.setProfile(QSurfaceFormat::CoreProfile);
QSurfaceFormat::setDefaultFormat(surfaceFormat);
```

before 

https://github.com/ignitionrobotics/ign-gazebo/blob/9f5f2d097dd15cfde78d155bd5b23f9dfb7685f7/src/gui/Gui.cc#L281

should fix this issue. But that would maybe introduce the OpenGL 3.3 requirement even for GUIs which do not use Scene3D.

According to https://doc.qt.io/qt-5/qsurfaceformat.html#QSurfaceFormat, QT asks for OpenGL 2.0 by default, and the highest "forward-compatible" with 2.0 is 3.1. But that wouldn't explain why does this work on nvidia GPUs with binary blob drivers. Maybe they tell that they can't provide anything less than 3.3?

iche033, 2020-09-08:
>  But I'd really be surprised if OpenGL allowed to share a context with different surface format

That's my guess too. Maybe you can give that a quick test by setting the context to use a 3.3 format (instead of `current->format()`)? Code to change is here:
https://github.com/ignitionrobotics/ign-gazebo/blob/master/src/gui/plugins/scene3d/Scene3D.cc#L1926

peci1, 2020-09-09:
Surprise! I experimented with setting the surface format, and it seems that setting it on a shared context is no problem (whatever that means). I was able to run Ignition Gazebo after setting the surface format in Scene3D on integreated Intel GPU:

[image]

```
03:15:30: ******************************
*** Starting GLX Subsystem ***
******************************
03:15:30: GL3PlusRenderSystem::_createRenderWindow "OgreWindow(0)_0", 1x1 windowed  miscParams: FSAA=0 border=none contentScalingFactor=1.000000 currentGLContext=true externalGLControl=true gamma=true stereoMode=Frame Sequential
03:15:30: GLXWindow::create used FBConfigID = 204
03:15:30: GL_VERSION = 4.6.0.0
03:15:30: GL_VENDOR = Intel Open Source Technology Center
03:15:30: GL_RENDERER = Mesa DRI Intel(R) UHD Graphics 620 (KBL GT2)
03:15:30: GL_EXTENSIONS =
...
03:15:30: **************************************
03:15:30: ***   OpenGL 3+ Renderer Started   ***
03:15:30: **************************************
...
03:15:30: RenderSystem capabilities
03:15:30: -------------------------
03:15:30: RenderSystem Name: OpenGL 3+ Rendering Subsystem
03:15:30: GPU Vendor: intel
03:15:30: Device Name: Mesa DRI Intel(R) UHD Graphics 620 (KBL GT2)
03:15:30: Driver Version: 4.6.0.0
```

where

```
$ glxinfo -B
name of display: :1
display: :1  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) UHD Graphics 620 (KBL GT2) (0x5917)
    Version: 20.0.8
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) UHD Graphics 620 (KBL GT2)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 3.0 Mesa 20.0.8
OpenGL shading language version string: 1.30
OpenGL context flags: (none)

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

Without the fix, GUI uses the 3.0 compat context for GzScene3d plugin, which is not sufficient.

This also means the GUI is able to run under xvfb (not of much use probably, as it is a "black-hole" X server) and TurboVNC server (I tried connecting to the VNC display and GUI is there).

To my surprise, trying to fix it via Gui.cc did not help.

However, to be able to run everything on the integrated GPU, there is still a crash of the sensor thread:

```
[... 383 lines cut ...]
03:21:46:    - With different bit depths: yes
03:21:46:  * Point Sprites: yes
03:21:46:  * Extended point parameters: yes
03:21:46:  * Max Point Size: 255
03:21:46:  * Vertex texture fetch: yes
03:21:46:  * Number of world matrices: 0
03:21:46:  * Number of texture units: 16
03:21:46:  * Stencil buffer depth: 8
03:21:46:  * Number of vertex blend matrices: 0
03:21:46:    - Max vertex textures: 32
03:21:46:    - Vertex textures shared: yes
03:21:46:  * Render to Vertex Buffer : yes
03:21:46:  * Hardware Atomic Counters: yes
03:21:46:  * GL 1.5 without VBO workaround: no
03:21:46:  * Frame Buffer objects: yes
03:21:46:  * Frame Buffer objects (ARB extension): no
03:21:46:  * Frame Buffer objects (ATI extension): no
03:21:46:  * PBuffer support: no
03:21:46:  * GL 1.5 without HW-occlusion workaround: no
03:21:46:  * Vertex Array Objects: yes
03:21:46:  * Separate shader objects: no
```
</details>

Do you have an idea whether that's a bug or whether the driver really doesn't offer what's required? But given the fact that the sensor thread runs fine under xvfb and xvnc, I'd really think it is more likely a bug than a missing feature.

peci1, 2020-09-09:
See  ignitionrobotics/ign-gazebo#339 .

peci1, 2020-09-09:
A similar effect is achieved by using `MESA_GL_VERSION_OVERRIDE=3.3FC` (the FC is important, otherwise it again selects compat profile).

iche033, 2020-09-15:
> terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  OGRE EXCEPTION(2:InvalidParametersException): All framebuffer formats with this texture internal format unsupported in GL3PlusFrameBufferObject::initialise at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusFrameBufferObject.cpp (line 229)

that error msg comes up often and it usually points to an issue with the graphics driver, e.g. this [post](https://answers.gazebosim.org//question/29/gzclient-throwing-ogreinvalidparametersexception/) and this [issue](https://github.com/ros-visualization/rviz/issues/1123). In this case, the mesa driver may not support the texture format that's being requested. I think it maybe an issue with floating point texture format used by depth/rgbd/lidar sensors. Normal rgb camera may work, hopefully.

iche033, 2023-03-09:
closing this as it looks like a few PRs addressing this issue were merged. The [troubleshooting guide](https://github.com/gazebosim/docs/blob/master/garden/troubleshooting.md#unable-to-create-the-rendering-window) also now has more info on useful env vars to set. Please reopen if anyone runs into this issue again.

closed 2023-03-09 by iche033