gazebosim/gz-rendering #134: COLLADA Rendering Failure: instance_geometry Conflict

jaelrod, 2020-08-31:
I have the need to programmatically generate some COLLADA files to simulate some dynamic assets in Gazebo. Loosely following COLLADA specification, I am declaring a simple triangles mesh in library_geometries and trying to instantiate that same geometry multiple times with different transformation matrices and materials applied to it. Attached is a file that demonstrates the below screenshots.
[mesh.dae.txt](https://github.com/ignitionrobotics/ign-rendering/files/5152607/mesh.dae.txt)

This is the mesh loaded into Blender, looking as expected:
[image]

However, the same exact COLLADA file loaded into Ignition Gazebo (v3.2.0) looks like the following. The rendered mesh has only one plane instead of two, and that plane takes the shape of the instance_geometry that is declared first and the bind_material of the instance_geometry declared second:
[image]

I seem to be able to work around this by defining duplicate geometries inside of library_geometries for each instance_geometry, but that sort of defeats the point of being able to reuse recurring shapes from the geometry library.

chapulina, 2020-09-01:
Thank you for ticketing the issue. The fix may need to go into [ignition::common::ColladaLoader](https://github.com/ignitionrobotics/ign-common/blob/master/graphics/src/ColladaLoader.cc), which is in a different library.

jaelrod, 2020-09-01:
Sounds good. My apologies for placing the issue in the wrong repo. After ticketing this yesterday, I encountered a separate but similar issue, so I'll put that one there. Is COLLADA the preferred asset representation format for Ignition Gazebo? Is there better support for a different format?

chapulina, 2020-09-01:
> My apologies for placing the issue in the wrong repo.

No worries. There's a possibility that the issue is coming from here, I just mentioned the collada loader for completeness.

> Is COLLADA the preferred asset representation format for Ignition Gazebo?

There's also support for OBJ, you may have better luck with that, depending on the asset. For animations, we also support BVH.

closed 2024-05-31 by azeey