gazebosim/gz-rendering #139: Textures from different models with same base name are reused in Ogre2

chapulina, 2020-09-10:
The Ogre 2 plugin is using a texture's file name to uniquely identify each texture. This means that all textures called `texture.png` will be treated as the same, even if they're in different directories.

For example, all (most?) models in this collection have texture files called `texture.png`. This means that all models will use the texture from the first model. For example, load this world on `ign-gazebo`:

<details><summary>world.sdf</summary>
<p>

```
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="shapes">
    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>

    <gui fullscreen="0">

      <!-- 3D scene -->
      <plugin filename="GzScene3D" name="3D View">
        <ignition-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </ignition-gui>

        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>1.0 1.0 1.0</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
      </plugin>
    </gui>

    <light type="directional" name="sun">
      <cast_shadows>true</cast_shadows>
      <pose>0 0 10 0 0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>

    <model name="ground_plane">
      <static>true</static>
      <link name="link">
        <collision name="collision">
          <geometry>
            <plane>
              <normal>0 0 1</normal>
            </plane>
          </geometry>
        </collision>
        <visual name="visual">
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.8 0.8 0.8 1</specular>
          </material>
        </visual>
      </link>
    </model>

<include>
<uri>
https://fuel.ignitionrobotics.org/1.0/GoogleResearch/models/adizero_F50_TRX_FG_LEA
</uri>
</include>
<include>
<pose>0 0.5 0 0 0 0</pose>
<uri>
https://fuel.ignitionrobotics.org/1.0/GoogleResearch/models/adistar_boost_m
</uri>
</include>
  </world>
</sdf>
```

</p>
</details>

See how the 2nd model uses the 1st model's texture:

[image]

As a comparison, change `ogre2` to `ogre` in the world, and see that each models loads its own texture:

[image]

I believe this can be fixed by using the whole texture path instead of just the basename here:

https://github.com/ignitionrobotics/ign-rendering/blob/46bf31837a347955a483de6dab3ad45e42370666/ogre2/src/Ogre2Material.cc#L441-L443

---

If I remember correctly, this was done on purpose so that all the cave tiles on SubT use the same texture. This should be resolved by allowing models to reference each other's textures (https://github.com/ignitionrobotics/ign-gazebo/issues/343) and then doing that for all the cave tiles.

iche033, 2020-09-10:
> If I remember correctly, this was done on purpose so that all the cave tiles on SubT use the same texture. This should be resolved by allowing models to reference each other's textures (ignitionrobotics/ign-gazebo#343) and then doing that for all the cave tiles.

yes that's correct. Unfortunately we found a significant increase in subt's VRAM usage after changing the basename to whole texture path (since each texture is now treated as unique texture) so we were able to apply the fix.

chapulina, 2021-11-15:
With SubT over, I'd like a path forward to closing this issue. Do you think we're ready for the proper fix, @iche033? I think that models that want to reuse textures should already be able to refer to those resources directly on the SDF, right?