gazebosim/gz-rendering #140: RenderEngine->CreateScene("scene") returns null in many tests

jacobperron, 2020-09-22:
`RenderEngine->CreateScene("scene")` is returning null is many places during testing causing all sorts of failures:

https://build.osrfoundation.org/job/ignition_rendering-ci-master-bionic-amd64/24/testReport/

The other half of test failures are caused by an Ogre exception:

```
unknown file
C++ exception with description "OGRE EXCEPTION(2:InvalidParametersException): All framebuffer formats with this texture internal format unsupported in GL3PlusFrameBufferObject::initialise at /var/lib/jenkins/workspace/ogre-2.1-debbuilder/repo/RenderSystems/GL3Plus/src/OgreGL3PlusFrameBufferObject.cpp (line 229)" thrown in the test body.
```

These failures started happening at the same time, though I'm not sure if they are related.

jacobperron, 2020-09-22:
This is also affecting `ign-rendering3`: https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/20/testReport/

chapulina, 2020-09-22:
I think the problem with those builds is that they were running on `r2d2`, which is having GPU issues. @j-rivero marked that node as non-GPU this morning so `ign-rendering` CI shouldn't run on it anymore. I triggered new builds, let's see how they look:

* Dome [image]
* Citadel [image]

iche033, 2020-09-22:
related [comment](https://github.com/ignitionrobotics/ign-rendering/pull/130#issuecomment-691557456)

chapulina, 2020-09-22:
The latest non-r2d2 builds don't have the Ogre exception failures.

closed 2020-09-22 by chapulina