gazebosim/gz-rendering #170: Camera/CameraTest.VisualAt/ogre2 flaky test on Bionic/amd64

j-rivero, 2020-10-28:
I've seen the `Camera/CameraTest.VisualAt/ogre2` failing in [some](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/21/testReport/) [builds](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/25/) for Bionic/amd64.
[Changes listed in build 20](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/20/)seems to be the starting point of the failures.

```
54: [ RUN      ] Camera/CameraTest.VisualAt/ogre2
54: [1;32m[Msg] [0m[1;32mLoading plugin [[0m[1;32mignition-rendering-ogre2[0m[1;32m][0m[1;32m[0m
54: /var/lib/jenkins/workspace/ignition_rendering-ci-ign-rendering3-bionic-amd64/ign-rendering/test/integration/camera.cc:243: Failure
54: Expected: (nullptr) != (vis), actual: (nullptr) vs 16-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
54: [  FAILED  ] Camera/CameraTest.VisualAt/ogre2, where GetParam() = "ogre2" (224 ms)
```

chapulina, 2020-11-05:
The test is flaky on Citadel:

https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/27/testReport/(root)/Camera_CameraTest/VisualAt_ogre2/history/

But it hasn't failed in Dome yet:

https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering4-bionic-amd64/1/testReport/(root)/Camera_CameraTest/VisualAt_ogre2/history/

chapulina, 2020-11-06:
On #174, I'm adding more information to the test so we can understand better how it's failing.

chapulina, 2021-02-24:
> But it hasn't failed in Dome yet:

Now that there are more builds, it's possible to see that the test is flaky on Dome too. And on Edifice:

https://build.osrfoundation.org/job/ignition_rendering-ci-main-bionic-amd64/lastStableBuild/testReport/(root)/Camera_CameraTest/VisualAt_ogre2/history/

#255 added a bit more debug information

chapulina, 2021-04-20:
I suspect this may be related to the issue that #221 is trying to fix, which seems to be caused by a difference in devices.

See a summary of the lastest builds:

Jenkins node | Passes | Failures
-- | -- | --
drogon | 0 | 5
optimus | 5 | 1
r2d2 | 4 | 0

The test seems to pass consistently on r2d2 and fail on drogon.

Note that the only failure on optimus happened on April 1st, and all successes happened before March 17th. I believe that optimus recently got an upgrade that may have changed its settings in a way that causes the test to fail.

jacobperron, 2021-07-19:
Another instance of the test failure for Citadel (from July 16th): https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/46 

```
 58: [ RUN      ] Camera/CameraTest.VisualAt/ogre2
58: [1;32m[Msg] [0m[1;32mLoading plugin [[0m[1;32mignition-rendering-ogre2[0m[1;32m][0m[1;32m[0m
58: /var/lib/jenkins/workspace/ignition_rendering-ci-ign-rendering3-bionic-amd64/ign-rendering/test/integration/camera.cc:251: Failure
58: Expected: (nullptr) != (vis), actual: (nullptr) vs 16-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
58: X: 300
58: /var/lib/jenkins/workspace/ignition_rendering-ci-ign-rendering3-bionic-amd64/ign-rendering/test/integration/camera.cc:259: Failure
58: Expected equality of these values:
58:   nullptr
58:     Which is: (nullptr)
58:   vis
58:     Which is: 16-byte object <20-0D 64-50 23-56 00-00 00-CC 24-50 23-56 00-00>
58: Found [box] at X [400]
58: /var/lib/jenkins/workspace/ignition_rendering-ci-ign-rendering3-bionic-amd64/ign-rendering/test/integration/camera.cc:265: Failure
58: Expected: (nullptr) != (vis), actual: (nullptr) vs 16-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
58: X: 600
58: /var/lib/jenkins/workspace/ignition_rendering-ci-ign-rendering3-bionic-amd64/ign-rendering/test/integration/camera.cc:265: Failure
58: Expected: (nullptr) != (vis), actual: (nullptr) vs 16-byte object <00-00 00-00 00-00 00-00 00-00 00-00 00-00 00-00>
58: X: 700
58: [  FAILED  ] Camera/CameraTest.VisualAt/ogre2, where GetParam() = "ogre2" (387 ms)
```

jacobperron, 2021-09-13:
Another instance: https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-bionic-amd64/50/testReport/junit/(root)/Camera_CameraTest/VisualAt_ogre2/

Blast545, 2021-10-04:
I took some time to gather metrics about how much this affects the ign-rendering builds across the ignition buildfarms. Taking into account only the builds with a test report available.

`ignition_rendering-ci-ign-rendering3-bionic-amd64` has this error 14 out of 25 builds. 56.0% Flaky
`ignition_rendering-ci-ign-rendering4-bionic-amd64` has this error 13 out of 23 builds. 56.5% Flaky
`ignition_rendering-ci-ign-rendering5-bionic-amd64` has this error 10 out of 21 builds. 47.6% Flaky
`ignition_rendering-ci-main-bionic-amd64` has this error 9 out of 34 builds. 26.5% Flaky

I think it would be a good idea to take a closer look into this one, it's a test that fails frequently.

iche033, 2021-10-04:
I'm testing this in https://github.com/ignitionrobotics/ign-rendering/pull/450. Setting the device pixel ratio to 1.0f seems to have fixed the issue on the CI machines:
https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering5/ogre2/src/Ogre2Camera.cc#L250

Blast545, 2022-06-08:
This error is failing consistently on our buildfarm `ign-rendering3` job, see:
https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-focal-amd64/11/

Is it possible to backport this one? or should we disable the test on citadel and document it as a known issue? @chapulina

Blast545, 2022-06-09:
I took some time to check/backport this one, the link you attached here is showing me this:

`src/Utils_TEST.cc`
```
 EXPECT_EQ(0u, rayResult.objectId);
  VisualPtr root = scene->RootVisual();
```

Am I correct? To me it looks like the important changes are happening in the `ogre2/src/Ogre2SelectionBuffer.cc`, and the changes there were not trivial for me to backport.

iche033, 2022-11-28:
issue is back on CI machine: https://github.com/gazebosim/gz-rendering/pull/758#issuecomment-1329246890

Blast545, 2022-11-29:
FYI @Crola1702

PR #647 (2022-06-10): Force scaling factor to be 1.0

# 🦟 Bug fix

Fixes #170 (on Citadel)

Forces scale factor to be 1.0, partial backport of #446 

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests -> Fixes tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))
- [ ] While waitin [... cut]