gazebosim/gz-rendering #175: 👩‍🌾 GaussianNoise/ogre2 test regression on Mac OS

caguero, 2020-11-11:
This test tailed on 2020/11/10 and it was passing before. 

https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering4-homebrew-amd64/4/

Failure log:

```
7: Test command: /Users/jenkins/workspace/ignition_rendering-ci-ign-rendering4-homebrew-amd64/build/bin/UNIT_GaussianNoisePass_TEST "--gtest_output=xml:/Users/jenkins/workspace/ignition_rendering-ci-ign-rendering4-homebrew-amd64/build/test_results/UNIT_GaussianNoisePass_TEST.xml"
7: Test timeout computed to be: 240
7: [==========] Running 2 tests from 1 test case.
7: [----------] Global test environment set-up.
7: [----------] 2 tests from GaussianNoise/GaussianNoisePassTest
7: [ RUN      ] GaussianNoise/GaussianNoisePassTest.GaussianNoise/ogre2
7: [1;31m[Err] [Ogre2RenderEngine.cc:436] [0m[1;31mUnable to find Ogre Plugin[[0m[1;31m/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus[0m[1;31m]. Rendering will not be possible.[0m[1;31mMake sure you have installed OGRE properly.[0m
7: bias mean 0.1
7: /Users/jenkins/workspace/ignition_rendering-ci-ign-rendering4-homebrew-amd64/ign-rendering/src/GaussianNoisePass_TEST.cc:88: Failure
7: Expected: (std::fabs(noisePass->Bias())) <= (biasMean + biasStdDev*3), actual: 0.126035 vs 0.121
7: [  FAILED  ] GaussianNoise/GaussianNoisePassTest.GaussianNoise/ogre2, where GetParam() = "ogre2" (114 ms)
```

j-rivero, 2021-01-25:
A similar error appeared today on GitHub actions:
```
  [==========] Running 1 test from 1 test case.
  [----------] Global test environment set-up.
  [----------] 1 test from GaussianNoise/GaussianNoisePassTest
  [ RUN      ] GaussianNoise/GaussianNoisePassTest.GaussianNoise/ogre2
  bias mean 0.1
  /github/workspace/src/GaussianNoisePass_TEST.cc:89: Failure
  Expected: (std::fabs(noisePass->Bias())) >= (biasMean - biasStdDev*3), actual: 0.0763668 vs 0.079
  [  FAILED  ] GaussianNoise/GaussianNoisePassTest.GaussianNoise/ogre2, where GetParam() = "ogre2" (121 ms)
  [----------] 1 test from GaussianNoise/GaussianNoisePassTest (121 ms total)
```

I remove the macos label to indicate that the problem affects all platforms.

jacobperron, 2021-02-08:
Seems like a different test failure, but still related to Gaussian noise: https://github.com/ignitionrobotics/ign-rendering/runs/1832710251?check_suite_focus=true

Let me know if you think this deserves a different ticket.

chapulina, 2021-02-22:
Here's the failure output from the GitHub actions run above :point_up: 

```
   [ RUN      ] GaussianNoise/RenderPassTest.DepthGaussianNoise/ogre2
  /github/workspace/test/integration/render_pass.cc:342: Failure
  The difference between mz and midLeftZ is 0.040650427341461182, which exceeds noiseTol, where
  mz evaluates to 0.043290950357913971,
  midLeftZ evaluates to 0.0026405230164527893, and
  noiseTol evaluates to 0.039999999105930328.
  [  FAILED  ] GaussianNoise/RenderPassTest.DepthGaussianNoise/ogre2, where GetParam() = "ogre2" (256 ms)
```

chapulina, 2021-02-23:
Original test failure: `GaussianNoisePassTest.GaussianNoise/ogre2` history on macOS for [Citadel](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/22/testReport/(root)/GaussianNoise_GaussianNoisePassTest/GaussianNoise_ogre2/history/) and [Dome](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering4-homebrew-amd64/22/testReport/(root)/GaussianNoise_GaussianNoisePassTest/GaussianNoise_ogre2/history/): No failures in visible history

chapulina, 2021-04-22:
Still no failures in the visible history:

* [ign-rendering3 in Jenkins](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/lastCompletedBuild/testReport/(root)/GaussianNoise_GaussianNoisePassTest/GaussianNoise_ogre2/history/)
* [ign-rendering4 in Jenkins](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering4-homebrew-amd64/lastCompletedBuild/testReport/(root)/GaussianNoise_GaussianNoisePassTest/GaussianNoise_ogre2/history/)
* [ign-rendering5 in Jenkins](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering5-homebrew-amd64/lastCompletedBuild/testReport/(root)/GaussianNoise_GaussianNoisePassTest/GaussianNoise_ogre2/history/)
* [ign-rendering3 in actions](https://github.com/ignitionrobotics/ign-rendering/actions/workflows/ci.yml?query=branch%3Aign-rendering3)
* [ign-rendering4 in actions](https://github.com/ignitionrobotics/ign-rendering/actions/workflows/ci.yml?query=branch%3Aign-rendering4)
* [ign-rendering5 in actions](https://github.com/ignitionrobotics/ign-rendering/actions/workflows/ci.yml?query=branch%3Aign-rendering5)

I'm going to close this for now. Please reopen if it comes back.

---

https://github.com/osrf/buildfarmer/issues/181

closed 2021-04-22 by chapulina