gazebosim/gz-rendering #181: Support sdf <laser_retro> for lidar

doisyg, 2020-11-30:
Hello,
In the process of evaluating the portability of our sim from gazebo to ignition, I tried the ignition lidar plugin but noticed that it seems to ignore the sdf <laser_retro> value.
Am I missing something ? Or is it not implemented yet ? And if not where should I start to implement it ?

iche033, 2020-12-01:
the `<laser_retro>` flag is not implemented yet. The implementation depends on which render engine you plan to use in ignition (ogre 1.x or 2.x) since the OgreGpuRays and Ogre2GpuRays implementations are a bit different. 

The work is likely not very trivial and requires some knowledge of how ogre / ogre2 render target and shader system works.
Here is a very high level idea: We need to add an extra pass in the GpuRays render pipeline to swap the visuals to use a custom laser retro material when the sensor is updating. The custom laser retro material is a set of shaders that change the visual's color to the laser retro value. The texture containing the retro values created in this step is then passed to the GpuRays' shaders so it can pack the retro values alongside of the range data. It works in a similar way to how a thermal camera reads temperature values of a visual. So the OgreThermalCamera / Ogre2ThermalCamera class maybe a good example to look at.

doisyg, 2020-12-30:
Hello,
I am completely new to ogre2 and shaders but by copy pasting from Ogre2ThermalCamera I got a prototype which is successfully applying retro values set to a visual with `SetUserData` (similar that what exists for Ogre2ThermalCamera):
```
  float laser_retro = 310.0;
  std::string userDataKey = "laser_retro";
  visual->SetUserData(userDataKey, laser_retro);
```
However, I am interested in these values to be set from a sdf file. The parsing of `laser_retro` can be easily added to the sdformat9 package but what I have trouble to understand is how the link between the parsed sdf and the `rendering::v3::Visual` is done.
Is there a typical example that I can use as a reference ?

chapulina, 2020-12-30:
> how the link between the parsed sdf and the rendering::v3::Visual is done

Ignition Gazebo creates a `rendering::Visual` from an `sdf::Visual` on [SceneManager::CreateVisual](https://github.com/ignitionrobotics/ign-gazebo/blob/a830b443989877b1e83170390f483b6c02952b25/src/rendering/SceneManager.cc#L223-L224).

doisyg, 2020-12-30:
Thanks! That helps a lot.
If I succeed I will then do a PR on 3 packages to support `laser_retro`:

1.  sdformat
2. ignition-gazebo
3. ignition-rendering (targeting Ogre2)

doisyg, 2021-01-06:
I have it now working locally, so pushing the first PR: https://github.com/osrf/sdformat/pull/454
I will proceed with the next one once the first is accepted

doisyg, 2021-02-02:
Last PR: https://github.com/ignitionrobotics/ign-gazebo/pull/603

chapulina, 2021-03-02:
All merged! Thanks for the contribution, @doisyg !

closed 2021-03-02 by chapulina