gazebosim/gz-rendering #212: Debian buster armhf build failed for 3.3.0

scpeters, 2021-01-23:
The debian buster armhf build failed with a linking errors starting with "multiple definition of `typeinfo name for" in the recent release of 3.3.0:

* [image] https://build.osrfoundation.org/job/ign-rendering3-debbuilder/56/

~~~
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x0): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Scene>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Scene>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x2c): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x2c): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x8): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x8): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x58): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Light>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x58): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x10): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Light>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x10): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x84): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Sensor>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x84): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x18): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Sensor>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x18): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0xb4): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Visual>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0xb4): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x20): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Visual>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x20): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0xe4): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Geometry>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0xe4): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x28): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Geometry>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x28): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x114): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::SubMesh>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x114): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x30): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::SubMesh>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x30): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x144): multiple definition of `typeinfo name for ignition::rendering::v3::Map<ignition::rendering::v3::Material>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x144): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x38): multiple definition of `typeinfo for ignition::rendering::v3::Map<ignition::rendering::v3::Material>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x38): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.rodata+0x174): multiple definition of `typeinfo name for ignition::rendering::v3::CompositeStore<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x174): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/TransformController.cc.o:(.data.rel.ro+0x40): multiple definition of `typeinfo for ignition::rendering::v3::CompositeStore<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x40): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.rodata+0x0): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Scene>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x0): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.data.rel.ro+0x0): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Scene>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x0): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.rodata+0x2c): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x2c): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.data.rel.ro+0x8): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x8): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.rodata+0x58): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Light>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x58): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.data.rel.ro+0x10): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Light>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x10): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.rodata+0x84): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Sensor>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0x84): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.data.rel.ro+0x18): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Sensor>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x18): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.rodata+0xb4): multiple definition of `typeinfo name for ignition::rendering::v3::Store<ignition::rendering::v3::Visual>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.rodata+0xb4): first defined here
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseRenderEngine.cc.o:(.data.rel.ro+0x20): multiple definition of `typeinfo for ignition::rendering::v3::Store<ignition::rendering::v3::Visual>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x20): first defined here
[... cut ...]
/usr/bin/ld: CMakeFiles/ignition-rendering3.dir/base/BaseScene.cc.o:(.data.rel.ro+0x40): multiple definition of `typeinfo for ignition::rendering::v3::CompositeStore<ignition::rendering::v3::Node>'; CMakeFiles/ignition-rendering3.dir/OrbitViewController.cc.o:(.data.rel.ro+0x40): first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [src/CMakeFiles/ignition-rendering3.dir/build.make:350: lib/libignition-rendering3.so.3.3.0] Error 1
make[3]: Leaving directory '/var/lib/jenkins/workspace/ign-rendering3-debbuilder/build/ignition-rendering-3.3.0/obj-arm-linux-gnueabihf'
make[2]: *** [CMakeFiles/Makefile2:1940: src/CMakeFiles/ignition-rendering3.dir/all] Error 2
~~~

scpeters, 2021-01-23:
looks like similar failures for 4.2.0 with armhf on Ubuntu focal and debian buster as well:

* focal armhf [image] https://build.osrfoundation.org/job/ign-rendering4-debbuilder/180/
* buster armhf [image] https://build.osrfoundation.org/job/ign-rendering4-debbuilder/182/

chapulina, 2021-01-25:
I don't think we have any armhf builds of `ign-rendering3`, right?

* http://packages.osrfoundation.org/gazebo/debian-stable/pool/main/i/ignition-rendering3/
* http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/i/ignition-rendering3/

scpeters, 2021-01-27:
> I don't think we have any armhf builds of `ign-rendering3`, right?
> 
> * http://packages.osrfoundation.org/gazebo/debian-stable/pool/main/i/ignition-rendering3/
> * http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/i/ignition-rendering3/

I don't see any successful armhf builds there. According to https://github.com/ignition-tooling/release-tools/issues/297, I think Buster is the only distro we intend to support for armhf for rendering4. I'll make pull requests to disable armhf for Ubuntu in the release repositories, since those are unsupported in that issue. Do we have a policy on platform support for Citadel? I'm wondering if we would try to support Buster/armhf there if we ever got it working in Dome.

scpeters, 2021-01-27:
disabling armhf builds on Ubuntu:

* https://github.com/ignition-release/ign-rendering4-release/pull/4
* https://github.com/ignition-release/ign-rendering3-release/pull/2

scpeters, 2021-12-20:
this is now working if a native arm machine is used for the build

https://github.com/ignition-release/ign-rendering3-release/pull/4

closed 2021-12-20 by scpeters