gazebosim/gz-rendering #223: Upgrading for OGRE 2.2

mjcarroll, 2021-02-02:
In order to take advantage of [EGL Support in OGRE2](https://github.com/OGRECave/ogre-next/pull/177), we will need to update ignition-rendering to use the OGRE 2.2 API.

Ogre 2.2 API Migration reference: https://ogrecave.github.io/ogre-next/api/2.2/_ogre22_changes.html

TODO List:
- [ ] Update OGRE2 packaging for OGRE 2.2:
 * Attempt for colcon ws: https://github.com/ignition-forks/ogre-next/pull/1 
- [ ] Update APIS for OGRE 2.2  
  - [ ] Texture -> TextureGpu
  - [ ] RenderTarget -> TextureGpu
  - [ ] HardwarePixelBuffer -> StagingTextures
  - [ ] PixelFormat -> PixelFormatGpu

chapulina, 2021-02-03:
It would be nice to enable the terrain component when packaging 2.2 so we can address https://github.com/ignitionrobotics/ign-rendering/issues/187.

ahcorde, 2021-02-24:
- ign-cmake2 needs to find OGRE 2-2 https://github.com/ignitionrobotics/ign-cmake/pull/157

chapulina, 2021-02-27:
Here's Ogre;s 2.2.5 release announcement introducing EGL support:

https://www.ogre3d.org/2021/02/06/ogre-2-2-5-cerberus-released-and-egl-headless-support

ahcorde, 2021-06-10:
Related with this PR https://github.com/ignitionrobotics/ign-rendering/pull/272

darksylinc, 2021-06-28:
I started merging changes from the main branch into 2.2 while I still have these changes fresh on my mind. It was almost over but it needed more testing before I uploaded them.

I've got two questions (for now):

1. Do you have anything specific I can look at for Ogre 2.2 branch that doesn't work correctly? I was expecting a mess and it was surprisingly working quite well :+1: .
2. I noticed ign-rendering [creates its RenderTargets](https://github.com/ignitionrobotics/ign-rendering/blob/fb0fb6ea1e94bb895f44c15dc049a58bef3dd9b0/ogre2/src/Ogre2RenderTarget.cc#L751) as `SaveToSystemRam` e.g.
```
  this->ogreTexture = textureMgr->createOrRetrieveTexture(
    this->name,
    Ogre::GpuPageOutStrategy::SaveToSystemRam,
    Ogre::TextureFlags::RenderToTexture,
    Ogre::TextureTypes::Type2D);
````
is there any rationale behind this? SaveToSystemRam was conceived mostly for:

  1. In the future Ogre will automatically unload regular textures that are not in use (i.e. when memory limits are hit). Not yet implemented. This flag basically tells how fast textures are unloaded and loaded again. This is specially important if the texture was loaded from memory instead of disk
  2. I can't remember if the feature was tested against RenderTargets. RenderTargets have a similar issue as loading from memory instead of disk: when made Resident again, there is no disk. So either the initial contents are restored from previously-saved RAM or by rendering everything again. Most RenderTargets would commonly use `GpuPageOutStrategy::Discard` while only a few would use SaveToSystemRam assuming you want to unload them at some point and don't want to or can't re-render its contents. For RenderTargets you never plan to unload you should prefer Discard.

mjcarroll, 2021-06-28:
> Do you have anything specific I can look at for Ogre 2.2 branch that doesn't work correctly? I was expecting a mess and it was surprisingly working quite well +1 .

I don't believe so. @ahcorde and @iche033 got everything working great.  The only thing that remains is a Windows packaging detail so that CI will pass there.

> is there any rationale behind this?

Not that I'm aware of.

iche033, 2021-06-28:
The last thing I was not sure of are the spam of console msgs, which were just fixed :)

I also don't think there is a specific reason to use `SaveToSystemRam`. We can change to `Discard`

srmainwaring, 2021-07-06:
I saw the note from @chapulina that you may have to drop macOS support in ignition-fortress
([Ogre 2 support on macOS in Ignition Fortress](https://community.gazebosim.org/t/ogre-2-support-on-macos-in-ignition-fortress/912))

I'd like to see continued support for macOS and would like to get involved in helping that happen.

Thanks to help from @darksylinc I'm pretty close to completing a patch to the `ogre-next/v2-2-egl` branch that has fallback support for OpenGL on macOS: [WIP: add GL3Plus support for macOS ](https://github.com/OGRECave/ogre-next/pull/209). It might be slow to load because of the shader compiling but then should run fine for PBS materials.

The next step would be to get the ignition rendering and gui components working for macOS - and I'd be happy to look further into that if the team can provide guidance.

chapulina, 2021-07-06:
Hi @srmainwaring , thanks for the help with macOS! We're happy to continue providing support if you and others manage to get the upfront work out of the way. We should be able to provide guidance between now and the Fortress release, but it will be best effort on our side. Hopefully it will be enough to get it working in time!

ahcorde, 2021-10-01:
The Ogre 2.2 PR is merged, This feature is available in Fortress. I'm going to close this issue and consider this as done!

closed 2021-10-01 by ahcorde