gazebosim/gz-rendering #226: OgreLightVisual.cc error C2664 cannot convert from 'const char [19]' to 'const Ogre::MaterialPtr &'

sloretz, 2021-02-03:
Ignition rendering nightly is failing to build on windows which appears to be caused by #202 (@ahcorde FYI)

https://build.osrfoundation.org/job/ign_rendering-ci-win/111/console

```
[453.891s] C:\Jenkins\workspace\ign_rendering-ci-win\ws\ign-rendering\ogre\src\OgreLightVisual.cc(78,58): error C2664: 'void Ogre::SimpleRenderable::setMaterial(const Ogre::MaterialPtr &)': cannot convert argument 1 from 'const char [19]' to 'const Ogre::MaterialPtr &' [C:\Jenkins\workspace\ign_rendering-ci-win\ws\build\ignition-rendering5\ogre\src\ignition-rendering5-ogre.vcxproj]
[453.891s] C:\Jenkins\workspace\ign_rendering-ci-win\ws\ign-rendering\ogre\src\OgreLightVisual.cc(78,58): message : Reason: cannot convert from 'const char [19]' to 'const Ogre::MaterialPtr' [C:\Jenkins\workspace\ign_rendering-ci-win\ws\build\ignition-rendering5\ogre\src\ignition-rendering5-ogre.vcxproj]
[453.891s] C:\Jenkins\workspace\ign_rendering-ci-win\ws\ign-rendering\ogre\src\OgreLightVisual.cc(78,58): message : Constructor for class 'Ogre::SharedPtr<Ogre::Material>' is declared 'explicit' [C:\Jenkins\workspace\ign_rendering-ci-win\ws\build\ignition-rendering5\ogre\src\ignition-rendering5-ogre.vcxproj]
[453.891s] C:\vcpkg\installed\x64-windows\include\OGRE\OgreSimpleRenderable.h(76,22): message : see declaration of 'Ogre::SimpleRenderable::setMaterial' (compiling source file C:\Jenkins\workspace\ign_rendering-ci-win\ws\ign-rendering\ogre\src\OgreLightVisual.cc) [C:\Jenkins\workspace\ign_rendering-ci-win\ws\build\ignition-rendering5\ogre\src\ignition-rendering5-ogre.vcxproj]
[453.891s] C:\Jenkins\workspace\ign_rendering-ci-win\ws\ign-rendering\ogre\src\OgreLightVisual.cc(114,48): error C2664: 'void Ogre::SimpleRenderable::setMaterial(const Ogre::MaterialPtr &)': cannot convert argument 1 from 'std::string' to 'const Ogre::MaterialPtr &' [C:\Jenkins\workspace\ign_rendering-ci-win\ws\build\ignition-rendering5\ogre\src\ignition-rendering5-ogre.vcxproj]
```

sloretz, 2021-02-03:
Opened PR to revert #202: https://github.com/ignitionrobotics/ign-rendering/pull/227

ahcorde, 2021-02-04:
Opened a fix PR https://github.com/ignitionrobotics/ign-rendering/pull/228

closed 2021-02-04 by chapulina