gazebosim/gz-rendering #232: Add support for configuring intensity of lights

ahcorde, 2021-02-04:
Add support for configuring intensity of lights

ahcorde, 2021-02-04:
- [ ] sdformat  https://github.com/osrf/sdformat/pull/484
 - [ ] ign-msgs https://github.com/ignitionrobotics/ign-msgs/pull/131
 - [ ] ign-rendering https://github.com/ignitionrobotics/ign-rendering/pull/233
 - [ ] ign-gazebo https://github.com/ignitionrobotics/ign-gazebo/pull/612

ahcorde, 2021-02-04:
Diferrent point light intensities:

 - Ogre1

[image]

 - Ogre2

[image]

iche033, 2021-02-04:
can you add `SetIntensity(1.0)` to the [Reset](https://github.com/ignitionrobotics/ign-rendering/blob/272885cb7b08c67b84a9ffbb70cb577905f8b215/include/ignition/rendering/base/BaseLight.hh#L142) function?

ahcorde, 2021-02-18:
All PR are merged. Closing.

closed 2021-02-18 by ahcorde