gazebosim/gz-rendering #237: ign-rendering4 Windows build failing due to internal compiler error

jacobperron, 2021-02-08:
## Environment
* OS Version: Windows
* Source: 38d1ec0e39c32f1744eb3ce93772f5bfa2668faa

## Description

See https://build.osrfoundation.org/job/ign_rendering-ign-4-win/23

## Output

```
D:\Jenkins\workspace\ign_rendering-ign-4-win\ws\ign-rendering\ogre\src\OgreDynamicLines.cc(198,1): fatal error C1001: Internal compiler error. [D:\Jenkins\workspace\ign_rendering-ign-4-win\ws\build\ignition-rendering4\ogre\src\ignition-rendering4-ogre.vcxproj]
```

There's also a CMake warning higher up in the log; possibly related:

```
CMake Warning (dev) at D:/Jenkins/workspace/ign_rendering-ign-4-win/ws/install/ignition-cmake2/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:378 (message):
   -- THIS MESSAGE IS INTENDED FOR IGNITION-RENDERING AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for IgnOGRE.  It was not provided by
  the find-module for the package, nor was it explicitly passed into the call
  to ign_find_package(~).  This is most likely an error in this project's use
  of ign-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:64 (ign_find_package)
```

peci1, 2021-02-08:
I also wondered how is IgnOGRE supposed to find OGRE, which selectively does not generate any pkg-config info on windows (https://github.com/OGRECave/ogre/blob/56833c49119da275669cc1c168fa53c0e5ed1d65/CMake/ConfigureBuild.cmake#L159).

I had to create the PkgConfig files myself with OGRE installed via vcpkg. Then it was found.

I build ign-rendering on Windows yesterday (VS 2019 + NMake Makefiles generator) and did not face this compiler issue.

chapulina, 2021-02-09:
I removed the `win-locutus` node from the Jenkins pool and the latest build passed. I believe the problem on our CI is due to some configuration problem on that node.

jacobperron, 2021-02-09:
@chapulina I think you are right. There were other jobs failing on `win-locutus` as well. I'll close this ticket and keep an eye out for similar failures.

closed 2021-02-09 by jacobperron