gazebosim/gz-rendering #253: New thermal camera integration test failure for Edifice

adlarkin, 2021-02-17:
After forward porting some changes from `ign-rendering4` to `main`, a new integration test failure appeared for the thermal camera. More information can be found in #246.

## Environment
* OS Version: MacOS
* Source build (commit https://github.com/ignitionrobotics/ign-rendering/commit/52c4ab0d5a57a793d90171cbc54f324e8aac37d4, `main` branch at the time of this writing)

## Description
* Expected behavior: `test/integration/thermal_camera` should pass on MacOS.
* Actual behavior: `test/integration/thermal_camera` failed on MacOS.
* `test/integration/thermal_camera` does not fail for MacOS on the `ign-rendering4` branch: https://github.com/ignitionrobotics/ign-rendering/pull/246#issuecomment-778477528

## Steps to reproduce

1. Clone the repo: `git clone https://github.com/ignitionrobotics/ign-rendering.git`
2. Complete the build so that tests are available:
```
cd ign-rendering
mkdir build
cd build
cmake ..
make
```
3. Run the thermal camera integration test:
```
# the following steps assume you are already in the ign-rendering/build directory
cd bin
./INTEGRATION_thermal_camera
```

## Output

Here is the original failure from CI in #246:
```
/Users/jenkins/workspace/ignition_rendering-ci-pr_any-homebrew-amd64/ign-rendering/test/integration/thermal_camera.cc:233
The difference between boxTemp and temp is 14, which exceeds boxTempRange, where
boxTemp evaluates to 310,
temp evaluates to 296, and
boxTempRange evaluates to 3.
```

adlarkin, 2021-02-17:
It seems like there are related test failures in https://github.com/ignitionrobotics/ign-gazebo/pull/622. I'm guessing that the thermal camera behavior is different in Edifice.

jacobperron, 2021-07-19:
Another instance of this test failure: https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering5-homebrew-amd64/17/

I noticed many instances of the following logged error (I'm not sure if it's relevant):

```
[Err] [Ogre2RenderEngine.cc:467] Unable to find Ogre Plugin[/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
```

chapulina, 2021-08-25:
> I'm guessing that the thermal camera behavior is different in Edifice.

The test has been flaky on all versions, so I don't think the issue is version-specific. Here are the histories:

* [ign-rendering3](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/lastSuccessfulBuild/testReport/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxes_ogre2/history/) - has the most failures
* [ign-rendering4](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering4-homebrew-amd64/lastSuccessfulBuild/testReport/junit/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxesHeatSignature_ogre2/history/) - surprisingly no failures
* [ign-rendering5](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering5-homebrew-amd64/lastSuccessfulBuild/testReport/junit/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxesHeatSignature_ogre2/history/) - few failures
* [main](https://build.osrfoundation.org/job/ignition_rendering-ci-main-homebrew-amd64/lastSuccessfulBuild/testReport/junit/(root)/ThermalCamera_ThermalCameraTest/ThermalCameraBoxesHeatSignature_ogre2/history/) - again, no failures

---
https://github.com/osrf/buildfarmer/issues/224