gazebosim/gz-rendering #260: Depth camera crash

peci1, 2021-02-26:
## Environment
* OS Version: Ubuntu 18.04.5
* Source or binary build? Both (4.4.0 works, 4.5.0 crashes; the crash starts with #251).
* Kernel: Linux cras-17 5.10.17-051017-generic #202102170631 SMP Wed Feb 17 11:37:41 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux (installed via "Ubuntu Mainline Kernel Installer" to get support for Renoir GPU)
* GPU info: AMD RX Vega 7 (Renoir)

```
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD RENOIR (DRM 3.40.0, 5.10.17-051017-generic, LLVM 11.0.1) (0x1636)
```
<details>

```
    Version: 20.3.4
    Accelerated: yes
    Video memory: 512MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2

OpenGL vendor string: AMD
OpenGL renderer string: AMD RENOIR (DRM 3.40.0, 5.10.17-051017-generic, LLVM 11.0.1)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.3.4 - kisak-mesa PPA
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.3.4 - kisak-mesa PPA
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
```

</details>

## Description
* Expected behavior: Depth cameras should not crash.
* Actual behavior:  PR #251 started causing me a crash on a notebook with AMD Vega RX 7 (Renoir) GPU. I found the exact commit by git bisect, and it points at merging #251. And that makes sense, because the crash is in depth camera GLSL code. Before, I was able to run the whole SubT simulator on this notebook, including robots with depth cameras.

## Steps to reproduce
1. Take a Lenovo T14s or similar notebook with RX Vega 7 GPU and Ubuntu 18.04.5.
2. Install mainline kernel 5.10, enable kisak-mesa PPA to get an updated mesa
3. Manually download firmware for the GPU in /lib/firmware/amdgpu from a newer linux-firmware package.
4. Boot into the new kernel
5. Install SubT simulator in a catkin workspace
6. Run e.g. `ign launch -v4 cloudsim_sim.ign  robotName1:=X1 robotConfig1:=EXPLORER_X1_SENSOR_CONFIG_2 ros:=true durationSec:=3600 worldName:=simple_cave_01`
7. As soon as the depth camera starts rendering, the sensor thread crashes.
8. Downgrade ign-rendering to 4.4.0 (or pre-#251), relaunch simulator.
9. No crash, depth cameras are rendering.

## Output

The cause of the crash is:

```
RenderSystems/GL3Plus/src/GLSL/OgreGLSLProgramManager.cpp:624:
void Ogre::GLSLProgramManager::extractUniformsFromProgram(...):
Assertion `size_t (arraySize) == newGLUniformReference.mConstantDef->arraySize && "GL doesn't agree with our array size!"' failed.
```

Whole stack trace is here:

```
[Dbg] [Sensors.cc:205] Rendering Thread initialized
[Dbg] [EntityComponentManager.cc:672] Using components of type [5681045647686850796] / [ign_gazebo_components.SlipComplianceCmd ].
[Dbg] [EntityComponentManager.cc:1069] Updated state thread iterators: 16 threads processing around 30 components each.
[... 851 lines cut ...]
20:26:47: Can't assign material scene::Material(64328) because this Material does not exist. Have you forgotten to define it in a .material script?
20:26:47: Can't assign material scene::Material(64327) because this Material does not exist. Have you forgotten to define it in a .material script?
20:26:47: Vertex Shader: 537002240VertexShader_vs
Fragment Shader: 537002240PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
20:26:47: Vertex Shader: 536936832VertexShader_vs
Fragment Shader: 536936832PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
20:26:47: Vertex Shader: 537067904VertexShader_vs
Fragment Shader: 537067904PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
20:26:47: Vertex Shader: 537133440VertexShader_vs
Fragment Shader: 537133440PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
20:26:47: Vertex Shader: 536969600VertexShader_vs
Fragment Shader: 536969600PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
```

</details>

peci1, 2021-02-26:
Pinging @iche033 as the author of #251. I know this might be a difficult to debug issue. Don't you have a quick clue what could be wrong here?

The same crash can also be observed by launching your sensor_particles.sdf world. If I launch it on 4.4.0, it doesn't crash (but there are white rectangles instead of particles, of course).

peci1, 2021-02-26:
I've narrowed down the crash to this pass:

https://github.com/ignitionrobotics/ign-rendering/blob/278f2747fa84699cf9f0e0d6772321e680f1a516/ogre2/src/Ogre2DepthCamera.cc#L705-L711

If I comment it out, the crash does not occur (but particles are rendered wrongly).

Also, maybe related: https://github.com/ignitionrobotics/ign-sensors/issues/67 .

iche033, 2021-03-02:
I think it could be issue happening on mesa driver / AMD card. I tried testing on two computers but they  both have nvidia cards so was not able to reproduce it.

Given that the other passes work fine, I made some minor changes so that the textures and this particular pass are setup the same way as the `particleDepthTexture` pass. Changes are in the [depth_pass_crash](https://github.com/ignitionrobotics/ign-rendering/commit/ef7d08dc6833c20bf6ea16a8f198ea5385fed18d). Can you see if that makes any difference for you?

peci1, 2021-03-02:
Thank you for trying to debug this issue, Ian. However, the suggested change did not help.

It can very well be a driver issue because my system setup is not really conventional (Ubuntu Bionic with manually installed kernel 5.10 and Mesa from some PPA that provides a newer version; all of that done to support the recent Renoir GPU). I've just run some OpenGL benchmarks and all of them ran without an issue. So I would still think that the system is in a more or less good shape. But I understand that OpenGL benchmarks do not usually try rendering depth cameras...

I can play around with some configuration myself, but I need some pointers at where to poke, because I don't understand OpenGL or OGRE that much...

peci1, 2021-03-04:
To rule out the influence of my nonstandard setup, I booted into Ubuntu 20.04.2 HWE, which has kernel 5.8 by default, and this should already have support for the Renoir GPUs. Even on this Ubuntu and with Ignition Dome, I got the exact same error.

peci1, 2021-03-04:
ogre2.log from Ubuntu 20.04:

<details>

```
00:24:44: Creating resource group General
00:24:44: Creating resource group Internal
00:24:44: Creating resource group Autodetect
00:24:44: SceneManagerFactory for type 'DefaultSceneManager' registered.
00:24:44: Registering ResourceManager for type Material
00:24:44: Registering ResourceManager for type Mesh
00:24:44: Registering ResourceManager for type Mesh2
00:24:44: Registering ResourceManager for type OldSkeleton
00:24:44: MovableObjectFactory for type 'ParticleSystem' registered.
00:24:44: ArchiveFactory for archive type FileSystem registered.
00:24:44: ArchiveFactory for archive type Zip registered.
00:24:44: ArchiveFactory for archive type EmbeddedZip registered.
00:24:44: DDS codec registering
00:24:44: FreeImage version: 3.18.0
00:24:44: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
00:24:44: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
00:24:44: OITD codec registering
00:24:44: Registering ResourceManager for type HighLevelGpuProgram
00:24:44: MovableObjectFactory for type 'Entity' registered.
00:24:44: MovableObjectFactory for type 'Item' registered.
00:24:44: MovableObjectFactory for type 'Light' registered.
00:24:44: MovableObjectFactory for type 'BillboardSet' registered.
00:24:44: MovableObjectFactory for type 'ManualObject2' registered.
00:24:44: MovableObjectFactory for type 'BillboardChain' registered.
00:24:44: MovableObjectFactory for type 'RibbonTrail' registered.
00:24:44: MovableObjectFactory for type 'WireAabb' registered.
00:24:44: *-*-* OGRE Initialising
00:24:44: *-*-* Version 2.1.0unstable ('B')
00:24:44: OverlayElementFactory for type Panel registered.
00:24:44: OverlayElementFactory for type BorderPanel registered.
00:24:44: OverlayElementFactory for type TextArea registered.
00:24:44: Registering ResourceManager for type Font
00:24:44: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so
00:24:44: Installing plugin: GL 3+ RenderSystem
00:24:44: OpenGL 3+ Rendering Subsystem created.
00:24:44: Plugin successfully installed
00:24:44: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/Plugin_ParticleFX.so
00:24:44: Installing plugin: ParticleFX
00:24:44: Particle Emitter Type 'Point' registered
00:24:44: Particle Emitter Type 'Box' registered
00:24:44: Particle Emitter Type 'Ellipsoid' registered
00:24:44: Particle Emitter Type 'Cylinder' registered
00:24:44: Particle Emitter Type 'Ring' registered
00:24:44: Particle Emitter Type 'HollowEllipsoid' registered
00:24:44: Particle Affector Type 'LinearForce' registered
00:24:44: Particle Affector Type 'ColourFader' registered
00:24:44: Particle Affector Type 'ColourFader2' registered
00:24:44: Particle Affector Type 'ColourImage' registered
00:24:44: Particle Affector Type 'ColourInterpolator' registered
00:24:44: Particle Affector Type 'Scaler' registered
00:24:44: Particle Affector Type 'Rotator' registered
00:24:44: Particle Affector Type 'DirectionRandomiser' registered
00:24:44: Particle Affector Type 'DeflectorPlane' registered
00:24:44: Plugin successfully installed
00:24:44: CPU Identifier & Features
00:24:44: -------------------------
00:24:44:  *   CPU ID: AuthenticAMD: AMD Ryzen 7 PRO 4750U with Radeon Graphics
00:24:44:  *   Logical cores: 16
00:24:44:  *      SSE: yes
00:24:44:  *     SSE2: yes
00:24:44:  *     SSE3: yes
00:24:44:  *      MMX: yes
00:24:44:  *   MMXEXT: yes
00:24:44:  *    3DNOW: no
00:24:44:  * 3DNOWEXT: no
[... 456 lines cut ...]
00:24:44: Texture: loading rescue_randy_roughness.png as rescue_randy_roughness.png
00:24:44: Texture: loading rescue_randy_metalness.png as rescue_randy_metalness.png
00:24:44: Texture: loading rescue_randy_albedo.png as rescue_randy_albedo.png
00:24:44: Texture: loading rescue_randy_normal.png as rescue_randy_normal.png
00:24:45: WARNING: normal map texture rescue_randy_normal.png is not BC5S compressed. This is encouraged for lower memory usage. If you don't want to see this message without compressing to BC5, set getDefaultTextureParameters()[TEXTURE_TYPE_NORMALS].pixelFormat to PF_R8G8_SNORM (or PF_BYTE_LA if RSC_TEXTURE_SIGNED_INT is not supported)
00:24:45: Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/caguero/models/smoke_generator/2/materials/textures/' of type 'FileSystem' to resource group 'General'
00:24:45: Texture: loading smoke.png as smoke.png
00:24:45: Vertex Shader: 537133056VertexShader_vs
Fragment Shader: 537133056PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
00:24:45: Vertex Shader: 537067648VertexShader_vs
Fragment Shader: 537067648PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
00:24:45: Vertex Shader: 537100416VertexShader_vs
Fragment Shader: 537100416PixelShader_ps
 GLSL validation result : 
active samplers with a different type refer to the same texture image unit
00:24:45: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
00:24:45: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
00:24:45: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
```

</details>

peci1, 2021-03-04:
For comparison, here's Ubuntu 18.04 with GeForce 1050:

<details>

```
00:54:10: Creating resource group General
00:54:10: Creating resource group Internal
00:54:10: Creating resource group Autodetect
00:54:10: SceneManagerFactory for type 'DefaultSceneManager' registered.
00:54:10: Registering ResourceManager for type Material
00:54:10: Registering ResourceManager for type Mesh
00:54:10: Registering ResourceManager for type Mesh2
00:54:10: Registering ResourceManager for type OldSkeleton
00:54:10: MovableObjectFactory for type 'ParticleSystem' registered.
00:54:10: ArchiveFactory for archive type FileSystem registered.
00:54:10: ArchiveFactory for archive type Zip registered.
00:54:10: ArchiveFactory for archive type EmbeddedZip registered.
00:54:10: DDS codec registering
00:54:10: FreeImage version: 3.17.0
00:54:10: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
00:54:10: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
00:54:10: OITD codec registering
00:54:10: Registering ResourceManager for type HighLevelGpuProgram
00:54:10: MovableObjectFactory for type 'Entity' registered.
00:54:10: MovableObjectFactory for type 'Item' registered.
00:54:10: MovableObjectFactory for type 'Light' registered.
00:54:10: MovableObjectFactory for type 'BillboardSet' registered.
00:54:10: MovableObjectFactory for type 'ManualObject2' registered.
00:54:10: MovableObjectFactory for type 'BillboardChain' registered.
00:54:10: MovableObjectFactory for type 'RibbonTrail' registered.
00:54:10: MovableObjectFactory for type 'WireAabb' registered.
00:54:10: *-*-* OGRE Initialising
00:54:10: *-*-* Version 2.1.0unstable ('B')
00:54:10: OverlayElementFactory for type Panel registered.
00:54:10: OverlayElementFactory for type BorderPanel registered.
00:54:10: OverlayElementFactory for type TextArea registered.
00:54:10: Registering ResourceManager for type Font
00:54:10: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so
00:54:10: Installing plugin: GL 3+ RenderSystem
00:54:10: OpenGL 3+ Rendering Subsystem created.
00:54:10: Plugin successfully installed
00:54:10: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/Plugin_ParticleFX.so
00:54:10: Installing plugin: ParticleFX
00:54:10: Particle Emitter Type 'Point' registered
00:54:10: Particle Emitter Type 'Box' registered
00:54:10: Particle Emitter Type 'Ellipsoid' registered
00:54:10: Particle Emitter Type 'Cylinder' registered
00:54:10: Particle Emitter Type 'Ring' registered
00:54:10: Particle Emitter Type 'HollowEllipsoid' registered
00:54:10: Particle Affector Type 'LinearForce' registered
00:54:10: Particle Affector Type 'ColourFader' registered
00:54:10: Particle Affector Type 'ColourFader2' registered
00:54:10: Particle Affector Type 'ColourImage' registered
00:54:10: Particle Affector Type 'ColourInterpolator' registered
00:54:10: Particle Affector Type 'Scaler' registered
00:54:10: Particle Affector Type 'Rotator' registered
00:54:10: Particle Affector Type 'DirectionRandomiser' registered
00:54:10: Particle Affector Type 'DeflectorPlane' registered
00:54:10: Plugin successfully installed
00:54:10: CPU Identifier & Features
00:54:10: -------------------------
00:54:10:  *   CPU ID: GenuineIntel: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
00:54:10:  *   Logical cores: 8
00:54:10:  *      SSE: yes
00:54:10:  *     SSE2: yes
00:54:10:  *     SSE3: yes
00:54:10:  *      MMX: yes
00:54:10:  *   MMXEXT: yes
00:54:10:  *    3DNOW: no
00:54:10:  * 3DNOWEXT: no
[... 594 lines cut ...]
00:54:10: Finished parsing scripts for resource group Internal
00:54:10: Creating resources for group Internal
00:54:10: All done
00:54:10: Can't assign material scene::Material(65506) because this Material does not exist. Have you forgotten to define it in a .material script?
00:54:10: Can't assign material scene::Material(65500) because this Material does not exist. Have you forgotten to define it in a .material script?
00:54:10: Can't assign material scene::Material(65500) because this Material does not exist. Have you forgotten to define it in a .material script?
00:54:10: Can't assign material scene::Material(65500) because this Material does not exist. Have you forgotten to define it in a .material script?
00:54:10: Can't assign material scene::Material(65500) because this Material does not exist. Have you forgotten to define it in a .material script?
00:54:10: Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/rescue randy/2/meshes' of type 'FileSystem' to resource group 'General' with recursive option
00:54:10: Initialising resource group General
00:54:10: Can't assign material scene::Material(65479) because this Material does not exist. Have you forgotten to define it in a .material script?
00:54:10: Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/rescue randy/2/materials/textures/' of type 'FileSystem' to resource group 'General'
[... cut ...]
00:54:12: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
00:54:12: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
00:54:12: Texture: RescueRandy_Thermal.png: Loading 1 faces(PF_R8G8B8,2048x2048x1) with 11 hardware generated mipmaps from Image. Internal format is PF_X8R8G8B8,2048x2048x1.
```

</details>

peci1, 2021-03-05:
I removed timestamps and have compared the ogre2.log from the working computer with nvidia GPU and the ubuntu 20.04 computer with AMD GPU.

<details>

```diff
--- "/home/peci1/Sta\305\276en\303\251/ogre.18.nvidia.log"	2021-03-05 01:06:43.380554984 +0100
+++ "/home/peci1/Sta\305\276en\303\251/ogre.20.amd.log"	2021-03-05 01:12:45.252972667 +0100
@@ -11,9 +11,9 @@
 ArchiveFactory for archive type Zip registered.
 ArchiveFactory for archive type EmbeddedZip registered.
 DDS codec registering
-FreeImage version: 3.17.0
+FreeImage version: 3.18.0
 This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
-Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
+Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
 OITD codec registering
 Registering ResourceManager for type HighLevelGpuProgram
 MovableObjectFactory for type 'Entity' registered.
@@ -54,8 +54,8 @@
 Plugin successfully installed
 CPU Identifier & Features
 -------------------------
- *   CPU ID: GenuineIntel: Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
- *   Logical cores: 8
+ *   CPU ID: AuthenticAMD: AMD Ryzen 7 PRO 4750U with Radeon Graphics
+ *   Logical cores: 16
  *      SSE: yes
  *     SSE2: yes
  *     SSE3: yes
@@ -72,16 +72,29 @@
 ******************************
 ng GLX Subsystem ***
 ********************
-GL3PlusRenderSystem::_createRenderWindow "OgreWindow(0)_0", 1x1 windowed  miscParams: FSAA=0 border=none contentScalingFactor=1.000000 currentGLContext=true externalGLControl=true gamma=true stereoMode=Frame Sequential 
-GLXWindow::create used FBConfigID = 33
-GL_VERSION = 3.3.0.0
-GL_VENDOR = NVIDIA Corporation
-GL_RENDERER = GeForce GTX 1050/PCIe/SSE2
+GL3PlusRenderSystem::_createRenderWindow "OgreWindow(0)_0", 1x1 windowed  miscParams: FSAA=0 border=none contentScalingFactor=1,000000 currentGLContext=true externalGLControl=true gamma=true stereoMode=Frame Sequential 
+GLXWindow::create used FBConfigID = 423
+GL_VERSION = 4.6.0.0
+GL_VENDOR = AMD
+GL_RENDERER = AMD RENOIR (DRM 3.40.0, 5.11.3-051103-generic, LLVM 11.0.1)
 GL_EXTENSIONS = 
+GL_AMD_conservative_depth
+GL_AMD_depth_clamp_separate
+GL_AMD_draw_buffers_blend
+GL_AMD_framebuffer_multisample_advanced
+GL_AMD_gpu_shader_int64
 GL_AMD_multi_draw_indirect
+GL_AMD_performance_monitor
+GL_AMD_pinned_memory
+GL_AMD_query_buffer_object
 GL_AMD_seamless_cubemap_per_texture
+GL_AMD_shader_stencil_export
+GL_AMD_shader_trinary_minmax
+GL_AMD_texture_texture4
 GL_AMD_vertex_shader_layer
 GL_AMD_vertex_shader_viewport_index
+GL_ANGLE_texture_compression_dxt3
+GL_ANGLE_texture_compression_dxt5
 GL_ARB_arrays_of_arrays
 GL_ARB_base_instance
 GL_ARB_bindless_texture
@@ -102,7 +115,6 @@
 GL_ARB_debug_output
 GL_ARB_depth_buffer_float
 GL_ARB_depth_clamp
-GL_ARB_depth_texture
[... 590 lines cut ...]
  * Number of world matrices: 0
  * Number of texture units: 16
@@ -677,6 +526,18 @@
 WARNING: normal map texture rescue_randy_normal.png is not BC5S compressed. This is encouraged for lower memory usage. If you don't want to see this message without compressing to BC5, set getDefaultTextureParameters()[TEXTURE_TYPE_NORMALS].pixelFormat to PF_R8G8_SNORM (or PF_BYTE_LA if RSC_TEXTURE_SIGNED_INT is not supported)
 Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/caguero/models/smoke_generator/2/materials/textures/' of type 'FileSystem' to resource group 'General'
 Texture: loading smoke.png as smoke.png
+Vertex Shader: 537133056VertexShader_vs
+Fragment Shader: 537133056PixelShader_ps
+ GLSL validation result :
+active samplers with a different type refer to the same texture image unit
+Vertex Shader: 537067648VertexShader_vs
+Fragment Shader: 537067648PixelShader_ps
+ GLSL validation result :
+active samplers with a different type refer to the same texture image unit
+Vertex Shader: 537100416VertexShader_vs
+Fragment Shader: 537100416PixelShader_ps
+ GLSL validation result :
+active samplers with a different type refer to the same texture image unit
+WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
 WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
 WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
-Texture: RescueRandy_Thermal.png: Loading 1 faces(PF_R8G8B8,2048x2048x1) with 11 hardware generated mipmaps from Image. Internal format is PF_X8R8G8B8,2048x2048x1.
```

</details>

peci1, 2021-03-05:
What caught my attention was

```diff
-GL_ARB_depth_texture
```

It seems that the selected OpenGL profile on AMD doesn't contain the depth_texture extension (although I don't know whether it is needed or not).

In glxinfo, I can see the extension, but only for the compat profile, not for core profile. So I tried launching with `MESA_GL_VERSION_OVERRIDE=4.6COMPAT`, which correctly selected the compat profile, OGRE has found the depth_texture extensions, but it still resulted in the very same exception.

iche033, 2021-03-05:
it weird that commenting that specific pass made things not crash because a few lines below, I'm doing the same depth pass with `particleDepthTexture` which has the same depth texture format and that worked. Just brainstorming some things to try:

* remove the `mVisibiliyMask` setting for the depthTexture pass, i.e this line:
```
 passScene->mVisibilityMask = IGN_VISIBILITY_ALL 
     & ~Ogre2ParticleEmitter::kParticleVisibilityFlags; 
```

* comment out the [colorTexture](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering4/ogre2/src/Ogre2DepthCamera.cc#L676) (and change `baseNodeDef->setNumTargetPass(5)` to baseNodeDef->`setNumTargetPass(4)`) pass and see if you still get the crash. The RGBD camera's color image won't work but just for testing.

* remove the `depthTexture`'s [clear ass](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering4/ogre2/src/Ogre2DepthCamera.cc#L705) (and change `depthTargetDef->setNumPasses(2)` to `depthTargetDef->setNumPasses(1)`)

peci1, 2021-03-05:
I've tried all of your suggestion. None of it helped. I also found #244 and tried it, but it also did not help.

I found a way to tell the GPU driver to write out which textures does it load. There are some differences between what is loaded when running sensors_demo.sdf and sensor_particles.sdf (I commented out all sensors except one depth camera):

```diff
--- /tmp/log_demo.txt	2021-03-05 13:15:36.312868546 +0100
+++ /tmp/log_particles.txt	2021-03-05 13:15:18.016786299 +0100
@@ -26,9 +26,20 @@
   Surf: size=4096, slice_size=4096, alignment=4096, swmode=22, epitch=31, pitch=32
   DCC: offset=4096, size=4096, alignment=4096, pitch_max=511, num_dcc_levels=1
 Texture:
-  Info: npix_x=1024, npix_y=1024, npix_z=1, blk_w=1, blk_h=1, array_size=10, last_level=10, bpe=4, nsamples=0, flags=0x0, r8g8b8a8_srgb
-  Surf: size=62914560, slice_size=6291456, alignment=65536, swmode=26, epitch=1535, pitch=1024
-  DCC: offset=62914560, size=262144, alignment=65536, pitch_max=1023, num_dcc_levels=2
+  Info: npix_x=2048, npix_y=2048, npix_z=1, blk_w=1, blk_h=1, array_size=10, last_level=11, bpe=1, nsamples=0, flags=0x0, r8_unorm
+  Surf: size=62914560, slice_size=6291456, alignment=65536, swmode=26, epitch=3071, pitch=2048
+  DCC: offset=62914560, size=262144, alignment=65536, pitch_max=2047, num_dcc_levels=2
+Texture:
+  Info: npix_x=2048, npix_y=2048, npix_z=1, blk_w=1, blk_h=1, array_size=2, last_level=11, bpe=4, nsamples=0, flags=0x0, r8g8b8a8_srgb
+  Surf: size=50331648, slice_size=25165824, alignment=65536, swmode=26, epitch=3071, pitch=2048
+  DCC: offset=50331648, size=196608, alignment=65536, pitch_max=2047, num_dcc_levels=3
+Texture:
+  Info: npix_x=2048, npix_y=2048, npix_z=1, blk_w=1, blk_h=1, array_size=2, last_level=11, bpe=2, nsamples=0, flags=0x0, r8g8_snorm
+  Surf: size=25165824, slice_size=12582912, alignment=65536, swmode=26, epitch=3071, pitch=2048
+  DCC: offset=25165824, size=131072, alignment=65536, pitch_max=2047, num_dcc_levels=2
+Texture:
+  Info: npix_x=256, npix_y=256, npix_z=1, blk_w=1, blk_h=1, array_size=40, last_level=8, bpe=4, nsamples=0, flags=0x0, r8g8b8a8_srgb
+  Surf: size=15728640, slice_size=393216, alignment=65536, swmode=26, epitch=383, pitch=256
 Texture:
   Info: npix_x=320, npix_y=240, npix_z=1, blk_w=1, blk_h=1, array_size=1, last_level=0, bpe=16, nsamples=0, flags=0x0, r32g32b32a32_float
   Surf: size=1310720, slice_size=1310720, alignment=65536, swmode=26, epitch=319, pitch=320
@@ -62,11 +73,3 @@
   Info: npix_x=160, npix_y=120, npix_z=1, blk_w=1, blk_h=1, array_size=1, last_level=0, bpe=4, nsamples=0, flags=0x820000, z32_float
   Surf: size=81920, slice_size=81920, alignment=4096, swmode=20, epitch=159, pitch=160
   HTile: offset=98304, size=32768, alignment=32768
-Texture:
-  Info: npix_x=320, npix_y=240, npix_z=1, blk_w=1, blk_h=1, array_size=1, last_level=0, bpe=4, nsamples=0, flags=0x60000, z32_float_s8x24_uint
-  Surf: size=524288, slice_size=393216, alignment=65536, swmode=24, epitch=383, pitch=384
-  HTile: offset=524288, size=262144, alignment=262144
-  Stencil: offset=393216, swmode=24, epitch=511
-Texture:
-  Info: npix_x=320, npix_y=240, npix_z=1, blk_w=1, blk_h=1, array_size=1, last_level=0, bpe=16, nsamples=0, flags=0x0, r32g32b32a32_float
-  Surf: size=1228800, slice_size=1228800, alignment=256, swmode=0, epitch=319, pitch=320
```

I understand that in case of the particle world, there are some more textures loaded, but the emitter texture is 256x256. That would explain one of the additional textures. And I don't see the 64x1 color texture loaded anywhere. And what are the three 2048x2048 textures that appear in the particle world?

Then there are the final two textures missing from particle world, but that is probably just because it throws the exception earlier than these textures are set up.

I tried also gpu_ray in the particle world, and that works.

peci1, 2021-03-05:
Big discovery! It's not the particle emitter that breaks stuff, it's the rescue randy that's in the particles world but not in sensors_demo (I haven't noticed, sorry for that; that would explain why the 2048x2048 textures were loaded in addition to sensors_demo world).

In particular, it's this definition of its texture:

```xml
<pbr>
  <metal>
    <albedo_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_albedo.png</albedo_map>
    <!--normal_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_normal.png</normal_map-->
    <!--metalness_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_metalness.png</metalness_map-->
    <!--roughness_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_roughness.png</roughness_map-->
  </metal>
</pbr>
```

If I uncomment any of the commented-out lines, I get a crash. 

Maybe related, if I look into ogre2.log from the crashed sessions, I can see

```
Vertex Shader: 537100416VertexShader_vs
Fragment Shader: 537100416PixelShader_ps
 GLSL validation result :
active samplers with a different type refer to the same texture image unit
```

Can't this be some kind of interference with the textures loaded for the rescue randy?

This is the randy without the three PBR textures on AMD GPU:
[image]

I also tried enabling e.g. the normal texture and resizing it to 512x512, but that did not help.

I tested all other SubT artifacts and all of them work except rescue randy (even rescue randy sitting works, as it doesn't use PBR).

darksylinc, 2021-06-21:
Where can I get cloudsim_sim.ign? I can't repro without that :(

I have similar HW so I should be hopefully able to repro the bug.

Usually this problem happens because the driver optimized out an uniform variable, e.g.

`uniform float myConst[4]; // it's not actually used`

or decides to shorten the length of the array

`uniform float myConst[4]; // the code actually uses range [0; 1] so driver decides to optimize it to myConst[2]`

The solution once the offending shader is identified is to either spuriously use the last value (e.g. use myConst[3] to force the array to be of size 4) or to remove it (i.e. the optimizer after all noticed the extra entries aren't used)

peci1, 2021-06-21:
cloudsim_sim.ign is part of the SubT simulator. You can install it following the instructions here: https://github.com/osrf/subt/wiki/Catkin%20System%20Setup . There are also docker images, but I'm not sure if they're the best to test GPU-related problems...

darksylinc, 2021-06-21:
Thanks! And ouch! That's a lot of setup work.

Were you able to repro the bug with a simple/modified SDF I can launch?

peci1, 2021-06-21:
Actually yes! I think this one causes the crash for me too, and does not require the SubT simulator:

sensor_particles.sdf:

<details>

```XML
<?xml version="1.0" ?>

<sdf version="1.6">
  <world name="particle_emitters">

    <physics name="1ms" type="ignored">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
    <plugin
      filename="ignition-gazebo-physics-system"
      name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin
      filename="ignition-gazebo-user-commands-system"
      name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin
      filename="ignition-gazebo-sensors-system"
      name="ignition::gazebo::systems::Sensors">
      <render_engine>ogre2</render_engine>
    </plugin>

    <plugin
      filename="ignition-gazebo-scene-broadcaster-system"
      name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>

    <gui fullscreen="0">

      <!-- 3D scene -->
      <plugin filename="GzScene3D" name="3D View">
        <ignition-gui>
          <title>3D View</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="string" key="state">docked</property>
        </ignition-gui>

        <engine>ogre2</engine>
        <scene>scene</scene>
        <ambient_light>0.4 0.4 0.4</ambient_light>
        <background_color>0.8 0.8 0.8</background_color>
        <camera_pose>-6 0 6 0 0.5 0</camera_pose>
      </plugin>

      <!-- World control -->
      <plugin filename="WorldControl" name="World control">
        <ignition-gui>
          <title>World control</title>
          <property type="bool" key="showTitleBar">false</property>
          <property type="bool" key="resizable">false</property>
          <property type="double" key="height">72</property>
          <property type="double" key="width">121</property>
          <property type="double" key="z">1</property>

          <property type="string" key="state">floating</property>
          <anchors target="3D View">
            <line own="left" target="left"/>
            <line own="bottom" target="bottom"/>
          </anchors>
        </ignition-gui>

        <play_pause>true</play_pause>
[... 360 lines cut ...]
            <clip>
              <near>0.1</near>
              <far>100</far>
            </clip>
          </camera>
          <always_on>1</always_on>
          <update_rate>30</update_rate>
          <visualize>true</visualize>
          <topic>thermal_camera</topic>
        </sensor>
      </link>
    </model>

    <include>
      <pose>1 0 0 0 0 1.570796</pose>
      <name>rescue_randy</name>
      <uri>https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Rescue Randy</uri>
    </include>

  </world>
</sdf>

```

</details>

darksylinc, 2021-06-21:
Mmmm... I could not repro it in this GPU:

```
Device: Radeon RX 560 Series (POLARIS11, DRM 3.40.0, 5.11.0-051100-generic, LLVM 10.0.1) (0x67ff)
Mesa Version: 20.1.3
```

I have a Vega iGPU on the laptop (Ryzen 5 2500U), I can't test it right now should be able to test it in another time.

I can't help noticing a couple things:

1. The LLVM version my Mesa driver is using is 10.0.1; yours is 11. This bug is sensitive to LLVM version that Mesa is using; so I can't ignore that.
2. GL_VENDOR = AMD, but normally Mesa reports X.Org. Are you on Wayland compositor? Are you using proprietary drivers? What's the output of glxinfo?

peci1, 2021-06-21:
AFAIK I use only the open-source drivers. Maybe it's the newest kernel?

Here's glxinfo:

<details>

```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_no_config_context, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
    GLX_ARB_context_flush_control, GLX_ARB_create_context, 
    GLX_ARB_create_context_no_error, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_get_proc_address, GLX_ARB_multisample, 
    GLX_EXT_buffer_age, GLX_EXT_create_context_es2_profile, 
    GLX_EXT_create_context_es_profile, GLX_EXT_fbconfig_packed_float, 
    GLX_EXT_framebuffer_sRGB, GLX_EXT_import_context, GLX_EXT_swap_control, 
    GLX_EXT_swap_control_tear, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_MESA_multithread_makecurrent, 
    GLX_MESA_query_renderer, GLX_MESA_swap_control, GLX_OML_swap_method, 
    GLX_OML_sync_control, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
GLX version: 1.4
GLX extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_no_error, 
    GLX_ARB_create_context_profile, GLX_ARB_create_context_robustness, 
    GLX_ARB_fbconfig_float, GLX_ARB_framebuffer_sRGB, 
    GLX_ARB_get_proc_address, GLX_ARB_multisample, GLX_EXT_buffer_age, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_swap_control, GLX_EXT_swap_control_tear, 
    GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, GLX_EXT_visual_rating, 
    GLX_INTEL_swap_event, GLX_MESA_copy_sub_buffer, GLX_MESA_query_renderer, 
    GLX_MESA_swap_control, GLX_OML_swap_method, GLX_OML_sync_control, 
    GLX_SGIS_multisample, GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, 
    GLX_SGIX_visual_select_group, GLX_SGI_make_current_read, 
    GLX_SGI_swap_control, GLX_SGI_video_sync
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD RENOIR (DRM 3.40.0, 5.12.10-051210-generic, LLVM 12.0.0) (0x1636)
    Version: 21.1.3
    Accelerated: yes
    Video memory: 2048MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 444 MB, largest block: 444 MB
[... 1922 lines cut ...]
0x4a2 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0  0  0  0  0  0  0  0 0 None
0x4a3 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 16  0  0  0  0  0  0 0 None
0x4a4 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 16  0  0  0  0  0  0 0 None
0x4a5 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 16  0  0  0  0  0  0 0 None
0x4a6 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x4a7 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x4a8 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  0  0  0  0  0  0 0 None
0x4a9 32 tc  0  32  0 r  . .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x4aa 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x4ab 32 tc  0  32  0 r  y .   8  8  8  8 .  .  0 24  8  0  0  0  0  0 0 None
0x4ac 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x4ad 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x4ae 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0  0  0  0  0  0  0  0 0 None
0x4af 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x4b0 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x4b1 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 16  0  0  0  0  0  0 0 None
0x4b2 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x4b3 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x4b4 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  0  0  0  0  0  0 0 None
0x4b5 32 tc  0  32  0 r  . .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x4b6 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
0x4b7 32 tc  0  32  0 r  y .   8  8  8  8 .  s  0 24  8  0  0  0  0  0 0 None
```

</details>

peci1, 2021-06-21:
Did you actually unpause the world? Sometimes it crashes only after unpause...

darksylinc, 2021-06-21:
Mmm your mesa version is much newer than mine. I should look into using that exact version.

Thanks!

peci1, 2021-11-03:
I don't know why, but this got fixed by #388 . Maybe my AMD card was not transferring the time variable to the shaders?

peci1, 2021-11-03:
I wanted to say that https://github.com/ignitionrobotics/ign-rendering/pull/388 resolved this issue for me, but I was too quick. Another run crashed again with the same error (now on Linux 5.15.0, Mesa 21.2.5).

peci1, 2021-11-03:
I got OGRE2 ignition fork built from source and used in runtime (both main and render plugins). However, I can't see any of the `printf` outputs anywhere (neither in console with `-v4`, nor in `~/.ignition/rendering/ogre2.log`). But I'm pretty sure the modified code gets executed.

Nevertheless, I got more info.

First of all, this assert does not go through:

https://github.com/ignition-forks/ogre-2.1-release/blob/063b485f91cfa9e9b7d03a63ffd7bfc3a4354fcd/OgreMain/src/OgreHlmsManager.cpp#L254-L257

If I comment it out, I get further. This is what I found in `ogre2.log` afterwards:

```
17:48:12: Parsing script depth_camera.material
17:48:12: OGRE EXCEPTION(5:ItemIdentityException): Parameter called time does not exist. Known names are: backgroundColor backgroundColor[0] colorTexture colorTexture[0] depthTexture depthTexture[0] far far[0] max max[0] min min[0] near near[0] particleDepthTexture particleDepthTexture[0] particleScatterRatio particleScatterRatio[0] particleStddev particleStddev[0] particleTexture particleTexture[0] projectionParams projectionParams[0] rnd rnd[0]  in GpuProgramParameters::_findNamedConstantDefinition at /media/data/ign/ogre/OgreMain/src/OgreGpuProgramParams.cpp (line 2218)
17:48:12: Compiler error: invalid parameters in depth_camera.material(33): setting of constant failed
17:48:12: Parsing script thermal.material
17:48:12: Parsing script gpu_rays.material
17:48:12: OGRE EXCEPTION(5:ItemIdentityException): Parameter called time does not exist. Known names are: colorTexture colorTexture[0] depthTexture depthTexture[0] far far[0] max max[0] min min[0] near near[0] particleDepthTexture particleDepthTexture[0] particleScatterRatio particleScatterRatio[0] particleStddev particleStddev[0] particleTexture particleTexture[0] projectionParams projectionParams[0] rnd rnd[0]  in GpuProgramParameters::_findNamedConstantDefinition at /media/data/ign/ogre/OgreMain/src/OgreGpuProgramParams.cpp (line 2218)
17:48:12: Compiler error: invalid parameters in gpu_rays.material(33): setting of constant failed
```

So my intuition that `time` shader variable could be the cause of the problem is probably right. But I don't know how to fix it.

iche033, 2021-11-03:
The `time` var are no longer used. Here's a quick PR to that removes them from the material script:
https://github.com/ignitionrobotics/ign-rendering/pull/485. See if that helps

peci1, 2021-11-04:
So the `time` problem was just a smoke wall. I removed it and now I'm back where I was at the very beginning - memory corruption.

It seems the Ogre assert is detecting an actual memory corruption, so disabling it lets me go a bit further but all reported errors afterwards may be just wrong.

The corruption start about here:

https://github.com/ignitionrobotics/ign-rendering/blob/46dd20ae11fdf2dd2ce25d7ee93299545ef8b224/ogre2/src/Ogre2Material.cc#L520-L523

Once inside `createDatablock()`, `getBlendBlock()` gets called with an argument that has at least two weird values:

```
$4 = (const Ogre::HlmsBlendblock &) @0x7fffb9303810: {<Ogre::BasicBlock> = {mRsData = 0x0, mRefCount = 0, mId = 0, mBlockType = 0 '\000', mAllowGlobalDefaults = 1 '\001'}, mAlphaToCoverageEnabled = false, 
  mBlendChannelMask = 1 '\001', mIsTransparent = true, mSeparateBlend = 67, mSourceBlendFactor = Ogre::SBF_SOURCE_COLOUR, mDestBlendFactor = Ogre::SBF_ONE, mSourceBlendFactorAlpha = Ogre::SBF_ONE, 
  mDestBlendFactorAlpha = Ogre::SBF_DEST_COLOUR, mBlendOperation = Ogre::SBO_MIN, mBlendOperationAlpha = 2999584171}
```

Notice `mBlockType = 0`, should be 1 for a blend block (this is what triggers the assert). Also notice `mBlendOperationAlpha = 2999584171` which is wrong because the alpha should be an enum with 4 values or so. I'm not sure whether `mRefCount = 0` is wrong here or not...

If I disable the assert and let the code run through it, this shows on stack:

```
0x00007fffb2a1a278 in Ogre::Hlms::createDatablock (this=0x7fffb45c80f0, name=..., refName=<error reading variable: Cannot access memory at address 0x64f0b678>, macroblockRef=..., blendblockRef=..., 
    paramVec=std::vector of length -15012014088548807, capacity -15012014088548807 = {...}, visibleToManager=128, filename="", 
    resourceGroup=<error reading variable: Cannot access memory at address 0x3ff0000000000008>) at /media/data/ign/ogre/OgreMain/src/OgreHlms.cpp:1572
```

The `paramVec` is obviously corrupted, thus it's no surprise searching for a parameter in it doesn't work and segfaults.

So, the question is where is this corruption coming from.

The first value where I noticed the corruption is the `blendblockRef` parameter of `Ogre::Hlms::createDatablock()` called from 

https://github.com/ignitionrobotics/ign-rendering/blob/46dd20ae11fdf2dd2ce25d7ee93299545ef8b224/ogre2/src/Ogre2Material.cc#L520-L523

As can be seen, the blend block object is freshly created and directly passed to `createDatablock()`, so I think the corruption has to happen even earlier or somewhere around here.

Just to add context, the assert fails when creating this material, which also happens to be the very first one created in `ignition::rendering::v4::BaseScene::CreateMaterials()`:

ignition::rendering::v4::Ogre2Scene::CreateMaterialImpl (this=0x7fffb47ff970, _id=65534, _name="Default/TransRed") at /media/data/ign/ign-rendering/ogre2/src/Ogre2Scene.cc:420

Complete stack trace up to the assert:

<details>

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff7571921 in __GI_abort () at abort.c:79
#2  0x00007ffff756148a in __assert_fail_base (fmt=0x7ffff76e8750 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7fffb23d9898 "baseParams.mBlockType == BLOCK_BLEND && \"baseParams.mBlockType should always be BLOCK_BLEND! You can ignore this assert,\" \" but it usually indicates memory corruption (or you created the block without"..., file=file@entry=0x7fffb23d9570 "/media/data/ign/ogre/OgreMain/src/OgreHlmsManager.cpp", line=line@entry=258, 
    function=function@entry=0x7fffb23d9fe0 <Ogre::HlmsManager::getBlendblock(Ogre::HlmsBlendblock const&)::__PRETTY_FUNCTION__> "const Ogre::HlmsBlendblock* Ogre::HlmsManager::getBlendblock(const Ogre::HlmsBlendblock&)") at assert.c:92
#3  0x00007ffff7561502 in __GI___assert_fail (
    assertion=0x7fffb23d9898 "baseParams.mBlockType == BLOCK_BLEND && \"baseParams.mBlockType should always be BLOCK_BLEND! You can ignore this assert,\" \" but it usually indicates memory corruption (or you created the block without"..., file=0x7fffb23d9570 "/media/data/ign/ogre/OgreMain/src/OgreHlmsManager.cpp", line=258, 
    function=0x7fffb23d9fe0 <Ogre::HlmsManager::getBlendblock(Ogre::HlmsBlendblock const&)::__PRETTY_FUNCTION__> "const Ogre::HlmsBlendblock* Ogre::HlmsManager::getBlendblock(const Ogre::HlmsBlendblock&)")
    at assert.c:101
#4  0x00007fffb1f78a10 in Ogre::HlmsManager::getBlendblock (this=0x7fffb43061c0, baseParams=...) at /media/data/ign/ogre/OgreMain/src/OgreHlmsManager.cpp:255
#5  0x00007fffb1f3920b in Ogre::Hlms::createDatablock (this=0x7fffb45d0ac0, name=..., refName=<error reading variable: Cannot access memory at address 0x64f0b678>, macroblockRef=..., blendblockRef=..., 
    paramVec=std::vector of length -15012014046490168, capacity -15012014088548807 = {...}, visibleToManager=240, filename="", resourceGroup="") at /media/data/ign/ogre/OgreMain/src/OgreHlms.cpp:1570
#6  0x00007fffb2c33000 in ignition::rendering::v4::Ogre2Material::Init (this=0x7fffb43497d0) at /media/data/ign/ign-rendering/ogre2/src/Ogre2Material.cc:524
#7  0x00007fffb2c984c8 in ignition::rendering::v4::Ogre2Scene::InitObject (this=0x7fffb47ff970, _object=std::shared_ptr<ignition::rendering::v4::Ogre2Object> (use count 2, weak count 1) = {...}, _id=65534, 
    _name="Default/TransRed") at /media/data/ign/ign-rendering/ogre2/src/Ogre2Scene.cc:471
[... cut ...]
#16 0x00007fffba2a371d in ignition::gazebo::v4::systems::SensorsPrivate::WaitForInit() () at ./src/systems/sensors/Sensors.cc:197
#17 0x00007fffba2a4542 in ignition::gazebo::v4::systems::SensorsPrivate::RenderThread() () at ./src/systems/sensors/Sensors.cc:284
#18 0x00007ffff538f6df in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#19 0x00007ffff73196db in start_thread (arg=0x7fffb9304700) at pthread_create.c:463
#20 0x00007ffff765271f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

</details>

darksylinc, 2021-11-04:
Sorry for taking my time. I know what's wrong but I need to be at my PC to look for the patch.

Basically in debug version ign needs to define OGRE_DEBUG_LEVEL to match (or alternatively disablr it on Ogre)

AllianceDarkSylinc, 2021-11-04:
OK I'm at the PC.

Here's the patch:

```diff
diff --git a/ogre2/src/CMakeLists.txt b/ogre2/src/CMakeLists.txt
index 012a52a3..d8423429 100644
--- a/ogre2/src/CMakeLists.txt
+++ b/ogre2/src/CMakeLists.txt
@@ -1,3 +1,8 @@
+if( UNIX )
+	# lld is much faster than ld for linking
+	set( CMAKE_SHARED_LINKER_FLAGS "-fuse-ld=lld" )
+endif()
+
 # Collect source files into the "sources" variable and unit test files into the
 # "gtest_sources" variable.
 ign_get_libsources_and_unittests(sources gtest_sources)
@@ -44,7 +49,7 @@ target_link_libraries(${ogre2_target}
     terra
     IgnOGRE2::IgnOGRE2)
 
-target_compile_definitions(${ogre2_target} PRIVATE $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
+-target_compile_definitions(${ogre2_target} PRIVATE $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
 
 
 set (versioned ${CMAKE_SHARED_LIBRARY_PREFIX}${PROJECT_NAME_LOWER}-${engine_name}${CMAKE_SHARED_LIBRARY_SUFFIX})
diff --git a/ogre2/src/terrain/Terra/CMakeLists.txt b/ogre2/src/terrain/Terra/CMakeLists.txt
index 83c9ac90..32404087 100644
--- a/ogre2/src/terrain/Terra/CMakeLists.txt
+++ b/ogre2/src/terrain/Terra/CMakeLists.txt
@@ -25,8 +25,8 @@ endif()
 
 add_definitions(-DOGRE_IGNORE_UNKNOWN_DEBUG)
 
-#target_compile_definitions(${PROJECT_NAME} PUBLIC
-#  $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
+target_compile_definitions(${PROJECT_NAME} PUBLIC
+  $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
 
 target_include_directories(${PROJECT_NAME}
   PRIVATE
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index dcfd6f2f..10d16225 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -1,3 +1,8 @@
+if( UNIX )
+	# lld is much faster than ld for linking
+	set( CMAKE_SHARED_LINKER_FLAGS "-fuse-ld=lld" )
+endif()
+
 # set compile definitions for tests
 set_property(
   SOURCE Camera_TEST.cc RenderTarget.cc
```

If the diff fails, what's really important is just:
`target_compile_definitions(${ogre2_target} PRIVATE $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)`

By defining DEBUG, Ogre headers will see this and ign-rendering will build correctly (assuming you're building with ` colcon build --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Debug -DBUILD_DOCS=OFF --merge-install`)

darksylinc, 2021-11-04:
OK I'm at the PC.

Here's the patch:

```diff
diff --git a/ogre2/src/CMakeLists.txt b/ogre2/src/CMakeLists.txt
index 012a52a3..d8423429 100644
--- a/ogre2/src/CMakeLists.txt
+++ b/ogre2/src/CMakeLists.txt
@@ -1,3 +1,8 @@
+if( UNIX )
+	# lld is much faster than ld for linking
+	set( CMAKE_SHARED_LINKER_FLAGS "-fuse-ld=lld" )
+endif()
+
 # Collect source files into the "sources" variable and unit test files into the
 # "gtest_sources" variable.
 ign_get_libsources_and_unittests(sources gtest_sources)
@@ -44,7 +49,7 @@ target_link_libraries(${ogre2_target}
     terra
     IgnOGRE2::IgnOGRE2)
 
-target_compile_definitions(${ogre2_target} PRIVATE $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
+-target_compile_definitions(${ogre2_target} PRIVATE $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
 
 
 set (versioned ${CMAKE_SHARED_LIBRARY_PREFIX}${PROJECT_NAME_LOWER}-${engine_name}${CMAKE_SHARED_LIBRARY_SUFFIX})
diff --git a/ogre2/src/terrain/Terra/CMakeLists.txt b/ogre2/src/terrain/Terra/CMakeLists.txt
index 83c9ac90..32404087 100644
--- a/ogre2/src/terrain/Terra/CMakeLists.txt
+++ b/ogre2/src/terrain/Terra/CMakeLists.txt
@@ -25,8 +25,8 @@ endif()
 
 add_definitions(-DOGRE_IGNORE_UNKNOWN_DEBUG)
 
-#target_compile_definitions(${PROJECT_NAME} PUBLIC
-#  $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
+target_compile_definitions(${PROJECT_NAME} PUBLIC
+  $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)
 
 target_include_directories(${PROJECT_NAME}
   PRIVATE
diff --git a/src/CMakeLists.txt b/src/CMakeLists.txt
index dcfd6f2f..10d16225 100644
--- a/src/CMakeLists.txt
+++ b/src/CMakeLists.txt
@@ -1,3 +1,8 @@
+if( UNIX )
+	# lld is much faster than ld for linking
+	set( CMAKE_SHARED_LINKER_FLAGS "-fuse-ld=lld" )
+endif()
+
 # set compile definitions for tests
 set_property(
   SOURCE Camera_TEST.cc RenderTarget.cc
```

If the diff fails, what's really important is just:
`target_compile_definitions(${ogre2_target} PRIVATE $<$<CONFIG:Debug>:DEBUG=1 _DEBUG=1>)`

By defining DEBUG, Ogre headers will see this and ign-rendering will build correctly (assuming you're building with ` colcon build --cmake-args -DBUILD_TESTING=OFF -DCMAKE_BUILD_TYPE=Debug -DBUILD_DOCS=OFF --merge-install`)

Ogre headers have a warning to detect this error but the fork stripped it :\

peci1, 2021-11-04:
Thanks, that did work. Now I'm back at "GL doesn't agree...".

This is the log with the added debug prints (I used `LogManager` instead of `printf`):

```
14:59:56: Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/rescue randy/2/meshes' of type 'FileSystem' to resource group 'General' with recursive option
14:59:56: Initialising resource group General
14:59:56: Can't assign material scene::Material(65487) because this Material does not exist. Have you forgotten to define it in a .material script?
14:59:56: Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/rescue randy/2/materials/textures/' of type 'FileSystem' to resource group 'General'
14:59:56: Texture: loading rescue_randy_roughness.png as rescue_randy_roughness.png
14:59:56: Texture: loading rescue_randy_metalness.png as rescue_randy_metalness.png
14:59:56: Texture: loading rescue_randy_albedo.png as rescue_randy_albedo.png
14:59:57: Texture: loading rescue_randy_normal.png as rescue_randy_normal.png
14:59:57: WARNING: normal map texture rescue_randy_normal.png is not BC5S compressed. This is encouraged for lower memory usage. If you don't want to see this message without compressing to BC5, set getDefaultTextureParameters()[TEXTURE_TYPE_NORMALS].pixelFormat to PF_R8G8_SNORM (or PF_BYTE_LA if RSC_TEXTURE_SIGNED_INT is not supported)
14:59:57: Added resource location '/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/caguero/models/smoke_generator/2/materials/textures/' of type 'FileSystem' to resource group 'General'
14:59:57: Texture: loading smoke.png as smoke.png
14:59:57: mVertexShader 536969216VertexShader_vs
14:59:57: mFragmetShader 536969216PixelShader_ps
14:59:57: paramName worldMatBuf
14:59:57: paramName f3dGrid
14:59:57: paramName f3dLightList
14:59:57: Vertex Shader: 537001984VertexShader_vs
Fragment Shader: 537001984PixelShader_ps
 GLSL validation result :
active samplers with a different type refer to the same texture image unit
14:59:57: mVertexShader 537001984VertexShader_vs
14:59:57: mFragmetShader 537001984PixelShader_ps
14:59:57: paramName worldMatBuf
14:59:57: paramName f3dGrid
14:59:57: paramName f3dLightList
14:59:57: paramName textureMaps
14:59:57: mVertexShader 1610612740VertexShader_vs
14:59:57: mFragmetShader 1610612740PixelShader_ps
14:59:57: paramName textureMapsArray
14:59:57: mVertexShader 536969344VertexShader_vs
14:59:57: paramName worldMatBuf
14:59:57: Vertex Shader: 537002112VertexShader_vs
Fragment Shader: 537002112PixelShader_ps
 GLSL validation result :
active samplers with a different type refer to the same texture image unit
14:59:57: mVertexShader 537002112VertexShader_vs
14:59:57: mFragmetShader 537002112PixelShader_ps
14:59:57: paramName worldMatBuf
14:59:57: paramName textureMaps
```

peci1, 2021-11-04:
And here are some details from GDB:

```
(gdb) frame 5
#5  0x00007fff94302538 in Ogre::GLSLMonolithicProgram::buildGLUniformReferences (this=0x7fffb551b5c0) at /media/data/ign/ogre/RenderSystems/GL3Plus/src/GLSL/OgreGLSLMonolithicProgram.cpp:297
297	            GLSLMonolithicProgramManager::getSingleton().extractUniformsFromProgram(
(gdb) p vertParams
$1 = (const Ogre::GpuConstantDefinitionMap *) 0x7fffb550fe10
(gdb) p hullParams
$2 = (const Ogre::GpuConstantDefinitionMap *) 0x0
(gdb) p domainParams
$3 = (const Ogre::GpuConstantDefinitionMap *) 0x0
(gdb) p fragParams
$4 = (const Ogre::GpuConstantDefinitionMap *) 0x7fffb5508870
(gdb) p geomParams
$5 = (const Ogre::GpuConstantDefinitionMap *) 0x0
(gdb) p computeParams
$6 = (const Ogre::GpuConstantDefinitionMap *) 0x0
(gdb) p mGLUniformReferences
$7 = std::vector of length 1, capacity 1 = {{mLocation = 9, mSourceProgType = Ogre::GPT_VERTEX_PROGRAM, mConstantDef = 0x7fffb5528990}}
(gdb) p mGLAtomicCounterReferences
$8 = std::vector of length 0, capacity 0
(gdb) p mGLUniformBufferReferences
$9 = std::vector of length 0, capacity 0
(gdb) p mSharedParamsBufferMap
$10 = std::map with 0 elements
(gdb) p mGLCounterBufferReferences
$11 = std::vector of length 0, capacity 0
(gdb) p mGLProgramHandle
$12 = 12
(gdb) frame 4
#4  0x00007fff94315328 in Ogre::GLSLProgramManager::extractUniformsFromProgram (this=0x7fffb40c6080, programObject=12, vertexConstantDefs=0x7fffb550fe10, hullConstantDefs=0x0, 
    domainConstantDefs=0x7fffb5508870, geometryConstantDefs=0x0, fragmentConstantDefs=0x0, computeConstantDefs=0x0, uniformList=std::vector of length 1, capacity 1 = {...}, 
    counterList=std::vector of length 0, capacity 0, uniformBufferList=std::vector of length 0, capacity 0, sharedParamsBufferMap=std::map with 0 elements, counterBufferList=std::vector of length 0, capacity 0)
    at /media/data/ign/ogre/RenderSystems/GL3Plus/src/GLSL/OgreGLSLProgramManager.cpp:624
624	                    assert(size_t (arraySize) == newGLUniformReference.mConstantDef->arraySize
(gdb) p foundSource
$13 = true
(gdb) p arraySize
$14 = 1
(gdb) p newGLUniformReference.mConstantDef
$15 = (const Ogre::GpuConstantDefinition *) 0x7fffb5514f80
(gdb) p newGLUniformReference.mConstantDef->arraySize
$16 = 3
(gdb) p newGLUniformReference                        
$17 = {mLocation = 8202, mSourceProgType = Ogre::GPT_FRAGMENT_PROGRAM, mConstantDef = 0x7fffb5514f80}
(gdb) p paramName
$18 = "textureMaps"
(gdb) p uniformName
$19 = "textureMaps[0]\000\065\065].indices4_7\000]\000cale[0]\000\f\316\270\264\377\177\000\000\377\377\377\377\000\000\000\000x%<\365\377\177\000\000 *\n\264\377\177\000\000\000\321\315G\r\327F%\300\220e\365\377\177\000\000 *\n\264\377\177\000\000\000U&\271\377\177\000\000\371\002<\365\377\177\000\000\230W\016\262\377\177\000\000NL>\365\377\177\000\000 *\n\264\377\177\000\000\362\v\036\262\377\177\000\000\240\066\202\265\377\177\000\000\000\216E\264\002\000\000\000\200U&\271\377\177\000\000 *\n\264\377\177\000\000\220U&\271\377\177\000\000\230R6\365\377\177\000\000\000\000\000\000\000\000\000"
(gdb) p std::string(uniformName)
Cannot look up value of a typedef `std::__cxx11::string'.
(gdb) p Ogre::String(uniformName)
A syntax error in expression, near `uniformName)'.
(gdb) p String(uniformName)
A syntax error in expression, near `uniformName)'.
(gdb) p uniformCount 
$20 = 2828
(gdb) p index
$21 = 2827
(gdb) p glType
$22 = 36289
(gdb) p uniformList
$23 = std::vector of length 1, capacity 1 = {{mLocation = 9, mSourceProgType = Ogre::GPT_VERTEX_PROGRAM, mConstantDef = 0x7fffb5528990}}
```

peci1, 2021-11-04:
I keep the GDB running if you wanted more info.

darksylinc, 2021-11-04:
What!?!? It is complaining about a PBS shader Ogre generated. I did not see this one coming.

I suspect it's textureMaps.

Find `hlmsPbs->setDebugOutputPath(false, false);` in ign-rendering/ogre2/src/Ogre2RenderEngine.cc and change it to:

`hlmsPbs->setDebugOutputPath(true, true, "/home/myusername/some_path_you_wish_to_dump/");`

(note it must end with '/')

Then repro the crash again. Look at the Ogre.log, the last entries where:

> 14:59:57: mVertexShader 537002112VertexShader_vs
> 14:59:57: mFragmetShader 537002112PixelShader_ps

Therefore go to /home/myusername/some_path_you_wish_to_dump/, locate these 2 files, and upload them here.

Cheers

peci1, 2021-11-04:
Weird, there is only one pair of files.

The log says

```
15:33:36: Texture: loading smoke.png as smoke.png
15:33:36: mVertexShader 536969216VertexShader_vs
15:33:36: mFragmetShader 536969216PixelShader_ps
15:33:36: paramName worldMatBuf
15:33:36: paramName f3dGrid
15:33:36: paramName f3dLightList
15:33:36: Vertex Shader: 537001984VertexShader_vs
Fragment Shader: 537001984PixelShader_ps
 GLSL validation result :
active samplers with a different type refer to the same texture image unit
15:33:36: mVertexShader 537001984VertexShader_vs
15:33:36: mFragmetShader 537001984PixelShader_ps
15:33:36: paramName worldMatBuf
15:33:36: paramName f3dGrid
15:33:36: paramName f3dLightList
15:33:36: paramName textureMaps
15:33:36: mVertexShader 1610612740VertexShader_vs
15:33:36: mFragmetShader 1610612740PixelShader_ps
15:33:36: paramName textureMapsArray
15:33:36: mVertexShader 536969344VertexShader_vs
15:33:36: paramName worldMatBuf
15:33:36: Vertex Shader: 537002112VertexShader_vs
Fragment Shader: 537002112PixelShader_ps
 GLSL validation result :
active samplers with a different type refer to the same texture image unit
15:33:36: mVertexShader 537002112VertexShader_vs
15:33:36: mFragmetShader 537002112PixelShader_ps
15:33:36: paramName worldMatBuf
15:33:36: paramName textureMaps
```

but I can only find debug info for 1610612740, which is not the last shader... Nevertheless, here they are: [1610612740.zip](https://github.com/ignitionrobotics/ign-rendering/files/7475596/1610612740.zip)

darksylinc, 2021-11-04:
Hi!

It sounds like you set HlmsUnlit to dump the shaders, instead of HlmsPbs. There should be two instances of setDebugOutputPath; try setting them both

peci1, 2021-11-04:
You're right. Here is the PBS log
[537002112.zip](https://github.com/ignitionrobotics/ign-rendering/files/7475681/537002112.zip)

darksylinc, 2021-11-04:
OK I was about to rule it out a driver bug; until I noticed something that's very wrong.

Ogre at some set `hlms_render_depth_only` which means we're not rendering colour, but depth. This is usually left for shadow maps; but when that happens Ogre disables the pixel shader entirely.

For some reason this pixel shader was not disabled, which means we're not at shadow mapping stage.

Another reason for rendering to only depth is early Z prepass which AFAIK ignition doesn't use. I have no idea why hlms_render_depth_only was set. I suspect Ignition must be rendering to a PF_DEPTH_* format directly but trying to use PBS (which is invalid).

When the crash happens, could you post the full call stack? I want to see ignition's calls to see where it is in its rendering.

darksylinc, 2021-11-04:
**Update:**

Yeah, definitely this could happen if ignition tries to render colour into a depth texture directly:

```cpp
setProperty( HlmsBaseProp::RenderDepthOnly,
             renderTarget->getForceDisableColourWrites() ? 1 : 0 );```

I remember a similar issue with the LIDAR rendering; where `Ogre2LaserRetroMaterialSwitcher::cameraPreRenderScene` would switch the materials for objects but accidentally leave other objects that failed the test untouched and would attempt to render them using standard PBS shaders.

I suspect something similar is happening.

At least now that I know what's happening I should be able to repro this problem; rather than relying on the driver screaming to us we could throw an exception if we're trying to render Pbs colour into a depth target; and from there see what is going on.

peci1, 2021-11-04:
I think it is a depth camera where it crashes.

Here's the stack:

<details>

```
ign gazebo -v4 -s /home/peci1/Stažené/sensor_particles.sdf: /media/data/ign/ogre/RenderSystems/GL3Plus/src/GLSL/OgreGLSLProgramManager.cpp:625: void Ogre::GLSLProgramManager::extractUniformsFromProgram(GLuint, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, const GpuConstantDefinitionMap*, Ogre::GLUniformReferenceList&, Ogre::GLAtomicCounterReferenceList&, Ogre::GLUniformBufferList&, Ogre::SharedParamsBufferMap&, Ogre::GLCounterBufferList&): Assertion `size_t (arraySize) == newGLUniformReference.mConstantDef->arraySize && "GL doesn't agree with our array size!"' failed.
Stack trace (most recent call last) in thread 27819:
#31   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-4/plugins/libignition-gazebo-sensors-system.so", at 0x7f737727a54f, in ignition::gazebo::v4::systems::SensorsPrivate::RenderThread()
#30   Object "/usr/lib/x86_64-linux-gnu/ign-gazebo-4/plugins/libignition-gazebo-sensors-system.so", at 0x7f737727a028, in ignition::gazebo::v4::systems::SensorsPrivate::RunOnce()
#29   Object "/usr/lib/x86_64-linux-gnu/libignition-sensors4.so.4", at 0x7f7376669d06, in ignition::sensors::v4::Manager::RunOnce(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
#28   Object "/usr/lib/x86_64-linux-gnu/libignition-sensors4.so.4", at 0x7f737667a745, in ignition::sensors::v4::Sensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, bool)
#27   Object "/usr/lib/x86_64-linux-gnu/libignition-sensors4-depth_camera.so", at 0x7f72f3de42ff, in ignition::sensors::v4::DepthCameraSensor::Update(std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&)
#26   Object "/usr/lib/x86_64-linux-gnu/libignition-sensors4-rendering.so.4", at 0x7f73768a4e33, in ignition::sensors::v4::RenderingSensor::Render()
#25   Object "/usr/lib/x86_64-linux-gnu/ign-rendering-4/engine-plugins/libignition-rendering-ogre2.so", at 0x7f737457c5b9, in ignition::rendering::v4::Ogre2DepthCamera::Render()
#24   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f73678d5cfc, in Ogre::Root::renderOneFrame()
#23   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f73678d85a8, in Ogre::Root::_updateAllRenderTargets()
#22   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367a81761, in Ogre::CompositorManager2::_update(Ogre::SceneManagerEnumerator&, Ogre::HlmsManager*)
#21   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367879620, in Ogre::RenderSystem::updateCompositorManager(Ogre::CompositorManager2*, Ogre::SceneManagerEnumerator&, Ogre::HlmsManager*)
#20   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367a81b1f, in Ogre::CompositorManager2::_updateImplementation(Ogre::SceneManagerEnumerator&, Ogre::HlmsManager*)
#19   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367ab2ba8, in Ogre::CompositorWorkspace::_update()
#18   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367a956b8, in Ogre::CompositorNode::_update(Ogre::Camera const*, Ogre::SceneManager*)
#17   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367ad85d0, in Ogre::CompositorPassScene::execute(Ogre::Camera const*)
#16   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f736789ada4, in Ogre::RenderTarget::_updateViewportRenderPhase02(Ogre::Viewport*, Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char, bool)
#15   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f7367a38d02, in Ogre::Viewport::_updateRenderPhase02(Ogre::Camera*, Ogre::Camera const*, unsigned char, unsigned char)
#14   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f73675caf50, in Ogre::Camera::_renderScenePhase02(Ogre::Camera const*, Ogre::Viewport*, unsigned char, unsigned char, bool)
#13   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f73678ec6d2, in Ogre::SceneManager::_renderPhase02(Ogre::Camera*, Ogre::Camera const*, Ogre::Viewport*, unsigned char, unsigned char, bool)
#12   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f736786dad5, in Ogre::RenderQueue::render(Ogre::RenderSystem*, unsigned char, unsigned char, bool, bool)
#11   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f736786e357, in Ogre::RenderQueue::renderGL3(bool, bool, Ogre::HlmsCache*, Ogre::RenderQueue::RenderQueueGroup const&, Ogre::IndirectBufferPacked*, unsigned char*, unsigned char*)
#10   Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreMain.so.2.1.0", at 0x7f73676a1d63, in Ogre::Hlms::getMaterial(Ogre::HlmsCache const*, Ogre::HlmsCache const&, Ogre::QueuedRenderable const&, unsigned char, bool)
#9    Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/libOgreHlmsPbs.so.2.1.0", at 0x7f73670b1c08, in Ogre::HlmsPbs::createShaderCacheEntry(unsigned int, Ogre::HlmsCache const&, unsigned int, Ogre::QueuedRenderable const&)
[... cut ...]
#4    Object "/media/data/ign/ogre/install/lib/x86_64-linux-gnu/OGRE-2.1/OGRE/RenderSystem_GL3Plus.so", at 0x7f7350115327, in Ogre::GLSLProgramManager::extractUniformsFromProgram(unsigned int, std::map<VERY_LONG_TEMPLATE_HELL>&)
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73b489a501, in __assert_fail
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73b489a489, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73b48aa920, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f73b48a8fb7, in gsignal
```

</details>

darksylinc, 2021-11-04:
OK I understand what's going on now.

I have a quick question for @iche033 :

Is the object (I suspect it's the human model) supposed to be rendered by Ogre2DepthCamera?

If the answer is no, it's an ignition bug
If the answer is yes, it's an Ogre bug

From Ogre side we should be able to fix it by adding this snippet:

```
@property( hlms_render_depth_only )
    @set( hlms_disable_stage, 1 )
@end
```

Into Hlms/Pbs/GLSL/PixelShader_ps.glsl

peci1, 2021-11-04:
Awesome! Adding your snippet to the shader fixed the issue. Let's wait for Ian to answer, but I think the answer will be yes, the randy should be rendered in depth camera.

peci1, 2021-11-04:
Maybe because it uses a full stack of PBR textures, while other objects usually don't?

```XML
                  <pbr>
                        <metal>
                            <albedo_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_albedo.png</albedo_map>
                            <normal_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_normal.png</normal_map>
                            <metalness_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_metalness.png</metalness_map>
                            <roughness_map>https://fuel.ignitionrobotics.org/1.0/openrobotics/models/rescue randy/2/files/materials/textures/rescue_randy_roughness.png</roughness_map>
                        </metal>
                    </pbr>

```

darksylinc, 2021-11-04:
> yep the depth camera should see the rescue randy model. hmm I don't know why only that particular model has this issue though since other models are work fine and appear in the depth camera view.

> Maybe because it uses a full stack of PBR textures, while other objects usually don't?
Yes, that's the reason

Just to clarify what's happening:

 1. Ogre creates a VS and PS
 2. Ogre's scripts see we're only outputting depth; so it leaves out the part where it writes to colour
     - This is done so in case a user extension wants to write to depth; though probably should be handled better (i.e. disable by default, force extension to re-enable)
 3. Driver (this is driver specific) sees all computation is redundant (no output to colour), thus it optimizes out a lot or all the code. Here's what's causing the crash; because Ogre expects textureMaps[3] but it got partially optimized out (apparently it got optimized into textureMaps[1]... for whatever reason)
 4. Later on (if the driver didn't optimize out the code -> no crash) the pixel shader is set; but the driver will never use it because it outputs to depth. The result is that it still works fine

The fix is to tell Ogre not to compile the Pixel Shader (i.e. null out the Hlms-generated output) when we detect we're only outputting depth. That way driver will never see this PS and Ogre will never have a PS either.

peci1, 2021-12-23:
@darksylinc Is this suggested fix something that should go into a PR? Or is it just a workaround?

> From Ogre side we should be able to fix it by adding this snippet:
> 
> ```
> @property( hlms_render_depth_only )
>     @set( hlms_disable_stage, 1 )
> @end
> ```
> 
> Into Hlms/Pbs/GLSL/PixelShader_ps.glsl (no need to update Ogre since Ignition bundles this file into ign-rendering's repo)

darksylinc, 2021-12-23:
Ouch ouch ouch

Now that I'm with a fresh brain (honestly I forgot this issue wasn't fixed), I realized my assessment was partially wrong (and the fix is sadly not the right one).

[What I said](https://github.com/ignitionrobotics/ign-rendering/issues/260#issuecomment-961334669) is right except:

1. We already set `hlms_disable_stage` when `hlms_render_depth_only` is on:
```
@property( hlms_render_depth_only && !alpha_test && !hlms_shadows_esm && !macOS)
	@set( hlms_disable_stage, 1 )
@end
````
2. However `alpha_test` is key. We still need to execute the pixel shader if alpha_test is enabled.
3. Thus after simplification the error boils down:
```glsl
uniform sampler2DArray textureMaps[3];
diffuseCol = texture( textureMaps[0], vec3( UV_DIFFUSE( inPs.uv0.xy ), diffuseIdx ) );
if( material.kD.w >= diffuseCol.a )
	discard;
```
4. The value `textureMaps[0]` gets used and is relevant thus the driver cannot optimize it out, but `textureMaps[1]` and `textureMaps[2]` although they're sampled, they can be optimized out since their results lead to nowhere.
5. Hence the mismatch textureMaps.size = 1 vs textureMaps.size = 3 in C++

I'll have to think of a better way to fix this so that only diffuse textures are sent in this case

darksylinc, 2021-12-23:
OK I noticed this problem should be gone from Ogre 2.2

Are you able to upgrade to Fortress (which uses 2.2)?

peci1, 2021-12-26:
> Are you able to upgrade to Fortress (which uses 2.2)?

That will have to happen in a short time. But Edifice is still affected.

I verified that in Fortress, I can successfully run the example that crashes in Dome.

azeey, 2024-05-31:
Closing since this is fixed in Fortress

closed 2024-05-31 by azeey