gazebosim/gz-rendering #283: Can't select capsule objects in the 3D scene

jennuine, 2021-03-25:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  
  
Source, `main` branch, latest commit 757afcdcb059f1aed4f70f9e6062bccf9483d754
(for Edifice)

## Description
* Expected behavior: Clicked capsule object should show a wirebox when selected from the 3D scene
* Actual behavior: Can't select capsule object from the scene, can only select it from the entity tree

## Steps to reproduce

1. `ign gazebo shapes.sdf`
2. Attempt to click the capsule object in the scene

## Output

[image]

chapulina, 2021-03-25:
The capsule is different from the other shapes because it doesn't use a `unit_*` mesh and has a custom class. I think we may need to set some flag to the geometry so the ray cast by the mouse intersects it.

chapulina, 2021-05-06:
> I think we may need to set some flag to the geometry so the ray cast by the mouse intersects it.

I believe that's `IGN_VISIBILITY_SELECTABLE`

ahcorde, 2021-05-10:
PR https://github.com/ignitionrobotics/ign-rendering/pull/319

closed 2021-05-11 by iche033