gazebosim/gz-rendering #300: Specific issue templates including GPU

peci1, 2021-04-05:
## Desired behavior
Issue templates were first added in #234 and then centralized organization-wide in #250. However, I think that it would be useful to ask the reporters for the GPU and driver versions they use when reporting rendering-related issues. I'm not sure about the best approach, though. 

## Alternatives considered
Leave it as it is. This requires extra effort from maintainers asking for the used GPUs and leaves some issues in a hard to replicate state when no concrete GPU is mentioned.

## Implementation suggestion
If you do not want to maintain two versions of the issue template, you might as well add the GPU-specific section to the general template with a note that it is only needed when reporting rendering-related issues.

The issue template should include easy to follow instructions for getting the requested info - e.g. run `glxinfo | head` and `uname -a` or something like that (+ of course some instructions for Windows). If the instructions would get too long, they might be written as an ign-docs tutorial linked from the issue template.

peci1, 2021-04-05:
This would also affect ign-sensors, ign-gui and ign-gazebo.

chapulina, 2021-04-05:
> you might as well add the GPU-specific section to the general template with a note that it is only needed when reporting rendering-related issues.

This sounds like a reasonable compromise for now. Mind opening a PR to https://github.com/ignitionrobotics/.github?

Another idea that came to mind was adding a specific issue template for rendering bugs, but that would need to be added to the entire org, and I worry it may cause confusion on repositories that don't touch rendering.

darksylinc, 2021-04-12:
On Linux, the following commands can give necessary information about GPU:

```
lspci -nn | grep VGA
lshw -C display -numeric
glxinfo -B
sudo X -version
```

peci1, 2021-04-12:
Done: https://github.com/ignitionrobotics/.github/pull/3 .

closed 2021-04-14 by chapulina