gazebosim/gz-rendering #360: CMake policy CMP0072

chapulina, 2021-07-07:
## Environment
* OS Version: Ubuntu Focal
* Source or binary build? Source
* CMake 3.16

## Description
* Expected behavior:  Builds with no warnings
* Actual behavior: 

```
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.
                                                                      
  FindOpenGL found both a legacy GL library:    

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for 
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  /home/chapulina/dev_focal/ws_fortress/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:189 (find_package)
  CMakeLists.txt:58 (ign_find_package)
```

## Steps to reproduce

Compile from source with a CMake version higher than 3.10

## Solution

If we want to keep compatibility with CMake 3.10, I think we should explicitly set the policy to `OLD`.

mjcarroll, 2021-07-07:
Further reference: https://cmake.org/cmake/help/git-stage/policy/CMP0072.html

I think the better behavior is to check if the policy exists and make use of the newer version.  This shouldn't have any impact on our current cmake mechanisms, as far as I can tell:

```
if (POLICY CMP0072)
  cmake_policy (SET CMP0072 NEW)
endif(POLICY CMP0072)
```

I believe we also need to propagate it out to the `ignition-renderingXX-ogre-config` and `ignition-renderingXX-ogre2-config` to make sure that downstream users don't also get the warning.

darksylinc, 2021-07-20:
A bit more background on what this error mean (to make a better informed decision):

The context is switchable graphics. Eons ago drivers would replace libGL.so with their own and call it a day.

But today it is very normal to have multiple GPUs installed (specially laptops with switchable graphics between Intel and NVIDIA; AMD + Intel combo isn't a problem because they are both provided by Mesa so it's the same libGL.so)

So a new library was made, libOpenGL.so, which decides at launch time which libGL.so to launch. This interface is known as GLVND.

If [configured properly](https://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/primerenderoffload.html) launching:

```
__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia glxinfo
__GLX_VENDOR_LIBRARY_NAME=mesa glxinfo
```

Today it works out of the box if using the NVIDIA installer (no need to mess with xorg.conf), but Ubuntu 20.04's packaged NVIDIA drivers broke this (maybe on purpose?) as trying to use mesa will give X11 interface errors.

So basically the error is saying link against libOpenGL instead of libGL to better support switchable graphics.

Ogre 2.1+ uses `dlopen( "libGL.so.1", RTLD_LAZY | RTLD_LOCAL)` at runtime rather than relying on the linker and it seems GLVND at that time can already intercept the call and choosing Mesa vs NVIDIA it seems to work (I use it all the time on gazebo to test NV and AMD discrete GPUs)

Ogre 1.x uses a very similar way to load OpenGL so it should be working, but I didn't check it.

Hence this CMake warning probably can be safely ignored because we don't actually link against libGL or libOpenGL during build time; unless you have other tools (other than Ogre) linking against GL directly.

chapulina, 2022-01-06:
I tried a few things here but I wasn't able to silence the warning while still using `ign_find_package(OpenGL...)`. The `cmake_policy` command has complicated scoping. These didn't work:

```cmake
  if (POLICY CMP0072)
    cmake_policy (SET CMP0072 NEW)
  endif(POLICY CMP0072)
  ign_find_package(OpenGL
    REQUIRED_BY ogre ogre2
    PKGCONFIG gl)
```

```cmake
  if (POLICY CMP0072)
    cmake_policy (SET CMP0072 NEW)
  endif(POLICY CMP0072)
  ign_find_package(OpenGL
    REQUIRED_BY ogre ogre2
    PKGCONFIG gl
    EXTRA_ARGS NO_POLICY_SCOPE)
```

```cmake
  if(POLICY CMP0072)
    cmake_policy(PUSH)
    cmake_policy(SET CMP0072 NEW)
  endif()
  ign_find_package(OpenGL
    REQUIRED_BY ogre ogre2
    PKGCONFIG gl)
  if(POLICY CMP0072)
    cmake_policy(POP)
  endif()
```

This does work, but we lose the `REQUIRED_BY` and `PKGCONFIG`:

```cmake
  if (POLICY CMP0072)
    cmake_policy (SET CMP0072 NEW)
  endif(POLICY CMP0072)
  find_package(OpenGL)
```

Other things that I thought of trying:

* Adding a `POLICY` parameter to `ign_find_package`
* Hardcoding the policy in `ign-cmake`'s `ign_find_package`
* Creating a custom `FindOpenGL.cmake` where the policy is set

Does anyone have a better idea? Now that Jenkins CI is using Focal  (https://github.com/ignition-tooling/release-tools/pull/565) this warning is turning the build yellow.

darksylinc, 2022-01-06:
What about [setting](https://github.com/Kitware/CMake/blob/2f244b4cb82e4a91162d2965a0cea6da12359665/Modules/FindOpenGL.cmake#L268) `OpenGL_GL_PREFERENCE` variable?

I guess push/pop policy doesn't make sense in this case because once FindOpenGL has executed, the effects are irreversible (variables are set with OGL libs and folders), even if the policy is popped

chapulina, 2022-01-06:
> What about setting OpenGL_GL_PREFERENCE variable?

Thanks, that does it! https://github.com/ignitionrobotics/ign-rendering/pull/528

That's still not propagated to downstream projects though, not sure if `ign-cmake` provides a way to propagate the variable to the config files like @mjcarroll mentioned above.

azeey, 2024-05-31:
Fixed by https://github.com/gazebosim/gz-rendering/pull/528

PR #528 (2022-01-06): Silence CMake policy CMP0072
# 🦟 Bug fix

* Fixes #360

## Summary

Calling `cmake_policy` is complicated in this situation because the `find_package` call is made in a different scope, within `ign_find_package`. Setting the variable directly does the trick though. See https://github.com/ignitionrobotics/ign-rendering/issues/360#issuecomment-1006278927.

This doesn't solve the problem for downstream projects though. I didn't find a way of passing variables to [ignition-component-config.cmake.in](https://github.com/ignitionrobotics/ign-cmake/blob/ign-cmake2/cmake/ignition-component-config.cmake [... cut]
closed 2024-05-31 by azeey