gazebosim/gz-rendering #362: Selection buffer crash when resizing window

chapulina, 2021-07-16:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build? Source, `main` (ac1ab2ca04143707804933aca64c6ea06e1305bd)
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: `ogre2`
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running in Docker/Singularity

<details><summary> Rendering system info</summary>
<p>

    ```
    $           LANG=C lspci -nn | grep VGA  # might require installing pciutils
    09:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:2503] (rev a1)

    $           echo "$DISPLAY"
    :1

    $           LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
    direct rendering: Yes
        Preferred profile: core (0x1)
        Max core profile version: 4.5
        Max compat profile version: 3.1
        Max GLES1 profile version: 1.1
        Max GLES[23] profile version: 3.2
    OpenGL vendor string: Mesa/X.org
    OpenGL renderer string: llvmpipe (LLVM 11.0.0, 256 bits)
    OpenGL core profile version string: 4.5 (Core Profile) Mesa 20.2.6
    OpenGL core profile shading language version string: 4.50
    OpenGL core profile context flags: (none)
    OpenGL core profile profile mask: core profile
    OpenGL version string: 3.1 Mesa 20.2.6
    OpenGL shading language version string: 1.40
    OpenGL context flags: (none)
    OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.2.6
    OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

    $           ps aux | grep Xorg
    develop+  566090  0.0  0.0   3312   412 pts/2    S+   17:26   0:00 grep --color=auto Xorg
    ```

</p>
</details>

- [x] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details><summary>~/.ignition/rendering/ogre2.log</summary>
  
```
17:19:26: Creating resource group General
17:19:26: Creating resource group Internal
17:19:26: Creating resource group Autodetect
17:19:26: SceneManagerFactory for type 'DefaultSceneManager' registered.
17:19:26: Registering ResourceManager for type Material
17:19:26: Registering ResourceManager for type Mesh
17:19:26: Registering ResourceManager for type Mesh2
17:19:26: Registering ResourceManager for type OldSkeleton
17:19:26: MovableObjectFactory for type 'ParticleSystem' registered.
17:19:26: ArchiveFactory for archive type FileSystem registered.
17:19:26: ArchiveFactory for archive type Zip registered.
17:19:26: ArchiveFactory for archive type EmbeddedZip registered.
17:19:26: DDS codec registering
17:19:26: FreeImage version: 3.18.0
17:19:26: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
17:19:26: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
17:19:26: OITD codec registering
17:19:26: Registering ResourceManager for type HighLevelGpuProgram
17:19:26: MovableObjectFactory for type 'Entity' registered.
17:19:26: MovableObjectFactory for type 'Item' registered.
17:19:26: MovableObjectFactory for type 'Light' registered.
[... 504 lines cut ...]

## Description

Ignition Gazebo crashed while resizing the window by dragging the top-right corner.

## Steps to reproduce

It's a flaky crash, it doesn't happen often.

## Output

Backtrace:

```
/home/developer/ign_ws/install/lib/ign-rendering-6/engine-plugins/libignition-rendering-ogre2.so(_ZN8ignition9rendering2v620Ogre2SelectionBu
ffer16OnSelectionClickEii+0x6b) [0x7f02edf63dcb] /home/developer/ign_ws/src/ign-rendering/ogre2/src/Ogre2SelectionBuffer.cc:221             
[0x7f02ededb78e]                                                                                                                            
/home/developer/ign_ws/install/lib/ign-gazebo-6/plugins/gui/libGzScene3D.so(0x7f03c80950dd) [0x7f03c80950dd]                                
/home/developer/ign_ws/install/lib/ign-gazebo-6/plugins/gui/libGzScene3D.so(_ZN8ignition6gazebo2v611IgnRenderer27HandleMouseTransformControl
Ev) /home/developer/ign_ws/install/include/ignition/math6/ignition/math/Vector2.hh:68
```

The offending line, according to the backtrace, doesn't seem to have anything suspicious to me:

https://github.com/ignitionrobotics/ign-rendering/blob/f3f75a58cddf150ec96d8ba1ad1e880069bdc2e7/ogre2/src/Ogre2SelectionBuffer.cc#L221

darksylinc, 2021-07-20:
Can you upload a video (e.g. OBS, or even capturing from a phone works) of how you're reproducing the error? 'dragging the top-right corner' can be interpreted in several ways.

> The offending line, according to the backtrace, doesn't seem to have anything suspicious to me:

But the previous one `vp->getTarget();` is suspicious. If OnSelectionClick got called too early (or worse, is a race condition) then the Viewport may be a dangling pointer.

chapulina, 2021-07-20:
> Can you upload a video (e.g. OBS, or even capturing from a phone works) of how you're reproducing the error? 'dragging the top-right corner' can be interpreted in several ways.

The crash itself is hard to reproduce, but this is what I did when it happened:

[image]

darksylinc, 2021-07-20:
Ahh thanks!

It probably reproduces when you move the mouse much faster, so that the cursor ends up in the 3D area to force a selectionbuffer, and resizing on the next frame (or viceversa, anyway, fast mouse movement to touch the center of the 3D area)

Probably getting rid of the right pane helps in reproducibility

chapulina, 2021-07-20:
> Probably getting rid of the right pane helps in reproducibility

Worked like a charm :grimacing: 

[image]

darksylinc, 2021-07-26:
I can't repro the bug in this environment:

 - Xubuntu 18.04 LTS
 - AMD Radeon RX 560 (Mesa drivers)

I suspect reproducibility has to do with the Qt version bundled by the distro, since I've seen the threading changes within Qt mess with code very close to the crash that has caused issues in the past.

I'll try to repro the bug in Ubuntu 20.04 at another time.

chapulina, 2021-11-17:
Fixed in https://github.com/ignitionrobotics/ign-gazebo/pull/969

closed 2021-11-17 by chapulina