gazebosim/gz-rendering #369: DepthCameraTest.DepthCameraBoxes/ogre2 failing on macOS

jacobperron, 2021-07-26:
## Environment
* OS Version: macOS
* Source or binary build?
  Source (latest Citadel 9341eb24f37a5c2b1ceb5c2dc07ab8473a2a0f9a and Ediface 44a19485aa32f48fd116ad6b5e938996eb5fbd90)

## Description

Example ign-rendering5 failure: https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering5-homebrew-amd64/17/testReport/(root)/DepthCamera_DepthCameraTest/DepthCameraBoxes_ogre2/

Example ign-rendering3 failure: https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/44/testReport/junit/(root)/DepthCamera_DepthCameraTest/DepthCameraBoxes_ogre2/

## Output

The output is long and repeats). Here is the first bunch of output before it starts repeating:

```
53: [ RUN      ] DepthCamera/DepthCameraTest.DepthCameraBoxes/ogre2
53: [Err] [Ogre2RenderEngine.cc:432] Unable to find Ogre Plugin[/usr/local/opt/ogre2.1/lib/OGRE-2.1/RenderSystem_GL3Plus]. Rendering will not be possible.Make sure you have installed OGRE properly.
53: /Users/jenkins/workspace/ignition_rendering-ci-ign-rendering3-homebrew-amd64/ign-rendering/test/integration/depth_camera.cc:254: Failure
53: Expected: (mb) > (0u), actual: 0 vs 0
53: /Users/jenkins/workspace/ignition_rendering-ci-ign-rendering3-homebrew-amd64/ign-rendering/test/integration/depth_camera.cc:455: Failure
53: Expected: (b) > (0u), actual: 0 vs 0
53: /Users/jenkins/workspace/ignition_rendering-ci-ign-rendering3-homebrew-amd64/ign-rendering/test/integration/depth_camera.cc:456: Failure
53: Expected equality of these values:
53:   255u
53:     Which is: 255
53:   a
53:     Which is: 0
```

darksylinc, 2021-07-31:
Ugh.

This bug was supposed to be fixed with https://github.com/ignitionrobotics/ign-rendering/pull/339/files

The root of the problem is that the value on memory in floating point representation was very close to 0 (e.g. blue = 0xff while everything else including alpha is 0x00 is 3.57331e-43f) so GPU was flushing it to 0.0f.

This is probably HW / driver specific so full info on your OS version and HW model is appreciated

chapulina, 2021-08-02:
These tests ran in one of our CI machines: https://build.osrfoundation.org/computer/mac-t1000.catalina/

Maybe :farmer: @scpeters can help digging into why it's still failing on that machine.

scpeters, 2021-08-16:
> These tests ran in one of our CI machines: https://build.osrfoundation.org/computer/mac-t1000.catalina/

Yes, it only seems to fail on that machine. Here is some info about the operating system and hardware:

## Some info from `brew config`

~~~
CPU: octa-core 64-bit ivybridge
Clang: 12.0.0 build 1200
Git: 2.32.0 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 10.15.7-x86_64
CLT: 12.0.0.32.29
Xcode: N/A
~~~

## Hardware Overview from system profiler

~~~
  Model Name:	Mac mini
  Model Identifier:	Macmini6,2
  Processor Name:	Quad-Core Intel Core i7
  Processor Speed:	2.3 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Hyper-Threading Technology:	Enabled
  Memory:	8 GB
  Boot ROM Version:	421.0.0.0.0
  SMC Version (system):	2.8f1
~~~

## Graphics/Displays from System Profiler

~~~
Intel HD Graphics 4000:

  Chipset Model:	Intel HD Graphics 4000
  Type:	GPU
  Bus:	Built-In
  VRAM (Dynamic, Max):	1536 MB
  Vendor:	Intel
  Device ID:	0x0166
  Revision ID:	0x0009
  Metal:	Supported, feature set macOS GPUFamily1 v4
~~~

jacobperron, 2021-09-13:
Same failure again on t1000: https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering5-homebrew-amd64/19/

darksylinc, 2021-09-13:
I forgot to comment on this ticket:

> Intel HD Graphics 4000

This GPU has multiple HW bugs which drivers have to workaround. I'm not surprised it randomly fails on macOS on this HW given that Apple isn't paying much attention to GL drivers.

Unfortunately this bug is nearly unfixable for that system. Perhaps a full rewrite on how data is stored could fix it (i.e. stop reinterpret casting ints to floats, that'd be a great start); but there are other bugs in Apple's GL implementation as well when dealing with UINT/SINT formats (vs UNORM/SNORM formats)

Blast545, 2022-07-27:
:man_farmer: This test and https://github.com/gazebosim/gz-rendering/issues/253 started failing recently on Citadel.

Here it was working fine: [ignition_rendering-ci-ign-rendering3-homebrew-amd64#72](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/72/)
Here [ignition_rendering-ci-ign-rendering3-homebrew-amd64#73/](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/73/) started failing DepthCamera_DepthCameraTest/DepthCameraBoxes_ogre2
And here started failing ThermalCameraBoxes_ogre2 [ignition_rendering-ci-ign-rendering3-homebrew-amd64#74/](https://build.osrfoundation.org/job/ignition_rendering-ci-ign-rendering3-homebrew-amd64/74/)

The problem may be triggered by new versions of the dependency packages for the job. For now we'll disable the tests on Citadel (Backporting https://github.com/gazebosim/gz-rendering/pull/405) and attach the logs of these jobs to this issue so they can be investigated later.

FYI: @Crola1702

darksylinc, 2022-12-24:
This bug will be fixed for good in Garden once PR #785 gets merged.

It may be possible to backport the changes to Fortress.

PR #785 (2022-12-28): Add Heightmap integration test
# 🦟 Bug fix

Almost all issues in #546 are covered.

Only Thermal Camera is not covered.

Fixes #332 (again) for macOS and possibly any OS/HW/Driver combination.
Fixes #369

## Summary

This PR adds a heightmap test.

This test has uncovered a lot of bugs recently. 

Note: This PR currently includes part of #784 because it otherwise will fail unit tests.

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them) [... cut]
commit 4de8d1e4 (2022-12-28): Add Heightmap integration test (#785)

This PR adds a heightmap test.

Almost all issues in #546 are covered.

Only Thermal Camera is not covered.

Fixes #332 (again) for macOS and possibly any OS/HW/Driver combination.
Fixes #369
PR #797 (2022-12-28): Fix Ogre2DepthCamera on macOS (Fortress)
# 🦟 Bug fix

Fixes #332 (again) for macOS and possibly any OS/HW/Driver combination.
Fixes #369 

## Summary

For a very long time macOS (and possibly other OS/HW/Driver combination) would fail the tests because the shader code was doing (simplified, in C to convey the explanation):

```cpp
uint32_t valueToStore = r << 24 | g << 16 | b << 8 | a;
float output;
output = (float*)&valueToStore;
return output;
```

In ticket #332 I made a deep analysis and found out that given the wrong conditions, the bit pattern would result in either a denormal or a NaN i [... cut]
commit 41be2408 (2022-12-28): Fix Ogre2DepthCamera on macOS (Fortress) (#797)

Fixes #332 (again) for macOS and possibly any OS/HW/Driver combination.
Fixes #369

For a very long time macOS (and possibly other OS/HW/Driver combination) would fail the tests because the shader code was doing (simplified, in C to convey the explanation):

```
uint32_t valueToStore = r << 24 | g << 16 | b << 8 | a;
float output;
output = (float*)&valueToStore;
return output;
```

In ticket #332 I made a deep analysis and found out that given the wrong conditions, the bit pattern would result in either a denormal or a NaN in floating point; whi [... cut]
closed 2022-12-28 by iche033