gazebosim/gz-rendering #374: rendering3-3.5.0 fails to build with ogre/1.10.12

mboisson, 2021-08-04:
## Environment
* OS Version: Gentoo on CentOS7
* Source build
* using rendering3-3.5.0 and ogre 1.10.12

## Description
I was trying to build ignition-citadel using ogre/1.10.12, since it appears that it has issues with ogre/1.12.12 (https://github.com/osrf/gazebo/issues/2700). However, I hit the following compilation error: 

```
22031 /tmp/mboisson/avx2/ignition/citadel/GCC-9.3.0/ign-rendering-ignition-rendering3_3.5.0/ogre/src/OgreMaterialSwitcher.cc: In member function virtual Ogre::Technique* ignition::rendering::v3::OgreMaterialSwitcher::handleSchemeNotFound(uint16_t, const String&, Ogre::Material*, uint16_t, const Ogre::Renderable*):
22032 /tmp/mboisson/avx2/ignition/citadel/GCC-9.3.0/ign-rendering-ignition-rendering3_3.5.0/ogre/src/OgreMaterialSwitcher.cc:86:53: error: no matching function for call to static_pointer_cast<Ogre::Material>(Ogre::ResourcePtr&)
22033    86 |         std::static_pointer_cast<Ogre::Material>(res);
22034       |
```

Is there a missing version requirement for ogre for rendering3-3.5.0 ?

The same recipe builds fine against ogre/1.12.12.

chapulina, 2021-08-04:
We're explicitly depending on 1.8.0:

https://github.com/ignitionrobotics/ign-rendering/blob/9341eb24f37a5c2b1ceb5c2dc07ab8473a2a0f9a/CMakeLists.txt#L64

mboisson, 2021-08-05:
Mmm, shouldn't it abort at configuration time if a different version is found then ?

chapulina, 2021-08-05:
Yup, I'm surprised it didn't. There must be something weird going on in [FindIgnOGRE](https://github.com/ignitionrobotics/ign-cmake/blob/ign-cmake2/cmake/FindIgnOGRE.cmake)

traversaro, 2021-08-06:
I may be wrong, but I guess that `1.8.0` means `>= 1.80 and < 2.0`, see https://github.com/ignitionrobotics/ign-cmake/blob/ign-cmake2/cmake/FindIgnOGRE.cmake#L94 . In fact, I don't think there is any environment in which ignition-rendering is built with Ogre 1.8.0. As far as I know, on apt-based distributions and Homebrew Ogre 1.9 is used (see https://repology.org/project/ogre/versions), while in conda-forge at the moment we  are building ignition-rendering (even if a newer version) with both Ogre 1.10 and 1.12, see https://github.com/conda-forge/libignition-rendering4-feedstock/pull/8 .

I noticed that in conda-forge we have a local patch (https://github.com/conda-forge/libignition-rendering4-feedstock/blob/master/recipe/ogre-version.patch) to correctly compile with Ogre 1.10, that I think is inspired by a comment by @diegoferigo in https://github.com/robotology/gym-ignition/issues/279#issuecomment-776264280 . Probably we just need to get that patch upstream to solve this issue, sorry for not submiting it before!