gazebosim/gz-rendering #400: Ubuntu CI not finding Ogre 1.9

chapulina, 2021-09-09:
Our CI is not finding Ogre 1.9, it may be related to https://github.com/ignitionrobotics/ign-rendering/pull/376

## Environment
* OS Version:  Ubuntu Bionic or Focal
* Source or binary build? Source, on CI
  
  

## Description
* Expected behavior: CI runs for both Ogre 1 and 2
* Actual behavior: CI is only finding Ogre 2

## Steps to reproduce

Run CI on Jenkins or GitHub actions. Jenkins is yellow because of the warning, but Actions doesn't fail on warnings, so it shows up green.

## Output

Ogre 1.9 is installed:

```
2021-09-09T00:23:39.9181608Z Selecting previously unselected package libogre-1.9.0v5:amd64.
2021-09-09T00:23:39.9217794Z Preparing to unpack .../270-libogre-1.9.0v5_1.9.0+dfsg1-10_amd64.deb ...
2021-09-09T00:23:39.9329386Z Unpacking libogre-1.9.0v5:amd64 (1.9.0+dfsg1-10) ...
2021-09-09T00:23:40.1933838Z Selecting previously unselected package libogre-1.9-dev.
2021-09-09T00:23:40.1977322Z Preparing to unpack .../271-libogre-1.9-dev_1.9.0+dfsg1-10_amd64.deb ...
2021-09-09T00:23:40.1989527Z Unpacking libogre-1.9-dev (1.9.0+dfsg1-10) ...
```

But not found:

```
2021-09-09T00:23:47.9754351Z    CONFIGURATION WARNINGS:
2021-09-09T00:23:47.9755220Z    -- Missing dependency [IgnOGRE] (Components: RTShaderSystem, Terrain, Overlay)
2021-09-09T00:23:47.9756184Z    -- Skipping component [optix]: Missing dependency [OptiX].
2021-09-09T00:23:47.9756851Z       ^~~~~ Set SKIP_optix=true in cmake to suppress this warning.
```

Example builds:

* [Jenkins](https://build.osrfoundation.org/job/ignition_rendering-ci-pr_any-ubuntu_auto-amd64/1732/consoleFull)
* [Actions](https://pipelines.actions.githubusercontent.com/25Afi1eCTV82xlgZ37QhXw2tVikon6IYGpydIyZAynnfiBqqik/_apis/pipelines/1/runs/3423/signedlogcontent/5?urlExpires=2021-09-09T20%3A54%3A24.9484720Z&urlSigningMethod=HMACV1&urlSignature=WVbzxYWpJ7ayTrfoM%2BSwv9qiKq0xJztvwpdkv4wj%2BCk%3D)

Blast545, 2021-09-14:
I'm not a Cmake expert, but I think I know what's happening. 

https://github.com/ignitionrobotics/ign-cmake/blob/eac0b654a19dc67c580e8ceab494915efc6ab796/cmake/IgnUtils.cmake#L196-L209

This portion of code tries to find a IgnOgre, but when it fails to find it, it still tries to find the selected components here:
https://github.com/ignitionrobotics/ign-rendering/blob/49e7a06024a4adb7f2ec68cac1ebd33318cb9b8b/CMakeLists.txt#L64

Then, the components are marked as IgnOgre dependency is not available, even when IgnOgre is found below. This is confirmed by the Jenkins output:

```
-- Looking for IgnOGRE - not found

-- Checking for module 'OGRE-RTShaderSystem >= 1.9'
--   Found OGRE-RTShaderSystem , version 1.9.0
-- Checking for module 'OGRE-Terrain >= 1.9'
--   Found OGRE-Terrain , version 1.9.0
-- Checking for module 'OGRE-Overlay >= 1.9'
--   Found OGRE-Overlay , version 1.9.0
-- Looking for IgnOGRE - found

...

CMake Warning at /usr/share/cmake/ignition-cmake2/cmake2/IgnConfigureBuild.cmake:53 (message):
   CONFIGURATION WARNINGS:
   -- Missing dependency [IgnOGRE] (Components: RTShaderSystem, Terrain, Overlay)
Call Stack (most recent call first):
  CMakeLists.txt:156 (ign_configure_build)
```

I think this should be fixed by changing the statement so the components require IgnOgre, otherwise do not import them. @j-rivero

chapulina, 2021-11-17:
I believe this is fixed:

https://build.osrfoundation.org/job/ignition_rendering-ci-pr_any-ubuntu_auto-amd64/lastBuild/consoleFull#console-section-6

PR #411 (2021-09-16): Avoid configure warning when checking ogre-1.10
# 🦟 Bug fix

Fixes #400

## Summary

After #376, the check for Ogre-1.10 is generating warnings during the configuration of the software. The PR changes the behavior to do a fake check with `QUIET` to avoid warnings and a real check afterwards if the unofficial ogre version is used.

## Checklist
- [x] Signed all commits for DCO
- [x] All tests passed (See [test coverage](https://ignitionrobotics.org/docs/all/contributing#test-coverage))

**Note to maintainers**: Remember to use **Squash-Merge**
closed 2021-11-17 by chapulina