gazebosim/gz-rendering #402: Ogre2.2 errors when launching ign gazebo

jennuine, 2021-09-09:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
  
  
   Source, `main` branch, commit: 7def756623029ef168cf159147d4df39e702ae3b
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2.2
      - [X] GUI rendering error. 

When running `ign gazebo -v 4 shapes.sdf`, the following errors appear:

```bash
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Err] [Ogre2RenderEngine.cc:319] Failed to initialize render-engine
[GUI] [Dbg] [RenderUtil.cc:2267] Create scene [scene]
[GUI] [Err] [BaseRenderEngine.cc:215] Render-engine has not been initialized
[GUI] [Err] [MarkerManager.cc:156] Scene pointer is invalid
[GUI] [Err] [Scene3D.cc:2546] Unable to initialize renderer
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Wrn] [BaseRenderEngine.cc:42] Render-engine has already been loaded
[GUI] [Dbg] [RenderUtil.cc:2267] Create scene [scene]
Error at line 57: @piece 'FresnelSwizzle' already definedError at line 16: @piece 'getSpecularFresnel' already definedError at line 6: @piece 'NormalNonPremul' already definedError at line 6: @piece 'detail_swizzle0' already definedError at line 57: @piece 'FresnelSwizzle' already definedError at line 102: @piece 'FresnelSwizzle' already definedError at line 14: @piece 'getSpecularFresnel' already definedError at line 5: @piece 'NormalNonPremul' already definedError at line 6: @piece 'detail_swizzle0' already definedError at line 102: @piece 'FresnelSwizzle' already definedError at line 8: @piece 'SampleSpecularMap' already definedError at line 102: @piece 'FresnelSwizzle' already definedError at line 14: @piece 'getSpecularFresnel' already definedError at line 5: @piece 'NormalNonPremul' already definedError at line 6: @piece 'detail_swizzle0' already definedError at line 102: @piece 'FresnelSwizzle' already definedError at line 8: @piece 'SampleSpecularMap' already defined
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
```
---

Rendering system info:

 - Rendering system info:
```bash
LANG=C lspci -nn | grep VGA  
### Output
05:00.0 VGA compatible controller [0300]: NVIDIA Corporation TU116 [GeForce GTX 1660 SUPER] [10de:21c4] (rev a1)

echo "$DISPLAY"
### Output
:1

LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  
### Output
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: GeForce GTX 1660 SUPER/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 460.91.03
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 460.91.03
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 460.91.03
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

ps aux | grep Xorg
### Output
root        1345  0.0  0.1 177356 50636 tty1     Sl+  08:24   0:02 /usr/lib/xorg/Xorg vt1 -displayfd 3 -auth /run/user/125/gdm/Xauthority -background none -noreset -keeptty -verbose 3
root        2923  2.1  0.2 230076 98156 tty2     Sl+  08:27  10:09 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
osrf      105452  0.0  0.0   9040  2428 pts/4    S+   16:28   0:00 grep --color=auto --exclude-dir=.bzr --exclude-dir=CVS --exclude-dir=.git --exclude-dir=.hg --exclude-dir=.svn --exclude-dir=.idea --exclude-dir=.tox Xorg

sudo env LANG=C X -version
### Output
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Ubuntu
Current Operating System: Linux osrf-MS-7885 5.11.0-34-generic #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021 x86_64
Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.11.0-34-generic root=UUID=5f4a51f9-a8a6-41f8-8317-6d55c97f96e9 ro quiet splash vt.handoff=7
Build Date: 06 July 2021  10:17:51AM
xorg-server 2:1.20.11-1ubuntu1~20.04.2 (For technical support please see http://www.ubuntu.com/support) 
[... 607 lines cut ...]
16:17:00: Parsing script gaussian_noise.material
16:17:00: Parsing script depth_camera.material
16:17:00: Parsing script GaussianNoise.compositor
16:17:00: Parsing script PbsMaterials.compositor
16:17:00: Compiler error: invalid parameters in PbsMaterials.compositor(4): Unrecognized PixelFormat
16:17:00: Compiler error: invalid parameters in PbsMaterials.compositor(8): Unrecognized PixelFormat
16:17:00: OGRE EXCEPTION(5:ItemIdentityException): Can't find texture with name: '[Hash 0x05f70bcc]' If it's a global texture, trying to use it in the output channel will fail. in CompositorNodeDef::getTextureSource at /home/jenkins/workspace/ogre-2.2-debbuilder/repo/OgreMain/src/Compositor/OgreTextureDefinition.cpp (line 214)
16:17:00: Compiler error: invalid parameters in PbsMaterials.compositor(25)
16:17:00: Parsing scripts for resource group Internal
16:17:00: Finished parsing scripts for resource group Internal
16:17:00: Creating resources for group Internal
16:17:00: All done
16:17:00: Added resource location '/home/osrf/Workspaces/ignFortress_ws/install/share/ignition/ignition-rendering6/media/materials/textures/' of type 'FileSystem' to resource group 'General'
16:17:00: Can't assign material scene::Material(65506) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:00: Can't assign material scene::Material(65500) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:00: Can't assign material scene::Material(65494) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:00: Can't assign material scene::Material(65488) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:00: Can't assign material scene::Material(65481) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:00: Can't assign material scene::Material(65488) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:00: Can't assign material scene::Material(65468) because this Material does not exist. Have you forgotten to define it in a .material script?
16:17:01: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
16:17:01: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
16:17:01: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
16:17:01: WARNING: Deleting mapped buffer without having it unmapped. This is often sign of a resource leak or a bad pattern. Umapping the buffer for you...
</details>

jennuine, 2021-09-09:
Also, when I try to expand the GUI, it crashes with:

```
ign gazebo gui: /home/osrf/Workspaces/ignFortress_ws/src/ign-rendering/ogre2/src/Ogre2Scene.cc:270: void ignition::rendering::v6::Ogre2Scene::StartRendering(): Assertion `("Started rendering without first calling Scene::PreRender. " "See Scene::SetCameraPassCountPerGpuFlush for details", this->dataPtr->frameUpdateStarted == true)' failed.
[Dbg] [SignalHandler.cc:141] Received signal[2].
[Dbg] [ServerPrivate.cc:108] Server received signal[2]
[Dbg] [ign.cc:353] Shutting down ign-gazebo-server
[Dbg] [SimulationRunner.cc:543] Exiting postupdate worker thread (0)
```

iche033, 2021-09-10:
I ran into the issue with failing to initialize the engine a few times. We recently merged https://github.com/ignitionrobotics/ign-rendering/pull/272 that upgrades ogre 2.1 to 2.2. The error and crash for me was due to leftover ogre2.1 media files in the colcon `install` directory. Looking at the `ogre2.log`:

```
Compiler error: invalid parameters in PbsMaterials.compositor(4): Unrecognized PixelFormat
```

The [PbsMaterials.compositor](https://github.com/ignitionrobotics/ign-rendering/tree/ign-rendering5/ogre2/src/media/2.0/scripts/Compositors) exists in ign-rendering versions <5 but should be removed in the latest ign-rendering `main` branch.  So it's trying to load a ogre 2.1 script that does work in ogre 2.2. So it may worth doing a clean rebuild and see if the issue still persists.

jennuine, 2021-09-10:
Thanks, a clean rebuild of the entire workspace worked for me. Yesterday I had only rebuilt rendering.

closed 2021-09-10 by jennuine