gazebosim/gz-rendering #414: Mouse functionality for segmentation camera demo is incomplete

adlarkin, 2021-09-18:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
  Source: #329
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2

## Description
There are several mouse features that don't work with the segmentation camera demo:
1. Ray query code causes a crash: https://github.com/ignitionrobotics/ign-rendering/blob/e185ecd74c3dd979a06740d54fd010bf146eb15b/examples/segmentation_camera/GlutWindow.cc#L160
2. Mouse wheel scrool zoom doesn't work: https://github.com/ignitionrobotics/ign-rendering/blob/e185ecd74c3dd979a06740d54fd010bf146eb15b/examples/segmentation_camera/GlutWindow.cc#L170-L171
3. Right mouse button zoom doesn't work: https://github.com/ignitionrobotics/ign-rendering/blob/e185ecd74c3dd979a06740d54fd010bf146eb15b/examples/segmentation_camera/GlutWindow.cc#L206-L207

## Steps to reproduce

1. build the segmentation camera demo example:
```
cd ign-rendering/examples/segmentation_camera
mkdir build
cd build
cmake .. && make
```
2. Run the example and try to use the scroll wheel or right click button on your mouse (they won't work):
```
./segmentation_camera
```
3. Try to uncomment the ray query code (point 1 in the `Description` section above) and the rebuild/re-test the example. A crash occurs when you click on the GUI display

darksylinc, 2021-09-19:
If you post a callstack of the crash to the rayquery:

```cpp
    // TODO(anyone) figure out why this code is causing a crash
    // g_rayQuery->SetFromCamera(rayCamera, ignition::math::Vector2d(nx, ny));
    // g_target  = g_rayQuery->ClosestPoint();
    // if (!g_target)
    // {
    //   // set point to be 10m away if no intersection found
    //   g_target.point = g_rayQuery->Origin() + g_rayQuery->Direction() * 10;
    //   return;
    // }
```

I might be able to spot what's going on, or at least give you a hint.

adlarkin, 2021-09-23:
> If you post a callstack of the crash to the rayquery

If I uncomment the code you highlighted in your comment and then run the example again, I get a crash with the only output being this:
```
Segmentation fault (core dumped)
```

It looks like it's coming from this line: https://github.com/ignitionrobotics/ign-rendering/blob/d2ff058e0b93d61cec514ead5293f6f53c2c8597/examples/segmentation_camera/GlutWindow.cc#L161

Doing some more digging, I find that the segfault actually occurs here: https://github.com/ignitionrobotics/ign-rendering/blob/d2ff058e0b93d61cec514ead5293f6f53c2c8597/ogre2/src/Ogre2RayQuery.cc#L67

It looks like the input camera pointer (`_camera`) is not null, but when it's converted to an `Ogre2Camera`, it is null. So, this must mean that the input `_camera` is not of type `Ogre2Camera`? Looking at the example code, the segmentation camera is [created here](https://github.com/ignitionrobotics/ign-rendering/blob/d2ff058e0b93d61cec514ead5293f6f53c2c8597/examples/segmentation_camera/Main.cc#L107). Is there something wrong with the inheritance structure of the segmentation camera?

iche033, 2021-09-23:
```
Ogre2SegmentationCamera :
      public BaseSegmentationCamera<Ogre2Sensor>
```
oh looks like it inherits from Ogre2Sensor and not Ogre2Camera so the cast fails.

adlarkin, 2021-09-23:
Would it be possible to change the inheritance template type to `Ogre2Camera` without breaking ABI since ogre2 is a separate component/plugin?

Also, I believe that mouse functionality with the ray query for the example in https://github.com/ignitionrobotics/ign-rendering/pull/334 works just fine (at least, it did last time I tried it) and it looks like the inheritance structure is similar to the segmentation camera :thinking:

iche033, 2021-09-23:
> Would it be possible to change the inheritance template type to Ogre2Camera without breaking ABI since ogre2 is a separate component/plugin?

we are installing the ogre2 comopnent headers so that'll break ABI.
 
looks like the bounding box demo creates 2 cameras: 1 regular camera, and 2 bounding box camera. Ray queries are created by the regular camera. We could probably do the same workaround for the segmentation camera demo

azeey, 2024-05-31:
Fixed by https://github.com/gazebosim/gz-rendering/pull/483

PR #483 (2021-11-05): Enable ray query for depth, segmentation and thermal cameras
# 🦟 Bug fix

Fixes #414

## Summary

This PR enables depth, segmentation and thermal cameras to set a ray query for the `ogre` and `ogre2` render engines.

The approach Introduces mix-in classes to provide access to underlying Ogre objects in the `OgreCamera` / `Ogre2Camera` and related camera classes.

Update the mouse handling in the `segmentation_camera` and `thermal_camera examples`.

[image]

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [x] Updated documentation (as needed)
- [x] Updated migration guide (as  [... cut]
closed 2024-05-31 by azeey