gazebosim/gz-rendering #427: RenderOrder in Simple Demo is not working (ogre2)

iche033, 2021-09-24:
## Environment
* OS Version: 18.04
* Source or binary build?
source build - `main` branch, commit d2ff058e0b93d61cec514ead5293f6f53c2c8597

* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2]. 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on real hardware

    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils

          00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:191b] (rev 06)
          01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM204M [GeForce GTX 980M] [10de:13d7] (rev a1)

          echo "$DISPLAY" 

          :0

          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package

          direct rendering: Yes
          OpenGL vendor string: NVIDIA Corporation
          OpenGL renderer string: GeForce GTX 980M/PCIe/SSE2
          OpenGL core profile version string: 4.6.0 NVIDIA 390.132
          OpenGL core profile shading language version string: 4.60 NVIDIA
          OpenGL core profile context flags: (none)
          OpenGL core profile profile mask: core profile
          OpenGL version string: 4.6.0 NVIDIA 390.132
          OpenGL shading language version string: 4.60 NVIDIA
          OpenGL context flags: (none)
          OpenGL profile mask: (none)
          OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 390.132
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

        
    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
  
```
11:52:04: Creating resource group General
11:52:04: Creating resource group Internal
11:52:04: Creating resource group Autodetect
11:52:04: SceneManagerFactory for type 'DefaultSceneManager' registered.
11:52:04: Registering ResourceManager for type Material
11:52:04: Registering ResourceManager for type Mesh
11:52:04: Registering ResourceManager for type Mesh2
11:52:04: Registering ResourceManager for type OldSkeleton
11:52:04: MovableObjectFactory for type 'ParticleSystem' registered.
11:52:04: ArchiveFactory for archive type FileSystem registered.
11:52:04: ArchiveFactory for archive type Zip registered.
11:52:04: ArchiveFactory for archive type EmbeddedZip registered.
11:52:04: DDS codec registering
11:52:04: FreeImage version: 3.17.0
11:52:04: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
11:52:04: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
11:52:04: OITD codec registering
11:52:04: Registering ResourceManager for type HighLevelGpuProgram
11:52:04: MovableObjectFactory for type 'Decal' registered.
11:52:04: MovableObjectFactory for type 'InternalCubemapProbe' registered.
11:52:04: MovableObjectFactory for type 'Entity' registered.
11:52:04: MovableObjectFactory for type 'Item' registered.
11:52:04: MovableObjectFactory for type 'Light' registered.
11:52:04: MovableObjectFactory for type 'Rectangle2Dv2' registered.
11:52:04: MovableObjectFactory for type 'BillboardSet' registered.
11:52:04: MovableObjectFactory for type 'ManualObject2' registered.
[... 571 lines cut ...]
11:52:04: Can't assign material scene::Material(65495) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65495) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65484) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65480) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65476) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65484) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65480) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65476) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65484) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65480) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65476) because this Material does not exist. Have you forgotten to define it in a .material script?
11:52:04: Can't assign material scene::Material(65448) because this Material does not exist. Have you forgotten to define it in a .material script?
```

</details>

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1. Follow the simple demo tutorial: https://ignitionrobotics.org/api/rendering/6.0/simple_demo.html
2. Run with ogre2, `./simple_demo ogre2`
3. You will notice that the red 2D rectangle flashes due to depth fighting issue. Expected behavior should be that the red rectangle does not flash and renders over the white ground plane

iche033, 2021-12-03:
@darksylinc I could use some advice on this issue. We are setting `Ogre::HlmsMacroblock`'s `mDepthBiasConstant` reduce z-fighting for coplanar polygons:
https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/ogre2/src/Ogre2Material.cc#L283

For the simple_demo example, [a small value](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/examples/simple_demo/Main.cc#L90) used to work in ogre 2.1 but with ogre 2.2 I had to increase the value by a factor of ~100 for it to take effect. I think this is because ogre 2.2 is now using reverse-z?  

Since the rendered result depends on a number of factors like distance to object and camera near and far planes, do you think if there is a way to preserve existing behavior of our API call, i.e. let user specify a small / or similar value while still keeping reverse-z in ogre 2.2?

darksylinc, 2021-12-03:
I freaked out for a second because I thought you were comparing 1.x vs 2.x but turns out it's just 2.1 vs 2.2

Phew!

>  I think this is because ogre 2.2 is now using reverse-z?

I think that is an excellent guess! You can test this theory by doing in `Ogre2RenderEngine::CreateRenderWindow`:

```cpp
params.["reverse_depth"] = "No" ;
```

The other possibility is that it's D32_FLOAT vs D24_UNORM difference, since bias is calculated differently depending on whether it's a FLOAT or UNORM buffer (we have support for slope bias but I don't think ign is using it)

```
Unorm: bias = depthBias * r + slopeScale * maxSlope
Float: bias = depthBias * 2^(exp(max abs(z) in primitive) - r) + slopeScale * maxSlope

// r is the smallest positive representable value in the depth render target
```

> Since the rendered result depends on a number of factors like distance to object and camera near and far planes, do you think if there is a way to preserve existing behavior of our API call, i.e. let user specify a small / or similar value while still keeping reverse-z in ogre 2.2?

I'll have to see it for myself to see what's going wrong.

Usually depth bias is a hack to avoid Z-fighting and simply activating reverse depth means the hack can be deactivated because it stops fighting.

But since this is Linux, I wouldn't be surprised if you got into worst case scenario:

- Reverse Depth makes sense when using D32_FLOAT, not when using the UNORM variants
  - When Reverse Depth is used on D32_FLOAT you get a massive boost in depth precision
  - When Reverse Depth is used on D24_UNORM, you actually get slightly worse depth precision than normal D24_UNORM.
- X11 doesn't support D32_FLOAT render _windows_ (or if it does, I don't know how to get it working); so it uses D24_UNORM
- GL3+ will turn reverse depth on based on GPU support. But on X11 that means you may end up using D24_UNORM + reverse_depth **which is worse**; instead of the intended D32_FLOAT + reverse_depth combo
  - Note this bug only affects when you're rendering directly to the screen
  - Note ign-gui should not have this bug because Ogre is rendered to a RenderTexture, and then the results sent to Qt as an image to be presented on screen
- Vulkan, Metal & D3D11 don't have this problem because the depth buffer used does not depend on the OS Windowing System
- A workaround to avoid this bug would be to always render to a RenderTexture on GL3+, then just copy to the window

I'd have to see it for myself, because applying depth bias can be for specific reasons, or simply to avoid Z fighting. If the latter, when reverse depth is being used then the need for a depth bias almost always disappears.

In any case though, it means from your perspective the API behavior is different (i.e. the depth bias is unwanted). Although I suspect you may be hitting D24_UNORM + reverse_depth which could explain why the 100x change in unit of scale.

iche033, 2021-12-03:
Just did a quick test setting `params["reverse_depth"] = "No" ;` and z-fighting is gone. Our use case is 2 overlapping planes with the same z pos, e.g. `[x, y, 0]`. We like to have one object rendered over another as a quick and easy way to do decals on a plane and avoid z-fighting.

Yeah we are rendering to a RenderTexture on GL3+. It's indeed the render texture used by ign-gui. As for `D32_FLOAT` vs `D24_UNORM`, we didn't specify the format when [creating the TextureGpu](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/ogre2/src/Ogre2RenderTarget.cc#L522) so not sure what OGRE uses underneath. I still need to look into this. Thanks for the pointers.

darksylinc, 2021-12-11:
I could not find info online about this problem, so i had to write a test myself.

The obvious solution is to multiply the depth bias by 100 or so; but I was wondering what would be the side effects on other geometry. I.e. what would happen to geometry that is meant to be in front of the plane? We know it works at close range, but would the plane get on top of everything as you move the camera away?

So [I made a test](https://www.youtube.com/watch?v=s2XdH3fYUac)!

Turns out we can safely multiply by something like 100 or so, and we will get **better** results with reverse Z.

So from ignition-rendering's end we just need to do:

```cpp
if( renderSystem->isReverseDepth() )
  depthBias *= 100.0f;
```

PR #514 (2021-12-13): RenderOrder is in different scale in Reverse Z
I could not find info online about this problem, so i had to write a
test myself.

The obvious solution is to multiply the depth bias by 100 or so; but I
was wondering what would be the side effects on other geometry.

I.e. what would happen to geometry that is meant to be in front of the
plane? We know a depth bias of 300 works at close range at reverse Z
(where at normal Z a depth bias of 3 was enough), but would the plane
get on top of everything as you move the camera away?

So I made a test: https://www.youtube.com/watch?v=s2XdH3fYUac

We can [... cut]
commit d06d2d45 (2021-12-13): RenderOrder is in different scale in Reverse Z (#514)

I could not find info online about this problem, so i had to write a
test myself.

The obvious solution is to multiply the depth bias by 100 or so; but I
was wondering what would be the side effects on other geometry.

I.e. what would happen to geometry that is meant to be in front of the
plane? We know a depth bias of 300 works at close range at reverse Z
(where at normal Z a depth bias of 3 was enough), but would the plane
get on top of everything as you move the camera away?

So I made a test: https://www.youtube.com/watch?v=s2XdH3fYUac [... cut]
closed 2021-12-13 by iche033