gazebosim/gz-rendering #432: GazeboSceneViewer is broken

iche033, 2021-09-24:
## Environment
* OS Version: Ubuntu 18.08
* Source or binary build?
 ign-rendering6 pre2 (`main` branch, commit d2ff058e0b93d61cec514ead5293f6f53c2c8597)

  

Here's the backtrace:

<details>
  
```
Thread 1 "gazebo_scene_vi" received signal SIGSEGV, Segmentation fault.
0x00007fffee3772a8 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > >, std::_Select1st<std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > > >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > > > >::find (
    __k=<synthetic pointer>: 0, this=0x30) at /usr/include/c++/8/bits/stl_tree.h:2556
2556	    _Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::
(gdb) bt
#0  0x00007fffee3772a8 in std::_Rb_tree<unsigned int, std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > >, std::_Select1st<std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > > >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > > > >::find(unsigned int const&) const (__k=<synthetic pointer>: 0, this=0x30) at /usr/include/c++/8/bits/stl_tree.h:2556
#1  0x00007fffee3772a8 in std::map<unsigned int, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > >, std::less<unsigned int>, std::allocator<std::pair<unsigned int const, std::vector<ignition::math::v6::Vector2<double>, std::allocator<ignition::math::v6::Vector2<double> > > > > >::find(unsigned int const&) const (__x=<synthetic pointer>: 0, this=0x30)
    at /usr/include/c++/8/bits/stl_map.h:1194
#2  0x00007fffee3772a8 in ignition::common::SubMesh::TexCoordCountBySet(unsigned int) const (this=this@entry=0x7fffffffcd50, _setIndex=_setIndex@entry=0)
    at /home/osrf/code/ign_f_ws2/src/ign-common/graphics/src/SubMesh.cc:457
#3  0x00007fffc320e9a5 in ignition::rendering::v6::OgreMeshFactory::LoadImpl(ignition::rendering::v6::MeshDescriptor const&) (this=0x55555606e5f0, _desc=...)
    at /home/osrf/code/ign_f_ws2/src/ign-rendering/ogre/src/OgreMeshFactory.cc:303
#4  0x00007fffc320d7f4 in ignition::rendering::v6::OgreMeshFactory::OgreEntity(ignition::rendering::v6::MeshDescriptor const&) (this=0x55555606e5f0, _desc=...)
    at /home/osrf/code/ign_f_ws2/src/ign-rendering/ogre/src/OgreMeshFactory.cc:76
#5  0x00007fffc32129a0 in ignition::rendering::v6::OgreMeshFactory::Create(ignition::rendering::v6::MeshDescriptor const&) (this=0x55555606e5f0, _desc=...)
    at /home/osrf/code/ign_f_ws2/src/ign-rendering/ogre/src/OgreMeshFactory.cc:59
#6  0x00007fffc32293d0 in ignition::rendering::v6::OgreScene::CreateMeshImpl(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ignition::rendering::v6::MeshDescriptor const&) (this=0x555555b03140, _id=65524, _name="scene::Cylinder(65524)", _desc=...)
    at /usr/include/c++/8/bits/shared_ptr_base.h:1018
#7  0x00007fffc322891f in ignition::rendering::v6::OgreScene::CreateMeshImpl(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (this=0x555555b03140, _id=65524, _name="scene::Cylinder(65524)", _meshName=...) at /home/osrf/code/ign_f_ws2/src/ign-rendering/ogre/src/OgreScene.cc:538
#8  0x00007fffc322a2f7 in ignition::rendering::v6::OgreScene::CreateCylinderImpl(unsigned int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (this=0x555555b03140, _id=65524, _name="scene::Cylinder(65524)") at /usr/include/c++/8/bits/basic_string.h:252
#9  0x00007ffff5b7e47e in ignition::rendering::v6::BaseScene::CreateCylinder() (this=0x555555b03140)
    at /home/osrf/code/ign_f_ws2/src/ign-rendering/src/base/BaseScene.cc:1156
#10 0x0000555555585eef in ignition::rendering::SubSceneManager::ProcessCylinder(gazebo::msgs::Geometry const&, std::shared_ptr<ignition::rendering::v6::Visual>) ()
#11 0x0000555555585a57 in ignition::rendering::SubSceneManager::ProcessGeometry(gazebo::msgs::Geometry const&, std::shared_ptr<ignition::rendering::v6::Visual>) ()
#12 0x0000555555585002 in ignition::rendering::SubSceneManager::ProcessVisual(gazebo::msgs::Visual const&, std::shared_ptr<ignition::rendering::v6::Visual>) ()
```

</details>

## Description
* Expected behavior: run gazebo_scene_viewer and the window shows up 
* Actual behavior: gazebo_scene_viewer crashes

## Steps to reproduce

1. follow the tutorial: https://ignitionrobotics.org/api/rendering/6.0/gazebo_scene_viewer.html
2. demo crashes when running `./gazebo_scene_viewer`

mayman99, 2021-11-06:
Hi Ian,
I tried to solve this one ... so somehow I figured out that the problem arises when linking with `${GAZEBO_LIBRARIES}`. 
You can confirm this by adding `${GAZEBO_LIBRARIES}` to the `CMakeLists.txt` of an example like `simple_demo` (it will break as well).
Do you perhaps have an idea of what could be happening?

iche033, 2021-11-08:
I think the problem is that gazebo has its own set of ignition dependencies with versions that are different from the ones in ign-rendering, e.g. gazebo11 depends on ign-common3 while ign-rendering depends on ign-common4, and this causes weird issues when the same library of different versions are loaded at the same time. I'm not sure what the best solution is though.

mayman99, 2021-11-08:
Yeah, this is most probably the issue here.
I'll probably think about it for a bit, if I came across a good solution I'll share it then.

42656E4B, 2022-09-27:
Are there any updates on this issue?

iche033, 2022-09-27:
no updates on this

iche033, 2023-10-06:
the gazebo_scene_viewer example is removed in harmonic.

closed 2023-10-06 by iche033