gazebosim/gz-rendering #435: Ogre2: Implement Global Illumination using VCT

darksylinc, 2021-09-26:
Note this ticket is for tracking my work. I'm the one implementing it.

The work can currently be found in [matias-global-illumination](https://github.com/darksylinc/ign-rendering/tree/matias-global-illumination) branch

## Desired behavior

Obtain realtime Global Illumination when using the Ogre2 engine

Ogre2 provides various [GI methods](https://ogrecave.github.io/ogre-next/api/2.3/_gi_methods.html), out of which VCT (Voxel Cone Tracing) is the most reliable and accurate one for simulations.

The class hierarchy is the following:

 - GlobalIlluminationBase
   - GlobalIlluminationVct
     - BaseGlobalIlluminationVct (ign-rendering abstraction / implementation detail)
       - Ogre2GlobalIlluminationVct

The reason for having `GlobalIlluminationBase` is that there may be multiple solutions implemented in the future; including raytracing

## What about render engines that have natural GI (e.g. OptiX)

It is unclear. Technically speaking `GlobalIlluminationBase` is an object where users can specify GI parameters. Users can create more than one if they wish to use different parameters, but only one can be active at the same time.

Right now `GlobalIlluminationBase` only contains simple properties such as `BounceCount`. Probably render engines with natural GI like ray and path tracers could move their bounce count settings to this class.

## Remaining tasks

 - [x] Write GI/VCT code
 - [ ] Light changes should notify GI to recalculate lighting
 - [ ] Make sample
 - [ ] Make unit test

darksylinc, 2021-10-02:
Quack quack

[image]

(voxelized version of the ogre2_demo example, which is the first step to get VCT lighting right)

darksylinc, 2021-10-03:
Ok it's definitely working, you can see the reflections:
[image]
[image]
[image]

And the GI contribution (see the green tint on the yellow duck):
[image]

Three things I'm noticing:
 - GI reflections break a a bit when the voxel resolution is non-square. Looks like an Ogre bug
 - Voxelization when octant auto division isn't `{ 1, 1, 1 }` is broken. This is definitely an Ogre bug
 - The example has 1 directional light and 12 additional lights. Because HDR is not being used, each spot/point light is just as powerful as the sun; thus their GI contributions blow out of proportion (*)

(*) This made me remember that VCT was thought mainly for "sun light" GI contribution, but it supports all types. To support HDR it would have to use `RGBA16_FLOAT` targets for the voxel lighting, which is 2x as expensive (in VRAM cost) than the default `RGBA8_UNORM` target.

We don't even support switching to `RGBA16_FLOAT` because that's a lot of memory and there was no need (it's easy to support, just a few more lines of code). But if you care about simulation accuracy and you have a monster GPU with 16-24GB of VRAM, I guess you have the luxury of not caring.

iche033, 2021-10-05:
> But if you care about simulation accuracy and you have a monster GPU with 16-24GB of VRAM, I guess you have the luxury of not caring.

hmm that sounds a little too memory intensive. I would hold off on supporting HDR for now until there is a need for it. 

The 12 lights were added to the ogre2 demo for testing a while back. Now I think about it, I wonder if we should just have a simple Cornell box environment to demo this feature :)

darksylinc, 2021-10-05:
> hmm that sounds a little too memory intensive. I would hold off on supporting HDR for now until there is a need for it.

OK without explanation that feels a little bit overblown. It's all about user settings and what they consider good enough.

We keep 4 voxels around (though we can reduce it to 1 if lights are never updated):

1. Albedo, RGBA8_UNORM, 4 bytes
2. Normals RGBA8_SNORM, 4 bytes
3. Emissive RGBA8_UNORM, 4 bytes (this one probably can go away if we don't allow emissive materials to emit light in GI)
4. Light calculation. RGBA8_UNORM, 4 bytes

So that's 12 bytes per voxel.

If we go to HDR, we'd need:

1. Albedo, RGBA8_UNORM, 4 bytes
2. Normals RGBA8_SNORM, 4 bytes
3. Emissive RGBA16_HALF, 8 bytes (this one probably can go away, or be kept at 4 bytes and hack a multiplier)
4. Light calculation. RGBA16_HALF, 8 bytes

So that's between 16 & 24 bytes per voxel.

A (medium-sized?) scene the user may want to use 1024x1024x32. Maybe less would be enough, maybe more. It depends on what the simulation expects.

So: 1024x1024x32 x (16|24) = between 512MB & 768MB of VRAM + mipmaps (and more if we turn on anisotropic, I can't remember how that was calculated). Mipmaps add a 1.15x overhead, so you actually need 588-883MB

If the user thinks, this is not enough, I need 4096x4096x64 then cost grows dramatically, between 16GB and 24GB (on a 24GB GPU congrats, you've ran out of memory, can't be done).

But a compromise 2048x2048x64: between 4GB and 6GB (+1.15x in mipmaps).

So how much memory you'll need depends on how much the user thinks he needs for his scene. If he's already thankful he has GI at all (i.e. vs having nothing) and considers 128x128x32 enough, then of course you need no monster GPU.

Note that it is perfectly valid to use one setting (e.g. 128x128x32) for real time preview on your laptop, and then go overboard with a different setting when you need accurate results in the simulation on a powerful workstation.

PR #675 (2022-12-22): Global illumination VCT & CI VCT
# 🦟 Bug fix

Fixes #435 

## Summary

This PR adds Real Time Global Illumination based on one of these two techniques:

 1. VCT (Voxel Cone Tracing)
 2. CI VCT (Cascaded Image Voxel Cone Tracing)

VCT can be more accurate but likes static objects (static objects can bounce GI, while dynamic objects can only receive).

CI VCT can be less accurate, but it is much faster dealing with objects and deal with very large scenes (and adapt more easily to any of them) and even though it prefers static objects, you can force all objects to bounce GI and it may just work.  [... cut]