gazebosim/gz-rendering #439: Ogre2: heightmap crashes if there is only one shadow casting spotlight in the scene

iche033, 2021-09-28:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
  ign-rendering6 source build
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre2]. 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [x] running on real hardware

    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
  
```
14:54:33: Creating resource group General
14:54:33: Creating resource group Internal
14:54:33: Creating resource group Autodetect
14:54:33: SceneManagerFactory for type 'DefaultSceneManager' registered.
14:54:33: Registering ResourceManager for type Material
14:54:33: Registering ResourceManager for type Mesh
14:54:33: Registering ResourceManager for type Mesh2
14:54:33: Registering ResourceManager for type OldSkeleton
14:54:33: MovableObjectFactory for type 'ParticleSystem' registered.
14:54:33: ArchiveFactory for archive type FileSystem registered.
14:54:33: ArchiveFactory for archive type Zip registered.
14:54:33: ArchiveFactory for archive type EmbeddedZip registered.
14:54:33: DDS codec registering
14:54:33: FreeImage version: 3.17.0
14:54:33: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
14:54:33: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
14:54:33: OITD codec registering
14:54:33: Registering ResourceManager for type HighLevelGpuProgram
14:54:33: MovableObjectFactory for type 'Decal' registered.
14:54:33: MovableObjectFactory for type 'InternalCubemapProbe' registered.
14:54:33: MovableObjectFactory for type 'Entity' registered.
14:54:33: MovableObjectFactory for type 'Item' registered.
14:54:33: MovableObjectFactory for type 'Light' registered.
14:54:33: MovableObjectFactory for type 'Rectangle2Dv2' registered.
14:54:33: MovableObjectFactory for type 'BillboardSet' registered.
14:54:33: MovableObjectFactory for type 'ManualObject2' registered.
14:54:33: MovableObjectFactory for type 'BillboardChain' registered.
14:54:33: MovableObjectFactory for type 'RibbonTrail' registered.
14:54:33: MovableObjectFactory for type 'WireAabb' registered.
14:54:33: *-*-* OGRE Initialising
14:54:33: *-*-* Version 2.2.6 (Cerberus)
14:54:33: OverlayElementFactory for type Panel registered.
14:54:33: OverlayElementFactory for type BorderPanel registered.
14:54:33: OverlayElementFactory for type TextArea registered.
14:54:33: Registering ResourceManager for type Font
14:54:33: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/RenderSystem_GL3Plus.so
14:54:33: Installing plugin: GL 3+ RenderSystem
14:54:33: OpenGL 3+ Rendering Subsystem created.
14:54:33: Plugin successfully installed
14:54:33: Loading library /usr/lib/x86_64-linux-gnu/OGRE-2.2/OGRE/Plugin_ParticleFX.so
14:54:33: Installing plugin: ParticleFX
14:54:33: Particle Emitter Type 'Point' registered
14:54:33: Particle Emitter Type 'Box' registered
14:54:33: Particle Emitter Type 'Ellipsoid' registered
14:54:33: Particle Emitter Type 'Cylinder' registered
14:54:33: Particle Emitter Type 'Ring' registered
14:54:33: Particle Emitter Type 'HollowEllipsoid' registered
14:54:33: Particle Affector Type 'LinearForce' registered
14:54:33: Particle Affector Type 'ColourFader' registered
14:54:33: Particle Affector Type 'ColourFader2' registered
14:54:33: Particle Affector Type 'ColourImage' registered
14:54:33: Particle Affector Type 'ColourInterpolator' registered
14:54:33: Particle Affector Type 'Scaler' registered
14:54:33: Particle Affector Type 'Rotator' registered
14:54:33: Particle Affector Type 'DirectionRandomiser' registered
[... 550 lines cut ...]

I noticed that it does not crash if there is also a directional light in the scene or if I turn off the spot light shadows

* Expected behavior: modified ogre2 heightmap demo runs fine
* Actual behavior: demo crashes

## Steps to reproduce
1. checkout the `heightmap_spotlight` branch
2. build and run `heightmap` demo
3. demo crashes

## Output

```
terminate called after throwing an instance of 'Ogre::RenderingAPIException'
  what():  OGRE EXCEPTION(3:RenderingAPIException): Fragment Program 100000000PixelShader_ps failed to compile. See compile log above for details. in GLSLShader::compile at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/GLSL/OgreGLSLShader.cpp (line 314)
Aborted (core dumped)
```

These lines in ogre2.log look suspicious:

```
14:54:33: GLSL compile log: 100000000PixelShader_ps
0(916) : error C1008: undefined variable "fShadow"
```

darksylinc, 2021-10-02:
I just saw this ticket.

Yeah the error in the log is the cause of the problem, at least the most immediate one: just bad shader codegen. This is easy to fix.

iche033, 2021-10-07:
fixed in #451