gazebosim/gz-rendering #444: Heightmaps don't work properly with segmentation camera

adlarkin, 2021-09-30:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
  Source: https://github.com/ignitionrobotics/ign-rendering/commit/eb7d6eaa5c6129fd76acbe222f6ca8655599342a

## Description
* Expected behavior: Heightmaps shouldn't be included in the segmentation output if they don't have a label.
* Actual behavior: Heightmaps can be seen if they aren't labeled.

A temporary fix that disables heightmap from the segmentation camera completely has been made in #443, but we should implement a solution that allows users to apply a label to the heightmap if they desire.

## Steps to reproduce

1. Run a world with a heightmap and semantic segmentation sensor in `ign-gazebo` (the fortress demo world has both)
2. View the segmentation output in the gazebo GUI by viewing the relevant colored map segmentation topic
3. You'll see a part of the heightmap in the segmentation output

## Output
[image]

darksylinc, 2021-12-06:
Hi!

This is with Ogre2 engine?

mjcarroll, 2021-12-06:
> This is with Ogre2 engine?

Yes, this was discovered in preparation for the fortress/ign-rendering6 release.

PR #545 (2022-02-01): Implement Hlms custom pieces swap for Objects & Terrain
# 🦟 Bug fix

Fixes #504
Fixes #444

## Summary

The 3rd and final PR to ticket #504 !

After this PR:

 - Terra now works with all Cameras (Thermal, SelectionBuffer, Segmentation, Laser) as documented in ticket #444 (backporting this to Fortress w/out breaking ABI could be a ton of work unfortunately)
 - Fixes various misc bugs of wrong values used in these cameras by regular objects when their properties weren't set (documented in #521, technically it fixes that one, but the fix can be backported so the ticket can will open until these v [... cut]
closed 2022-11-07 by iche033