gazebosim/gz-rendering #488: ABI checker failing to compile for ign-rendering6

chapulina, 2021-11-05:
## Environment
* OS Version:  Linux
* Source or binary build? Source
 
## Description
* Expected behavior:  The [ABI checker](https://build.osrfoundation.org/job/ignition_rendering-abichecker-any_to_any-ubuntu_auto-amd64/) compiles successfully
* Actual behavior:  It's failing to compile

## Steps to reproduce

Run the ABI checker, for example: https://build.osrfoundation.org/job/ignition_rendering-abichecker-any_to_any-ubuntu_auto-amd64/810/console

## Output

From https://build.osrfoundation.org/job/ignition_rendering-abichecker-any_to_any-ubuntu_auto-amd64/810/console:

```
ERROR: some errors occurred when compiling headers
ERROR: see log for details:
  /home/jenkins/workspace/ignition_rendering-abichecker-any_to_any-ubuntu_auto-amd64/abi_checker/logs/ign-rendering/X/log.txt
```

From https://build.osrfoundation.org/job/ignition_rendering-abichecker-any_to_any-ubuntu_auto-amd64/ws/abi_checker/logs/ign-rendering/X/log.txt:

```
The GCC parameters:
  gcc -fdump-lang-raw -fkeep-inline-functions -c -x c++ -fpermissive -w  -std=c++17 "/tmp/MZxkwdymxg/dump1.h"  -I/usr/local/destination_branch/include/ignition/rendering6 -I/usr/include/OGRE-2.2 -I/usr/include/OGRE -I/usr/include/ignition/utils1 -I/usr/include/ignition/math6 -I/usr/include/ignition/common4 -I/usr/include/OGRE/Paging

In file included from /usr/local/destination_branch/include/ignition/rendering6/ignition/rendering/ogre2/Ogre2Includes.hh:86,
                 from /usr/local/destination_branch/include/ignition/rendering6/ignition/rendering/ogre2/Ogre2Camera.hh:24,
                 from /tmp/MZxkwdymxg/dump1.h:6:
/usr/include/OGRE-2.2/Hlms/Unlit/OgreHlmsUnlit.h:32:10: fatal error: OgreHlmsBufferManager.h: No such file or directory
 #include "OgreHlmsBufferManager.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

j-rivero, 2021-11-12:
The problem is that the path where the header is `/usr/include/OGRE-2.2/Hlms/Unlit/` is not being included in the paths being used by the abi-checker `-I/usr/local/destination_branch/include/ignition/rendering6 -I/usr/include/OGRE-2.2 -I/usr/include/OGRE -I/usr/include/ignition/utils1 -I/usr/include/ignition/math6 -I/usr/include/ignition/common4 -I/usr/include/OGRE/Paging`

It has been a bit of time since I don't look into the abi-checker to remember how it computes the include paths, we can workaround the problem by setting it in the XML configuration files generated in Jenkins.

chapulina, 2021-11-16:
Just saw a similar issue on `ign-gazebo`:

https://build.osrfoundation.org/job/ignition_gazebo-abichecker-any_to_any-ubuntu_auto-amd64/4420/console

j-rivero, 2021-11-18:
Fix for ign-rendering is in https://github.com/ignition-tooling/release-tools/pull/560

> Just saw a similar issue on `ign-gazebo`:
> 
> https://build.osrfoundation.org/job/ignition_gazebo-abichecker-any_to_any-ubuntu_auto-amd64/4420/console

Seems like a completely different problem:
```
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/QtCore:4,
                 from /usr/include/ignition/gui5/ignition/gui/qt.h:25,
                 from /usr/include/ignition/gui5/ignition/gui/Application.hh:25,
                 from /usr/local/destination_branch/include/ignition/gazebo5/ignition/gazebo/gui/Gui.hh:22,
                 from /usr/local/destination_branch/include/ignition/gazebo5/ignition/gazebo.hh:122,
                 from /tmp/y5ci5Z7P6j/dump1.h:3:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qglobal.h:1135:4: error: #error "You must build your code with position independent code if Qt was built with -reduce-relocations. " "Compile your code with -fPIC (-fPIE is not enough)."
 #  error "You must build your code with position independent code if Qt was built with -reduce-relocations. "\
    ^~~~~
```
Looking into it.

closed 2021-11-18 by j-rivero

j-rivero, 2021-11-22:
> Seems like a completely different problem:
>
> ```
> In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/QtCore:4,
>                  from [...]

Can not reproduce it, ign-gazebo5 branch seems not to suffer this problem and ign-gazebo6 branch seems to be broken currently https://build.osrfoundation.org/job/ignition_gazebo-abichecker-any_to_any-ubuntu_auto-amd64/4438/. I leave the release-tools branch `fix_abi_qt` ready.