gazebosim/gz-rendering #500: Camera HFOV always returns same value even if it changed

chapulina, 2021-12-03:
## Environment
* OS Version:  Ubuntu Focal
* Source or binary build? Source, `ign-rendering6`
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin:  Both ogre and ogre2

## Description
* Expected behavior: `Camera::HFOV` returns the current angle
* Actual behavior: Always returns the initial angle

## Steps to reproduce

On a Gazebo or GUI plugin, print the camera's HFOV at every frame. The number is constant.

However, looking at the 3D view, it looks like the HFOV varies a lot when the image size changes, but the VFOV remains about the same:

[image]

## Additional info

I believe what happens is that `ign-rendering`'s API exposes HFOV, but Ogre uses VFOV. So when the user sets HFOV, we store it, then calculate VFOV to pass to Ogre:

https://github.com/ignitionrobotics/ign-rendering/blob/ae292a88bd752822e0c0f6ed6fd6029e3fc64d0e/ogre2/src/Ogre2Camera.cc#L75-L81

Then when the image size changes, Ogre keeps the VFOV fixed and changes the HFOV to compensate, but our API doesn't pick that up.

I think that a solution would be to override `Camera::HFOV` and return the current angle calculated from VFOV. I haven't opened a PR because I wanted to check that all my assumptions are correct and I'm not missing some important detail.

iche033, 2021-12-03:
yeah I think that's what's happening. Alternatively, we can try to maintain the HFOV by calculating the new VFOV based on the new aspect ratio whenever window size changes. Either way works.

chapulina, 2021-12-06:
> Alternatively, we can try to maintain the HFOV by calculating the new VFOV based on the new aspect ratio whenever window size changes. 

I thought of that, but I assumed that the current behaviour was preferred because I feel like most simulations are spread horizontally rather than vertically.

I'll look into fixing the returned value.

azeey, 2024-05-31:
Fixed by https://github.com/gazebosim/gz-rendering/pull/636

closed 2024-05-31 by azeey