gazebosim/gz-rendering #506: Roughness map has no affect on environment map reflections

ColeOSRF, 2021-12-07:
## Environment
* OS Version: Ubuntu 20.04 
Ignition Fortress, source

## Description
* Expected behavior: Roughness maps should affect the sharpness of the environment map applied to objects.
* Actual behavior: Roughness maps have no effect on the reflectivity from the environment map

## Steps to reproduce

1. Create an object and apply an environment map.
2. Adjust the roughness value from 0 to 1. Note the reflection remains sharp and unchanged regardless of the roughness value.
3. A high roughness value should be making the reflection more diffuse like this.
[image]

## Output
Based on this image on fuel it appears that roughness used to work as you would expect. Note the roughness map is causing the reflection on the body of the extinguisher to become diffuse.
[image]

Compared to current functionality where the reflection is 100% sharp and the reflection intensity is even on the entire model since the roughness map is not being used to control the roughness of the different surfaces of the model.
[image]

darksylinc, 2021-12-11:
Ogre2 does support roughness maps on env maps.

However we need the texture to have mipmaps for that. I suspect that, for whatever reason, the env. map does not have mipmaps.

How is the env. map reflection loaded? How can we repro it?

Cheers

iche033, 2021-12-13:
We did some more tests afterwards using [SaintPetersBasilica.dds](https://github.com/OGRECave/ogre-next/blob/v2-2/Samples/Media/materials/textures/Cubemaps/SaintPetersBasilica.dds) from ogre and we do see differences in the result. I now see a blurry / low res image in the reflection. I opened the dds file from ogre using gimp-dds and noticed that there are mipmaps and ours don't so that could be it. I think @ColeOSRF is still testing to see if it's possible to tune more params and produce the desired effect.