gazebosim/gz-rendering #508: Ogre2 renders incorrect size of DEM heightmap

jennuine, 2021-12-09:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build?
    Source, https://github.com/ignitionrobotics/ign-rendering/tree/jennuine/dem_ex, 58800ec88374f7084d3615659a7fd61f005f5601
  

## Description
Size of heightmap (which is a `common::Dem`) is rendering incorrectly for ogre2 but correctly in ogre1.

| **ogre1** | **ogre2**|
| ------------ | ------------ |
|[image] | [image]|

## Steps to reproduce

This requires https://github.com/ignitionrobotics/ign-common/commit/57f2d087ff5229a7b0cf964ccd9377920472bee9 with `ign-rendering` branch https://github.com/ignitionrobotics/ign-rendering/tree/jennuine/dem_ex

1. `cd /path/to/ign-rendering/examples/heightmap`
2. `mkdir build && cd build`
3. `cmake .. && make`
4. `./heightmap ogre` or `./heightmap ogre2`

chapulina, 2022-03-09:
See #580

jennuine, 2022-03-24:
Resolved in https://github.com/ignitionrobotics/ign-rendering/pull/580, thanks @chapulina! :sparkles:

PR #580 (2022-03-15): [ogre2] Fix heightmap height and update docs
# 🦟 Bug fix

* Closes #508 
* Builds on top of #560 

## Summary

The Ogre 2 implementation was incorrectly multiplying the final size by the height difference for Z, which resulted on extremely tall heightmaps. According to the [SDF spec](http://sdformat.org/spec?ver=1.9&elem=geometry#heightmap_size), the size is meant to represent the final size of the heightmap; it's not a scaling factor. Gazebo [passes](https://github.com/ignitionrobotics/ign-gazebo/blob/f5bb284884de9c5d4d7fa5614e53b528654b6611/src/rendering/SceneManager.cc#L705) the SDF's size  [... cut]