gazebosim/gz-rendering #516: Throwless Variant?

darksylinc, 2021-12-18:
ignition::rendering::Variant throws whenever it is interpreted as the wrong type.

e.g.
```cpp
Variant labelAny = ogreVisual->UserData("label");
int label;
try
{
  label = std::get<int>(labelAny);
}
catch(std::bad_variant_access &e)
{
  // items with no class are considered background
  label = this->segmentationCamera->BackgroundLabel();
}
```

This looks fine except it makes debugging hard because "break on thrown exceptions" hits dozen/hundreds of times per frame

## Desired behavior
Code should work the same as now, but could avoid throwing exceptions.

## Implementation suggestion

Something like the following could work:

```cpp
const int *label = std::get_if<int>(&labelAny, );
if (!label)
{
  // items with no class are considered background
  label = this->segmentationCamera->BackgroundLabel();
}
```

## Additional context

This is a quality of life improvement for developers trying to troubleshoot a problem. IMHO exceptions should be used as a rare error that warrants attention or special handling.

What are exceptions useful for are subject to lots of debate though (e.g. Python by default treats every tiny mistake via exceptions)

darksylinc, 2021-12-18:
If there are no objections, I can submit a PR using get_if

iche033, 2021-12-20:
yep, no objections, we can use `get_if`

azeey, 2024-05-31:
The task here is to change existing code that uses `std::get` to `std::get_if`. Examples:
* https://github.com/gazebosim/gz-rendering/blob/f3d30738726d11d240907e30699ce4c66e8a0f50/ogre2/src/Ogre2SegmentationMaterialSwitcher.cc#L91
* https://github.com/gazebosim/gz-rendering/blob/f3d30738726d11d240907e30699ce4c66e8a0f50/ogre/src/OgreThermalCamera.cc#L224
* https://github.com/gazebosim/gz-rendering/blob/f3d30738726d11d240907e30699ce4c66e8a0f50/ogre2/src/Ogre2BoundingBoxMaterialSwitcher.cc#L122

dheerubhai-101, 2025-04-22:
hey, saw about this issue on LinkedIn. I'll give it a shot.

iche033, 2025-04-23:
sure go ahead, I've assigned the issue to you

kscottz, 2025-04-24:
@dheerubhai-101 thanks for the help. Please include this issue number in your PR. Let us know if you need help.

dheerubhai-101, 2025-04-24:
Hey both, thank you. I'm almost done with my changes. This is my first time contributing, so I'm not completely sure how to test. After going through the Gazebo Contribution guide, will it be enough to follow the Test Coverage block and try the relevant tests in the tests folder?

iche033, 2025-04-24:
yeah depending on what you changed, you can run the relevant tests manually. The tests executables are located in the `build/bin` directory. I'm guessing that the affected tests are:

```
INTEGRATION_segmentation_camera
INTEGRATION_thermal_camera
INTEGRATION_boundingbox_camera
```

When you create a pull request, a number of CI builds will be started. They will also build and run all the tests on your branch on all platforms.

closed 2025-05-02 by iche033