gazebosim/gz-rendering #526: Fortress (Ogre 2.2?) broke GUI with VirtualGL

peci1, 2022-01-04:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? binary 6.1.0
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: ogre2.
      - [ ] Sensor rendering error.
      - [x] GUI rendering error.
    - EGL headless mode:
      - [ ] Running in EGL headless mode
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [x] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [x] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [x] GPU is concurrently used for other tasks
        - [x] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA
            06:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Renoir [1002:1636] (rev d1)
          echo "$DISPLAY"
            :0
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
              direct rendering: Yes
                  Preferred profile: core (0x1)
                  Max core profile version: 4.6
                  Max compat profile version: 4.6
                  Max GLES1 profile version: 1.1
                  Max GLES[23] profile version: 3.2
              OpenGL vendor string: AMD
              OpenGL renderer string: AMD RENOIR (DRM 3.42.0, 5.15.12-051512-generic, LLVM 13.0.0)
              OpenGL core profile version string: 4.6 (Core Profile) Mesa 21.3.3 - kisak-mesa PPA
              OpenGL core profile shading language version string: 4.60
              OpenGL core profile context flags: (none)
              OpenGL core profile profile mask: core profile
              OpenGL version string: 4.6 (Compatibility Profile) Mesa 21.3.3 - kisak-mesa PPA
              OpenGL shading language version string: 4.60
              OpenGL context flags: (none)
              OpenGL profile mask: compatibility profile
              OpenGL ES profile version string: OpenGL ES 3.2 Mesa 21.3.3 - kisak-mesa PPA
              OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
          ps aux | grep Xorg
            peci1     2917  2.9  0.8 2125692 245400 tty2   Sl+  led03  45:28 /usr/lib/xorg/Xorg vt2 -displayfd 3 -auth /run/user/1000/gdm/Xauthority -background none -noreset -keeptty -verbose 3
          sudo env LANG=C X -version
              X.Org X Server 1.20.8
              X Protocol Version 11, Revision 0
              Build Operating System: Linux 4.15.0-163-generic x86_64 Ubuntu
              Current Operating System: Linux cras-17 5.15.12-051512-generic #202112290738 SMP Wed Dec 29 13:02:47 UTC 2021 x86_64
              Kernel command line: BOOT_IMAGE=/boot/vmlinuz-5.15.12-051512-generic root=UUID=d982512a-7987-4852-992c-9aeeddc50c46 ro delayacct
              Build Date: 14 December 2021  02:34:57PM
              xorg-server-hwe-18.04 2:1.20.8-2ubuntu2.2~18.04.6 (For technical support please see http://www.ubuntu.com/support)
              Current version of pixman: 0.34.0
                Before reporting problems, check http://wiki.x.org
                to make sure that you have the latest version.
          ```
    - [x] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
  
```
[... 118 lines cut ...]
```

</details>

## Description
* Expected behavior: Running GUI with 3D operations performed by VirtualGL should work.
* Actual behavior: GUI crashes.

## Steps to reproduce
1. Install VirtualGL from https://sourceforge.net/projects/virtualgl/files/3.0/
2. Either configure the VGL server [according to docs](https://rawcdn.githack.com/VirtualGL/virtualgl/3.0/doc/index.html#hd006) or just for quick testing call `sudo chmod g+rw /dev/dri/card0`
3. `vglrun +v -d /dev/dri/card0 ign gazebo -v4`
4. GUI crashes

## Output

```
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
```

VirtualGL is a nice way to run Gazebo in many constrained environments where e.g. X server cannot be run, or it is running on a wrong GPU etc. We've been using it all the time in SubT challenge with Dome without problems. The above example works correctly in Dome and Edifice and fails in Fortress.

The EGL backend of VirtualGL works so that it intercepts GLX calls from the application and substitutes them with relevant EGL calls. This translation/faking layer is not 100% feature complete. However, with the error being so non-informative, I have no idea what could be wrong (whether it's gazebo-side or virtualgl-side).

Here is a trace of the VirtualGL function interposer, but I can't make anything useful from it: https://pastebin.com/nsCJ8SSE

darksylinc, 2022-01-05:
Why it broken in Fortress/Ogre2.2 I do not know.

What I can tell you, is that when "currentGLContext" = "1" (which is what Gazebo does in both Edifice & Fortress) both Ogre 2.1 and 2.2 will perform the following:

```cpp
if( "currentGLContext" == "1" )
{
                if( !glXGetCurrentContext() )
                {
                    OGRE_EXCEPT( Exception::ERR_RENDERINGAPI_ERROR,
                                 "currentGLContext was specified with no current GL context",
                                 "GLXWindow::create" );
                }
}
```

`glXGetCurrentContext` is a GLX call.

The algorithm is quite simple: if "currentGLContext" is set, then we expect a GL context to be bound.
(i.e. someone already called `glXMakeContextCurrent` or `glXMakeCurrent`)

Given that the context is created by Qt when using GUI, I suspect the cause is somewhere in Qt code, or some subtle change in Ogre2RenderEngine::CreateRenderWindow (or parent caller, or code executed up until now) that no longer sets the context.

Or alternatively, a routine caused a call to `glXMakeCurrent( display, None, None );` which unsets the context.
I'd put a breakpoint in `void GLXContext::endCurrent() { glXMakeCurrent( mGLSupport->getGLDisplay(), None, None ); }` on `RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXContext.cpp` to be sure it's not on the Ogre side.

TL;DR there is a missing call to  `glXMakeCurrent` that must happen before Ogre is created, or an extra `glXMakeCurrent( display, None, None );` that shouldn't be there.
But given that this only happens with VGL (instead of always) I have no idea why that would happen.

Perhaps Qt no longer uses `glXMakeCurrent` and uses something else that can still be retrieved with `glXGetCurrentContext` but broke with VGL.

peci1, 2022-01-05:
Here is a gdb trace of just the GUI printing every call of `glXMakeCurrent` (without VGL):

<details>

```
[Msg] Ignition Gazebo GUI    v6.3.0
[New Thread 0x7fffe2438700 (LWP 15279)]
[New Thread 0x7fffd82f7700 (LWP 15290)]
[New Thread 0x7fffd7af6700 (LWP 15291)]
[New Thread 0x7fffd605f700 (LWP 15292)]
[Dbg] [Application.cc:88] Initializing application.
[New Thread 0x7fffd585e700 (LWP 15294)]
[GUI] [Dbg] [Application.cc:432] Create main window
[New Thread 0x7fffb7afa700 (LWP 15295)]
[New Thread 0x7fffb6ab7700 (LWP 15296)]
[New Thread 0x7fffb62b6700 (LWP 15297)]
[New Thread 0x7fffb5ab5700 (LWP 15298)]
[New Thread 0x7fffb52b4700 (LWP 15299)]
[New Thread 0x7fffb4ab3700 (LWP 15300)]
[New Thread 0x7fff97fff700 (LWP 15301)]
[New Thread 0x7fff9ffff700 (LWP 15302)]
[New Thread 0x7fff9f7fe700 (LWP 15303)]
[New Thread 0x7fff9effd700 (LWP 15304)]
[New Thread 0x7fff9e7fc700 (LWP 15305)]
[New Thread 0x7fff9dffb700 (LWP 15306)]
[New Thread 0x7fff9d7fa700 (LWP 15307)]
[New Thread 0x7fff9cff9700 (LWP 15308)]
[New Thread 0x7fff977fe700 (LWP 15309)]
[New Thread 0x7fff96ffd700 (LWP 15310)]
[New Thread 0x7fff967fc700 (LWP 15311)]
[New Thread 0x7fff95ffb700 (LWP 15312)]
[New Thread 0x7fff957fa700 (LWP 15313)]
[New Thread 0x7fff94ff9700 (LWP 15314)]

Thread 1 "gui" hit Breakpoint 1, glXMakeCurrent (dpy=0x5555558bd970, drawable=37748743, context=0x555555cd5010) at ../src/GLX/libglx.c:1119
1119	../src/GLX/libglx.c: No such file or directory.

Thread 1 "gui" hit Breakpoint 1, glXMakeCurrent (dpy=0x5555558bd970, drawable=0, context=0x0) at ../src/GLX/libglx.c:1119
1119	in ../src/GLX/libglx.c
[Thread 0x7fff94ff9700 (LWP 15314) exited]
[New Thread 0x7fff94ff9700 (LWP 15333)]
[New Thread 0x7fff5fba4700 (LWP 15334)]
[New Thread 0x7fff5f3a3700 (LWP 15335)]
[New Thread 0x7fff5eba2700 (LWP 15336)]
[New Thread 0x7fff5e3a1700 (LWP 15337)]
[New Thread 0x7fff5dba0700 (LWP 15338)]
[GUI] [Dbg] [PathManager.cc:66] Requesting resource paths through [/gazebo/resource_paths/get]
[GUI] [Dbg] [Gui.cc:156] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [PathManager.cc:55] Received resource paths.
[GUI] [Dbg] [Gui.cc:215] Requesting GUI from [/world/default/gui/info]...
[GUI] [Dbg] [GuiRunner.cc:114] Requesting initial state from [/world/default/state]...
[GUI] [Msg] Loading config [/home/peci1/.ignition/gazebo/6/gui.config]
[GUI] [Dbg] [Application.cc:309] Loading plugin [MinimalScene]
[New Thread 0x7fff49e39700 (LWP 15342)]
[New Thread 0x7fff491c9700 (LWP 15343)]
[New Thread 0x7fff489c8700 (LWP 15344)]
[GUI] [Msg] Added plugin [3D View] to main window
[GUI] [Msg] Loaded plugin [MinimalScene] from path [/usr/lib/x86_64-linux-gnu/ign-gui-6/plugins/libMinimalScene.so]
[GUI] [Dbg] [Application.cc:309] Loading plugin [EntityContextMenuPlugin]
[GUI] [Msg] Added plugin [EntityContextMenu] to main window
[GUI] [Msg] Loaded plugin [EntityContextMenuPlugin] from path [/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/gui/libEntityContextMenuPlugin.so]
[GUI] [Dbg] [Application.cc:309] Loading plugin [GzSceneManager]
[GUI] [Msg] Added plugin [Scene Manager] to main window
[GUI] [Msg] Loaded plugin [GzSceneManager] from path [/usr/lib/x86_64-linux-gnu/ign-gazebo-6/plugins/gui/libGzSceneManager.so]
[GUI] [Dbg] [Application.cc:309] Loading plugin [InteractiveViewControl]
[GUI] [Msg] Camera view controller topic advertised on [/gui/camera/view_control]
[GUI] [Msg] Added plugin [] to main window
[GUI] [Msg] Loaded plugin [InteractiveViewControl] from path [/usr/lib/x86_64-linux-gnu/ign-gui-6/plugins/libInteractiveViewControl.so]
[GUI] [Dbg] [Application.cc:309] Loading plugin [CameraTracking]
[GUI] [Msg] Added plugin [Camera tracking] to main window
[... 235 lines cut ...]
[GUI] [Msg] Follow service on [/gui/follow]
[GUI] [Msg] Move to pose service on [/gui/move_to/pose]
[GUI] [Msg] Camera pose topic advertised on [/gui/camera/pose]
[GUI] [Msg] Follow offset service on [/gui/follow/offset]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [InteractiveViewControl.cc:130] InteractiveViewControl plugin is moving camera [scene::Camera(65527)]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Msg] Loading plugin [ignition-rendering-ogre2]
[GUI] [Dbg] [EntityContextMenuPlugin.cc:79] Entity context menu plugin is using camera [scene::Camera(65527)]
[Switching to Thread 0x7ffff7f5d800 (LWP 15144)]

Thread 1 "gui" hit Breakpoint 1, glXMakeCurrent (dpy=0x5555558bd970, drawable=37748753, context=0x555559193060) at ../src/GLX/libglx.c:1119
1119	in ../src/GLX/libglx.c

Thread 1 "gui" hit Breakpoint 1, glXMakeCurrent (dpy=0x5555558bd970, drawable=37748753, context=0x555559193060) at ../src/GLX/libglx.c:1119
1119	in ../src/GLX/libglx.c

Thread 1 "gui" hit Breakpoint 1, glXMakeCurrent (dpy=0x5555558bd970, drawable=37748753, context=0x555559193060) at ../src/GLX/libglx.c:1119
1119	in ../src/GLX/libglx.c

Thread 1 "gui" hit Breakpoint 1, glXMakeCurrent (dpy=0x5555558bd970, drawable=37748753, context=0x555559193060) at ../src/GLX/libglx.c:1119
1119	in ../src/GLX/libglx.c
```

</details>

peci1, 2022-01-05:
And the same kind of trace with VGL:

<detail>

```
[Msg] Ignition Gazebo GUI    v6.3.0
[VGL] Shared memory segment ID for vglconfig: 65564
[VGL] VirtualGL v3.0.1 64-bit (Build 20220105)
[New Thread 0x7fffe1c98700 (LWP 19257)]
[New Thread 0x7fffd7c43700 (LWP 19258)]
[New Thread 0x7fffd7442700 (LWP 19259)]
[New Thread 0x7fffd59ab700 (LWP 19260)]
[Dbg] [Application.cc:88] Initializing application.
[New Thread 0x7fffd51aa700 (LWP 19262)]
[GUI] [Dbg] [Application.cc:432] Create main window
[VGL] Opening EGL device /dev/dri/card0
[New Thread 0x7fffb7262700 (LWP 19273)]
[New Thread 0x7fffb60de700 (LWP 19274)]
[New Thread 0x7fffb58dd700 (LWP 19275)]
[New Thread 0x7fffb50dc700 (LWP 19276)]
[New Thread 0x7fffb48db700 (LWP 19277)]
[New Thread 0x7fff9ffff700 (LWP 19278)]
[New Thread 0x7fff9f7fe700 (LWP 19279)]
[New Thread 0x7fff9effd700 (LWP 19280)]
[New Thread 0x7fff9e7fc700 (LWP 19281)]
[New Thread 0x7fff9dffb700 (LWP 19282)]
[New Thread 0x7fff9d7fa700 (LWP 19283)]
[New Thread 0x7fff9cff9700 (LWP 19284)]
[New Thread 0x7fff87fff700 (LWP 19285)]
[New Thread 0x7fff877fe700 (LWP 19286)]
[New Thread 0x7fff86ffd700 (LWP 19287)]
[New Thread 0x7fff867fc700 (LWP 19288)]
[New Thread 0x7fff85ffb700 (LWP 19289)]
[New Thread 0x7fff857fa700 (LWP 19290)]
[New Thread 0x7fff84ff9700 (LWP 19291)]
[Thread 0x7fff877fe700 (LWP 19286) exited]
[Thread 0x7fff867fc700 (LWP 19288) exited]
[Thread 0x7fff84ff9700 (LWP 19291) exited]
[Thread 0x7fff85ffb700 (LWP 19289) exited]
[Thread 0x7fff857fa700 (LWP 19290) exited]
[Thread 0x7fff86ffd700 (LWP 19287) exited]
[Thread 0x7fff87fff700 (LWP 19285) exited]
[Thread 0x7fff9cff9700 (LWP 19284) exited]
[Thread 0x7fff9d7fa700 (LWP 19283) exited]
[Thread 0x7fff9dffb700 (LWP 19282) exited]
[Thread 0x7fff9e7fc700 (LWP 19281) exited]
[Thread 0x7fff9effd700 (LWP 19280) exited]
[Thread 0x7fff9f7fe700 (LWP 19279) exited]
[Thread 0x7fff9ffff700 (LWP 19278) exited]
[Thread 0x7fffb48db700 (LWP 19277) exited]
[Thread 0x7fffb50dc700 (LWP 19276) exited]
[Thread 0x7fffb58dd700 (LWP 19275) exited]
[Thread 0x7fffb60de700 (LWP 19274) exited]
[Thread 0x7fffb7262700 (LWP 19273) exited]
[New Thread 0x7fffb7262700 (LWP 19292)]
[New Thread 0x7fffb60de700 (LWP 19293)]
[New Thread 0x7fff84ff9700 (LWP 19294)]
[New Thread 0x7fff857fa700 (LWP 19295)]
[New Thread 0x7fffb58dd700 (LWP 19296)]
[New Thread 0x7fffb50dc700 (LWP 19297)]
[New Thread 0x7fffb48db700 (LWP 19298)]
[New Thread 0x7fff9ffff700 (LWP 19299)]
[New Thread 0x7fff9f7fe700 (LWP 19300)]
[New Thread 0x7fff9effd700 (LWP 19301)]
[New Thread 0x7fff9e7fc700 (LWP 19302)]
[New Thread 0x7fff9dffb700 (LWP 19303)]
[New Thread 0x7fff9d7fa700 (LWP 19304)]
[New Thread 0x7fff9cff9700 (LWP 19305)]
[New Thread 0x7fff87fff700 (LWP 19306)]
[New Thread 0x7fff877fe700 (LWP 19307)]
[... 272 lines cut ...]
[Thread 0x7ffec6b32700 (LWP 19434) exited]
[Thread 0x7ffec7333700 (LWP 19433) exited]
[Thread 0x7ffe7f7fe700 (LWP 19432) exited]
[Thread 0x7ffe877fe700 (LWP 19431) exited]
[Thread 0x7ffe86ffd700 (LWP 19430) exited]
[Thread 0x7ffec7b34700 (LWP 19429) exited]
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/GLX/OgreGLXWindow.cpp (line 165)
[GUI] [Err] [Ogre2RenderEngine.cc:958]  Unable to create the rendering window: OGRE EXCEPTION(3:RenderingAPIException): currentGLContext was specified with no current GL context in GLXWindow::create at /var/li [... cut ...]
0x00007ffed923d177 in Ogre::Hlms::createDatablock(Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Ogre::HlmsMacroblock const&, Ogre::HlmsBlendblock const&, std::vector<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::STLAllocator<std::pair<Ogre::IdString, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, Ogre::CategorisedAllocPolicy<(Ogre::MemoryCategory)0> > > const&, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11
```

</detail>

peci1, 2022-01-05:
If I found the right Qt source, it should be this one handling the context: https://code.woboq.org/qt5/qtbase/src/plugins/platforms/xcb/gl_integrations/xcb_glx/qglxintegration.cpp.html . It seems it is still full of `glXMakeCurrent()` calls...

iche033, 2022-12-09:
I looked into this a little and was able to reproduce this issue using the container suggested in https://github.com/gazebosim/gz-sim/issues/1746#issuecomment-1264614220. For me, I noticed that the GL context was lost (`glXMakeCurrent` returns null) after [loading the OGRE GL3Plus plugin](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2RenderEngine.cc#L393). Specifically it happens after the plugin tried to query EGL support. If I comment out this line then I no longer get the crash: 
https://github.com/OGRECave/ogre-next/blob/v2-2/RenderSystems/GL3Plus/src/windowing/OgreGlSwitchableSupport.cpp#L69

Similarly if I build ogre-next from source without EGL support then gazebo works works fine.

darksylinc, 2022-12-09:
Mmm, `EglPBufferSupport::initDevice` will start querying all the GPUs and try to create them; in order to see which ones are compatible.

Including calls to `eglMakeCurrent`.

Perhaps if Gazebo manually saves the GL context before loading the GL3+ plugin, and restores the context after loading it (and before creating a window) this problem can be fixed?

I don't know why the problem is specific to VirtualGL though.

peci1, 2022-12-18:
From https://registry.khronos.org/EGL/sdk/docs/man/html/eglMakeCurrent.xhtml:

> For purposes of eglMakeCurrent, the client API type of all OpenGL ES and OpenGL contexts is considered the same. In other words, if any OpenGL ES context is currently bound and context is an OpenGL context, or if any OpenGL context is currently bound and context is an OpenGL ES context, the currently bound context will be made no longer current and context will be made current. 

So it makes sense the `EglPBufferSupport::initDevice()` calls reset the GLX buffer.

peci1, 2022-12-19:
I think it now makes sense to me.

When running without VirtualGL, the GLX context is created, and then several EGL contexts are probed, each with its own display.

When running with VirtualGL, the GLX context [is emulated](https://github.com/VirtualGL/virtualgl/blob/main/server/faker-glx.cpp#L1734) with [`eglMakeCurrent()`](https://github.com/VirtualGL/virtualgl/blob/main/server/backend.cpp#L816), and then further EGL contexts are probed, also using `eglMakeCurrent()`. One of these contexts' displays however matches the emulated GLX display, which results in the observed error.

I don't think there's anything that could be fixed on the VirtualGL side. It just can't know if you want to actually change the current EGL context, or if it should remain as it was for the GLX emulation. Or, it would have to report the EGL display used for emulation as unavailable for direct EGL operation. I'll ask about this on the VirtualGL repo.

I've written an MWE where you can easily test this behavior. The two lines with comment `// FIX` suggest how to fix this problem (they would normally go around [`loadPlugins()`](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2RenderEngine.cc#L593)). I've tested Gazebo 6 with this fix and it worked!

Compile with `g++ -o mwe mwe.cpp -lGL -lGLU -lX11 -lEGL`.

<details>

```c++
#include <GL/glx.h>
#include <X11/Xlib.h>
#include <X11/Xutil.h>
#include <iostream>
#include <EGL/egl.h>
#include <EGL/eglext.h>
#include <vector>

int main(int argc, char** argv)
{

auto dummyDisplay = XOpenDisplay(0);
Display *x11Display = static_cast<Display*>(dummyDisplay);
int screenId = DefaultScreen(x11Display);
int attributeList[] = {
    GLX_RENDER_TYPE, GLX_RGBA_BIT,
    GLX_DOUBLEBUFFER, True,
    GLX_DEPTH_SIZE, 16,
    GLX_STENCIL_SIZE, 8,
    None
};
int nelements = 0;
auto dummyFBConfigs = glXChooseFBConfig(x11Display, screenId, attributeList, &nelements);
auto dummyWindowId = XCreateSimpleWindow(x11Display, RootWindow(dummyDisplay, screenId), 0, 0, 1, 1, 0, 0, 0);
PFNGLXCREATECONTEXTATTRIBSARBPROC glXCreateContextAttribsARB = 0;
glXCreateContextAttribsARB = (PFNGLXCREATECONTEXTATTRIBSARBPROC)glXGetProcAddress((const GLubyte *)"glXCreateContextAttribsARB");
int contextAttribs[] = {
        GLX_CONTEXT_MAJOR_VERSION_ARB, 3,  //
        GLX_CONTEXT_MINOR_VERSION_ARB, 3,  //
        None                               //
};
auto dummyContext = glXCreateContextAttribsARB(x11Display, dummyFBConfigs[0], nullptr, 1, contextAttribs);
// Create the GLX context and set it as current
GLXContext x11Context = static_cast<GLXContext>(dummyContext);
glXMakeCurrent(x11Display, dummyWindowId, x11Context);
std::cerr << glXGetCurrentContext() << " " << glXGetCurrentDrawable() << " " << glXGetCurrentDisplay() << std::endl;

typedef EGLBoolean ( *EGLQueryDevicesType )( EGLint, EGLDeviceEXT *, EGLint * );
auto eglQueryDevices = (EGLQueryDevicesType)eglGetProcAddress( "eglQueryDevicesEXT" );
auto eglQueryDeviceStringEXT = (PFNEGLQUERYDEVICESTRINGEXTPROC)eglGetProcAddress( "eglQueryDeviceStringEXT" );
EGLint numDevices = 0;
eglQueryDevices( 0, 0, &numDevices );

std::vector<EGLDeviceEXT> mDevices;
if( numDevices > 0 )
{
      mDevices.resize( static_cast<size_t>( numDevices ) );
      eglQueryDevices( numDevices, mDevices.data(), &numDevices );
}

for( int i = 0u; i < numDevices; ++i )
{
   EGLDeviceEXT device = mDevices[size_t( i )];
   auto name = std::string(eglQueryDeviceStringEXT( device, EGL_EXTENSIONS ));

   const char *gpuCard = eglQueryDeviceStringEXT( device, EGL_DRM_DEVICE_FILE_EXT );
   if( gpuCard ) name += std::string(" ") + gpuCard;

   std::cerr << i << " " << name << std::endl;

   EGLAttrib attribs[] = { EGL_NONE };
   auto eglDisplay = eglGetPlatformDisplay( EGL_PLATFORM_DEVICE_EXT, mDevices[i], attribs );
   EGLint major = 0, minor = 0;
   eglInitialize( eglDisplay, &major, &minor );

   const EGLint configAttribs[] = {
            EGL_SURFACE_TYPE,    EGL_PBUFFER_BIT, EGL_BLUE_SIZE, 8, EGL_GREEN_SIZE, 8, EGL_RED_SIZE, 8,
            EGL_RENDERABLE_TYPE, EGL_OPENGL_BIT,  EGL_NONE
   };

   EGLint numConfigs;
   EGLConfig eglCfg;
   eglChooseConfig( eglDisplay, configAttribs, &eglCfg, 1, &numConfigs );
   const EGLint pbufferAttribs[] = {
            EGL_WIDTH, 1, EGL_HEIGHT, 1, EGL_NONE,
   };
   auto eglSurf = eglCreatePbufferSurface( eglDisplay, eglCfg, pbufferAttribs );
   eglBindAPI( EGL_OPENGL_API );
   EGLint contextAttrs[] = {
            EGL_CONTEXT_MAJOR_VERSION,
            4,
            EGL_CONTEXT_MINOR_VERSION,
            5,
            EGL_CONTEXT_OPENGL_PROFILE_MASK,
            EGL_CONTEXT_OPENGL_CORE_PROFILE_BIT_KHR,
            EGL_NONE
   };

   // Create the EGL context and make it current
   auto eglCtx = eglCreateContext( eglDisplay, eglCfg, 0, contextAttrs );
   std::cerr << glXGetCurrentContext() << " " << glXGetCurrentDrawable() << " " << glXGetCurrentDisplay() << std::endl;
   auto ctx = glXGetCurrentContext(); auto dpy = glXGetCurrentDisplay(); auto drw = glXGetCurrentDrawable();  // FIX
   eglMakeCurrent( eglDisplay, eglSurf, eglSurf, eglCtx );
   glXMakeCurrent(dpy, drw, ctx);  // FIX
   std::cerr << glXGetCurrentContext() << " " << glXGetCurrentDrawable() << " " << glXGetCurrentDisplay() << std::endl;
}

}
```

</details>

rgov, 2022-12-20:
I found this issue while trying to run Gazebo Garden under VirtualGL inside a container. I don't mean to derail the conversation here but I have an observation:

Gazebo always crashes as described when VirtualGL is run with the EGL backend, but I have gotten it to run to a limited extent with the GLX backend. But even with the GLX backend, there is a crash when I open some environments, (like `tethys_at_empty_environment.sdf` from the [osrf/lrauv](https://github.com/osrf/lrauv) project), but not others (like `multi_lrauv_race.sdf`). The crashing condition was filed as https://github.com/gazebosim/gz-sim/issues/1746 and duped here. Are these really the same issue?

(I don't know enough about VirtualGL or the graphics stack to know the difference between the backends or which one should be preferred. I'd be happy if either of them worked fully.)

iche033, 2022-12-20:
I don't know why the different worlds from the lrauv project result in different behavior. Just looking at the ogre log in https://github.com/gazebosim/gz-sim/issues/1746, it has the same errors as the one reported here about missing gl context so it could be due to the same issue.

peci1, 2022-12-20:
Here is a DEB I've built from the main branch of virtualgl with the patch from https://github.com/VirtualGL/virtualgl/issues/220#issuecomment-1358529845: [virtualgl_3.0.91_amd64-fixed.zip](https://github.com/gazebosim/gz-rendering/files/10272578/virtualgl_3.0.91_amd64-fixed.zip). With this patched version of VirtualGL, the crashes no more occur to me.

However, following the discussion in the VirtualGL issue, I still think it'd make sense to also implement the fix on Gazebo side - it seems that the fact the current procedure worked is more likely to be an accident than design. It seems that mixing both GLX and EGL calls in a single program is something nobody does and we haven't found any references saying how the interaction between the two should work. I'll prepare the Gazebo PR.

peci1, 2022-12-20:
Fix in https://github.com/gazebosim/gz-rendering/pull/794 .

rgov, 2022-12-21:
I cherry-picked this change to the `gz-rendering7` branch then ran Gazebo Garden with @peci1's build of VirtualGL using the EGL backend, with the `multi_lrauv_race.sdf` world.

Indeed I no longer get the error mentioning `GLXWindow::create`, but it crashes reliably. However the GLX backend appears to load worlds I couldn't before (maybe a red herring).

<details>
<summary>Crash log</summary>

```
[GUI] [Msg] Loading plugin [gz-rendering-ogre2]
libEGL warning: DRI2: failed to create dri screen
libEGL warning: DRI2: failed to create dri screen
[VGL] ERROR: in eglMakeCurrent()--
[VGL]    815: EGL_NOT_INITIALIZED
terminate called after throwing an instance of 'backend::EGLError'
  what():  40: EGL_NOT_INITIALIZED
Stack trace (most recent call last):
#31   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f43abb10809, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#30   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f43ab2aa0ef, in QApplication::notify(QObject*, QEvent*)
#29   Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5", at 0x7f43ab2a0a65, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#28   Object "/usr/lib/x86_64-linux-gnu/libQt5Quick.so.5", at 0x7f43aae77ca2, in QQuickItem::event(QEvent*)
#27   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5", at 0x7f43abb3cc29, in QObject::event(QEvent*)
#26   Object "/home/developer/gz_ws/install/lib/gz-gui-7/plugins/libMinimalScene.so", at 0x7f43850a2124, in gz::gui::plugins::RenderWindowItem::Ready()
#25   Object "/home/developer/gz_ws/install/lib/gz-gui-7/plugins/libMinimalScene.so", at 0x7f43850a1f90, in gz::gui::plugins::RenderThread::Initialize[abi:cxx11]()
#24   Object "/home/developer/gz_ws/install/lib/gz-gui-7/plugins/libMinimalScene.so", at 0x7f43850b174b, in gz::gui::plugins::RenderThreadRhiOpenGL::Initialize[abi:cxx11]()
#23   Object "/home/developer/gz_ws/install/lib/gz-gui-7/plugins/libMinimalScene.so", at 0x7f43850a48ef, in gz::gui::plugins::GzRenderer::Initialize[abi:cxx11]()
#22   Object "/home/developer/gz_ws/install/lib/libgz-rendering7.so.7", at 0x7f4385028b57, in gz::rendering::v7::RenderEngineManager::Engine(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#21   Object "/home/developer/gz_ws/install/lib/libgz-rendering7.so.7", at 0x7f4385028944, in gz::rendering::v7::RenderEngineManagerPrivate::Engine(EngineInfo, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#20   Object "/home/developer/gz_ws/install/lib/libgz-rendering7.so.7", at 0x7f43850369a3, in gz::rendering::v7::BaseRenderEngine::Load(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)
#19   Object "/home/developer/gz_ws/install/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f435bb86f6d, in gz::rendering::v7::Ogre2RenderEngine::LoadImpl(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)
#18   Object "/home/developer/gz_ws/install/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f435bb86210, in gz::rendering::v7::Ogre2RenderEngine::LoadAttempt()
#17   Object "/home/developer/gz_ws/install/lib/gz-rendering-7/engine-plugins/libgz-rendering-ogre2.so", at 0x7f435bb7fc9d, in gz::rendering::v7::Ogre2RenderEngine::LoadPlugins()
#16   Object "/lib/libvglfaker.so", at 0x7f43b1accfdb, in 
#15   Object "/lib/libvglfaker.so", at 0x7f43b1ae88e5, in 
#14   Object "/lib/libvglfaker.so", at 0x7f43b1b48f38, in 
#13   Object "/lib/libvglfaker.so", at 0x7f43b1b48d95, in 
#12   Object "/lib/libvglfaker.so", at 0x7f43b1b3f848, in 
#11   Object "/lib/libvglfaker.so", at 0x7f43b1b3ea5b, in 
[... cut ...]
04:22:53: OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device EGL_EXT_device_drm #1 /dev/dri/card0 in EGLSupport::getGLDisplay at /home/jenkins/workspace/ogre-2.3-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 320)
04:22:53: Destroying device: EGL_EXT_device_drm #1 /dev/dri/card0...
04:22:53: EGL Device: EGL_MESA_device_software #2
04:22:53: Trying to init device: EGL_MESA_device_software #2...
04:22:53: Created GL 3.3 context for device EGL_MESA_device_software #2
04:22:53: Destroying device: EGL_MESA_device_software #2...
04:22:53: Plugin successfully installed
```
</details>

peci1, 2022-12-21:
Actually, the PR to gz-rendering itself should be sufficient to fix the rendering issues. The patched VirtualGL binary was another approach to fix the issue "on the other end". Could you try with the non-patched VirtualGL?

I'll close this issue as the original one has been resolved. Let's continue this discussion in https://github.com/gazebosim/gz-sim/issues/1746 .

closed 2022-12-21 by peci1

peci1, 2022-12-22:
The VirtualGL-side fix has also been implemented: https://github.com/VirtualGL/virtualgl/issues/220 . So now this bug should be prevented from both sides :) I did a thorough compatibility test and all combinations I used worked.

PR #794 (2022-12-21): Fix OGRE 2 with VirtualGL
# 🦟 Bug fix

Fixes #526 and maybe https://github.com/gazebosim/gz-sim/issues/1746 .

## Summary
Restore GLX context after loading the OGRE plugin. It is possible the loading process changes the current GLX context.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test cov [... cut]