gazebosim/gz-rendering #556: Potential non-deterministic behavior in particle emitter (ogre2)

iche033, 2022-02-08:
## Environment
* OS Version: Ubuntu Focal
* Source or binary build? Affects both source and binary in all versions

## Description

The particle emitter in ogre2 uses the Particle FX plugin. The plugin likely animates the particles using real time so the effect could be non-deterministic.

* Expected behavior: Speed of particles are dependent on user-specified time steps
* Actual behavior: Particles are animated using real time

## Steps to reproduce
1. Easier to reproduce using ign gazebo. Change `<real_time_factor>` in [particle_emitter2.sdf](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo6/examples/worlds/particle_emitter2.sdf#L28) file to `0` so simulation runs as fast as possible.
2. Launch ign gazebo with the modified `particle_emitter2.sdf` world
3. See the particles flow upwards in regular speed when it should be flowing as fast as possible.

darksylinc, 2022-02-14:
I haven't looked much into this, for what is worth `ParticleSystemUpdateValue` register itself to update the particle, which gets called in `ControllerManager::updateAllControllers` and gets updated when we (ignition) directly or indirectly call `updateSceneGraph` (normally Ogre does that call but I think it was in Edifice that I changed it so that we call it explicitly).

I've been thinking there's two possible solutions:

1. Somehow figure out how to manually update the controlled ourselves so we use simulation time.
2. Modify Ogre so that `Root::populateFrameEvent` can alternatively user-specified time (i.e. simulation time) instead of real time.

The 2nd option is highly desirable because it will also fix *other* stuff I just realized we were overlooking: For example any time-based postprocessing FX (if there's any, I think the gaussian noise is using it) should be using simulation time, not real time.

darksylinc, 2022-04-04:
We should either reopen this ticket or create a new one because there is still one source of non-determinism to fix.

iche033, 2022-04-04:
github auto-closed the issue with the merge of #584 . Reopening.

darksylinc, 2022-04-30:
I've pushed https://github.com/OGRECave/ogre-next/commit/d3bd8b423d53b08276d7d5b4501f36160a0a25c5 to branches:
 - 2.2
 - 2.3
 - master

AFAICT it won't break ABI nor API.

By calling `ParticleSystemManager::getSingleton().setSimulationTickRate( positive_non_zero_value )` as soon as possible during initialization and then setting the tick rate to the desired one at any time will do the trick.

Now the thing is getting this patch into Ignition's fork.

iche033, 2022-05-05:
thanks, we'll likely just have this fix in Garden with ogre 2.3. So I'll keep in mind to test this as part of the review of your existing PR, https://github.com/ignitionrobotics/ign-rendering/pull/553.

PR #584 (2022-04-04): Use custom simulation time variants for Ogre
# 🦟 Bug fix

Fixes #556

## Summary

This fixes Particle FXs not respecting simulation time.

After researching Ogre behavior, turns out we already had the functionality to use custom simulation (deterministic) time. We just had to use it. Turned out to be much easier than I expected (I actually started writing more code until I realized it... :( )

**Note:** This is a draft because I was unsure _where to pull gazebo's simulation time from_. Therefore it is currently hardcoded with:

```cpp
static const Ogre::Real kSimulationTime = Ogre::Real(1.0 /  [... cut]
commit 3b0426e7 (2022-04-04): Use custom simulation time variants for Ogre (#584)

* Use custom simulation time variants for Ogre

This fixes Particle FXs not respecting simulation time
Fixes #556