gazebosim/gz-rendering #561: Example on how to leverage this with QOpenglWidget?

Levi-Armstrong, 2022-02-15:
Are there any examples on how you might leverage this with QOpenglWidget?

Levi-Armstrong, 2022-03-19:
I created a simple example using QOpenglWidget but not sure why I am getting weird rendering like what is shown below. I can sometime get it to render correctly after multiple resizes, but not sue what could be causing this. Below is a condensed version of the class. The `this->dataPtr->renderer.Render();` just uses the scene capture method to create an image. The saved image in the temp directory looks good. @chapulina do you know who might be able to shed some light on what I am doing wrong?

``` c++
class SimpleRenderWidget : public QOpenGLWidget, protected QOpenGLFunctions
{
  Q_OBJECT
public:
void SimpleRenderWidget::resizeGL(int w, int h)
{
  this->dataPtr->renderer.Resize(w, h);
  glViewport(0, 0, w, h);
}

/////////////////////////////////////////////////
void SimpleRenderWidget::paintGL()
{
  makeCurrent();
  ignition::rendering::Image image = this->dataPtr->renderer.Render();
  ignition::rendering::PixelFormat f = image.Format();
  ignerr << "Format: " << f << std::endl;
  auto* data = image.Data<unsigned char>();

  ignerr << "Image, w: " << image.Width() << " h: " << image.Height() << std::endl;
  ignition::common::Image tmp;
  tmp.SetFromData(data, image.Width(), image.Height(), ignition::common::Image::RGB_INT8);
  tmp.SavePNG("/tmp/ign_image.png");
  doneCurrent();

  makeCurrent();
  GLint m_viewport[4];
  glGetIntegerv( GL_VIEWPORT, m_viewport );
  ignerr << "ViewPort, x: " << m_viewport[0] << " y: " << m_viewport[1] << " w: " << m_viewport[2] << " h: " << m_viewport[3] << std::endl;
  glClearColor(0.0f, 0.0f, 0.0f, 1.0f);
  glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT);
  glDrawPixels(image.Width(), image.Height(), GL_RGB, GL_UNSIGNED_BYTE, data);
  doneCurrent();
}
};
```

[image]

Levi-Armstrong, 2022-03-20:
Here is a minimal example.

minimal_example.h
``` c++
#ifndef MINIMAL_EXAMPLE_H
#define MINIMAL_EXAMPLE_H

#include <QWidget>
#include <QOpenGLWidget>
#include <QOpenGLFunctions>

#include <ignition/rendering/Camera.hh>
#include <ignition/rendering/RayQuery.hh>
#include <ignition/rendering/RenderEngine.hh>
#include <ignition/rendering/RenderingIface.hh>
#include <ignition/rendering/Scene.hh>
#include <ignition/rendering/Grid.hh>
#include <ignition/common/Console.hh>
#include <ignition/common/Image.hh>

class IssueRenderWidget : public QOpenGLWidget, protected QOpenGLFunctions
{
  Q_OBJECT
public:

  void initialize()
  {
    std::map<std::string, std::string> params;
    params["useCurrentGLContext"] = "1";
    params["winID"] = std::to_string(this->winId());
  //  params["winID"] = std::to_string(
  //    ignition::gui::App()->findChild<ignition::gui::MainWindow *>()->
  //      QuickWindow()-winId());
    auto* engine = ignition::rendering::engine(engine_name_, params);
    if (engine == nullptr)
    {
      ignerr << "Engine [" << this->engine_name_ << "] is not supported" << std::endl;
      return;
    }

    // Scene
    auto scene = engine->SceneByName(scene_name_);
    if (!scene)
    {
      igndbg << "Create scene [" << scene_name_ << "]" << std::endl;
      scene = engine->CreateScene(scene_name_);
      scene->SetAmbientLight(ambient_light_);
      scene->SetBackgroundColor(background_color_);
    }

    scene->SetSkyEnabled(true);

    ignition::rendering::VisualPtr visual = scene->VisualByName("tesseract_grid");
    if (visual == nullptr)
    {
      ignition::rendering::VisualPtr root = scene->RootVisual();

      // create gray material
      ignition::rendering::MaterialPtr gray = scene->CreateMaterial();
      gray->SetAmbient(0.7, 0.7, 0.7);
      gray->SetDiffuse(0.7, 0.7, 0.7);
      gray->SetSpecular(0.7, 0.7, 0.7);

      // create grid visual
      unsigned id = 1000; //static_cast<unsigned>(this->dataPtr->entity_manager.addVisual("tesseract_grid"));
      ignition::rendering::VisualPtr visual = scene->CreateVisual(id, "tesseract_grid");
      ignition::rendering::GridPtr gridGeom = scene->CreateGrid();
      if (!gridGeom)
      {
        ignwarn << "Failed to create grid for scene ["
[... 145 lines cut ...]

CMakeLists.txt
``` cmake
find_package(ignition-rendering6 REQUIRED)
find_package(ignition-common4 COMPONENTS profiler events av REQUIRED)
find_package(Qt5 COMPONENTS Core Widgets Gui OpenGL REQUIRED)
find_package(OpenGL REQUIRED)

QT5_WRAP_CPP(MINIMAL_EXAMPLE_WIDGET_headers_MOC minimal_example.h)

add_executable(${PROJECT_NAME}_minimal_example
  ${MINIMAL_EXAMPLE_WIDGET_headers_MOC}
  main.cpp)
target_link_libraries(${PROJECT_NAME}_minimal_example PUBLIC
  ignition-rendering6::ignition-rendering6
  ignition-common4::ignition-common4
  Qt5::Core Qt5::Widgets Qt5::Gui Qt5::OpenGL
   ${OPENGL_LIBRARIES})
target_include_directories(${PROJECT_NAME}_minimal_example PUBLIC
    "$<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/include>"
    "$<INSTALL_INTERFACE:include>")

install_targets(TARGETS
  ${PROJECT_NAME}_minimal_example)
```

Levi-Armstrong, 2022-04-08:
This was resolved by adding these lines. @chapulina I have little experience with opengl but learning more each day. Do you or anyone else know why these would be needed when using the QOpenglWidget but not in the glut example? I assuming it may be a configuration settings not exactly being the same but not sure where to start looking.

```
  glPixelStorei(GL_UNPACK_ALIGNMENT, 1);
  glPixelStorei(GL_UNPACK_ROW_LENGTH, 0);
  glPixelStorei(GL_UNPACK_SKIP_PIXELS, 0);
  glPixelStorei(GL_UNPACK_SKIP_ROWS, 0);
```

chapulina, 2022-04-08:
Hi @Levi-Armstrong , glad you found a workaround. Unfortunately I also don't have much experience there. Maybe one of the `ign-rendering` maintainers can pitch in if they have time.

iche033, 2022-04-08:
I haven't had time to looked into this. My understanding is that since `GL_UNPACK_ALIGNMENT` defaults to 4 and we're using RGB image format here (3 bytes) with resizable image width, I think it makes sense to set it to `1` as that should work for tightly packed image data with no padding. You may not have to set `GL_UNPACK_ALIGNMENT` if the data is RGBA I think.

Levi-Armstrong, 2022-04-08:
@iche033 Is it possible to when capturing the image from camera to get different formats?

iche033, 2022-04-08:
in ign-rendering6 (or above) and ogre2 render engine, you should be able to do:

```
camera->SetImageFormat(PF_R8G8B8A8);
```

and the `Image` created by the camera should be RGBA. Then just change `GL_RGB` to `GL_RGBA` when calling `glDrawPixels`. See if that works

Levi-Armstrong, 2022-04-08:
Thanks I will try it and report back.

@iche033 Another thing I had to do which was different from the glut example was set the matrix modes to identity.

``` c++
      glMatrixMode(GL_PROJECTION);
      glLoadIdentity();
      glMatrixMode(GL_MODELVIEW);
      glLoadIdentity();
      glMatrixMode(GL_TEXTURE);
      glLoadIdentity();

      glPixelStorei(GL_UNPACK_ALIGNMENT, 1);
      glPixelStorei(GL_UNPACK_ROW_LENGTH, 0);
      glPixelStorei(GL_UNPACK_SKIP_PIXELS, 0);
      glPixelStorei(GL_UNPACK_SKIP_ROWS, 0);

      glPixelZoom(1, -1);
      glRasterPos2f(-1, 1);
      glDrawPixels(imgw, imgh, GL_RGB, GL_UNSIGNED_BYTE, data);
```

This was required if I added these two lines. Do you have an idea why this would be required with QtOpenGLWidget but not when using glut?

``` c++
      glPixelZoom(1, -1);
      glRasterPos2f(-1, 1);
```

Levi-Armstrong, 2022-04-08:
Another interesting thing is if I switch from `ogre` to `ogre2` I end up with the following but using the glut example every works fine.

ogre2:
[image]

ogre1:
[image]

iche033, 2022-04-18:
> This was required if I added these two lines. Do you have an idea why this would be required with QtOpenGLWidget but not when using glut?

I'm not too sure why the behavior is different from glut. One guess is that the QtOpenGLWidget might also be changing opengl parameters. The two lines were there to flip the image when drawn otherwise it would be upside down. 

> Another interesting thing is if I switch from ogre to ogre2 I end up with the following but using the glut example every works
fine.

ahh that's not great. I don't know why though. If you're ok with using `ogre`, I think you can stick with RGB format and call `glPixelStorei(GL_UNPACK_ALIGNMENT, 1);` as you don't really need RGBA format for displaying the result to screen

Levi-Armstrong, 2022-04-19:
Is the default format of the data the same when using capture between ogre, ogre2, and optic?

iche033, 2022-04-19:
> Is the default format of the data the same when using capture between ogre, ogre2, and optix?

format of the data captured from a camera? yes the should be the same format, [PF_R8G8B8](https://github.com/ignitionrobotics/ign-rendering/blob/ign-rendering6/include/ignition/rendering/PixelFormat.hh#L39)

Levi-Armstrong, 2022-04-20:
> format of the data captured from a camera? yes the should be the same format, [...]

huh, then I am not sure why it is having an issue when switching to ogre2 if the data is formatted the same.

Levi-Armstrong, 2022-11-08:
@iche033 Here is the video showing the behavior when adding object to the scene that contain transparency.

https://user-images.githubusercontent.com/9803128/200689672-54c20aea-0a5e-4ae1-9e38-bfedb37bb139.mp4

Levi-Armstrong, 2022-11-09:
Okay, so I have narrowed it down a little further. I decided to just load the image from a file and use `glDrawPixels` which works fine. Then is just enabled calling capture on the camera but not use the data and interestingly the green and blue showed up over the image that is loading from a file. In video you can see that I use the mouse to manipulate the camera in the scene, but I am only drawing a static image. This indicates to me that something in render engine is using the texture id that `glDrawPixels` uses but not sure how best to resolve this. @iche033 Any thoughts?

```
   ignition::common::Image image("/tmp/ign_image.png");
   unsigned char *data = nullptr;
   unsigned int cnt = 0;
   image.Data(&data, cnt);
```

https://user-images.githubusercontent.com/9803128/200748015-ea305ae3-a148-42d1-b942-c654e94e15bf.mp4

Levi-Armstrong, 2022-11-09:
I also tried using Qt objects to use the camera texture id and blit. Which has its own issue where the upper right triangle is black but the lower right triangle looks fine. See video. Any thoughts on why this would be happening?

```
  makeCurrent();
  glClearColor(0.0f, 0.0f, 0.0f, 1.0f);
  glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT | GL_ACCUM_BUFFER_BIT | GL_STENCIL_BUFFER_BIT);
  glBindTexture(GL_TEXTURE_2D, this->dataPtr->renderer.texture_id);
  this->dataPtr->texture_blitter.bind(GL_TEXTURE_2D);
  const QRect targetRect(QPoint(0, 0), size());
  QMatrix4x4 target = QOpenGLTextureBlitter::targetTransform(targetRect, QRect(QPoint(0, 0), this->dataPtr->renderer.textureSize));
  target.translate(0, 0, -1); // If all zeros the scene is randomly visible.
  this->dataPtr->texture_blitter.blit(this->dataPtr->renderer.texture_id, target, QOpenGLTextureBlitter::OriginTopLeft);
  this->dataPtr->texture_blitter.release();
  doneCurrent();
```

https://user-images.githubusercontent.com/9803128/200879457-c454a45b-c66c-474d-b755-1014ea206077.mp4

iche033, 2022-11-09:
> This indicates to me that something in render engine is using the texture id that glDrawPixels uses but not sure how best to resolve this. @iche033 Any thoughts?

It sounds to me that the ogre camera is rendering correctly but it's just that there are issues when the images are displayed to the screen?  

Another thing to try is disable the selection buffer that the camera is using for mousing picking. This is only other thing that is rendering apart from the camera. You can disable the [render update here](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/ogre2/src/Ogre2SelectionBuffer.cc#L461) and see if it makes any difference. Mouse picking and camera control may not work but see if you still get the green, blue colors.

> I also tried using Qt objects to use the camera texture id and blit. Which has its own issue where the upper right triangle is black but the lower right triangle looks fine. See video. Any thoughts on why this would be happening?

I haven't used  `QOpenGLWidget` / `QOpenGLTextureBlitter`. A quick look at the doc and examples, it seems fine. Just checking that your `this->dataPtr->renderer.texture_id` is set to the `GLId` that you get from [RenderTextureGLId](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/include/gz/rendering/Camera.hh#L318)?

Levi-Armstrong, 2022-11-09:
>I haven't used QOpenGLWidget / QOpenGLTextureBlitter before. A quick look at the doc and examples, it seems fine. Just checking that your this->dataPtr->renderer.texture_id is set to the GLId that you get from [RenderTextureGLId](https://github.com/gazebosim/gz-rendering/blob/gz-rendering7/include/gz/rendering/Camera.hh#L318)?

That is correct. I should have included the full paintGL function which essentially does the following.

```
  makeCurrent();
  this->dataPtr->renderer.camera->Capture();
  this->dataPtr->renderer.texture_id = this->dataPtr->renderer.camera->RenderTextureGLId();
 doneCurrent();

  makeCurrent();
  glClearColor(0.0f, 0.0f, 0.0f, 1.0f);
  glClear(GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT | GL_ACCUM_BUFFER_BIT | GL_STENCIL_BUFFER_BIT);
  glBindTexture(GL_TEXTURE_2D, this->dataPtr->renderer.texture_id);
  this->dataPtr->texture_blitter.bind(GL_TEXTURE_2D);
  const QRect targetRect(QPoint(0, 0), size());
  QMatrix4x4 target = QOpenGLTextureBlitter::targetTransform(targetRect, QRect(QPoint(0, 0), this->dataPtr->renderer.textureSize));
  target.translate(0, 0, -1); // If all zeros the scene is randomly visible.
  this->dataPtr->texture_blitter.blit(this->dataPtr->renderer.texture_id, target, QOpenGLTextureBlitter::OriginTopLeft);
  this->dataPtr->texture_blitter.release();
  doneCurrent();
```

Levi-Armstrong, 2022-11-11:
Well after working on this off and on for six months I finally was able to get it working. All I had to do was take pieces from this demo `simple_demo_qml` around using shared context and using QOffscreenSurface to have it render here then use the  QOpenGLTextureBlitter render it to the widget and everything is now working. I will spend time cleaning  things up and creating a demo to add here.

https://user-images.githubusercontent.com/9803128/201255803-2ddf4bae-9e80-43bf-a134-b6e913d9a347.mp4

Levi-Armstrong, 2022-11-14:
If anyone is interested in a working example take a look here at [tesseract_qt/rendering](https://github.com/tesseract-robotics/tesseract_qt/tree/main/rendering). Next step is to work on adding back the functionality for rendering on another thread. After I get everything working, I will create an example similar to the qml example to add to this repository.