gazebosim/gz-rendering #568: Weird Render Queue usage, transparents and sky

darksylinc, 2022-02-27:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
  Technically ogre-2.3 but it's based off main

## Description

I was having trouble with a GI helper tool not showing up on screen, but it was previous before merging with latest main changes.
Upon research I discovered a few WTF (What a Terrible Failure) situations.

The problem is in `Ogre2RenderTarget::BuildCompositor`. The [node definition is as follows](https://github.com/ignitionrobotics/ign-rendering/blob/4d4e3dbb73974c15920d2a7de2639f6f7e29e270/ogre2/src/Ogre2RenderTarget.cc#L194-L231):

```cpp
// scene pass - opaque
{
  Ogre::CompositorPassSceneDef *passScene =
      static_cast<Ogre::CompositorPassSceneDef *>(
      rt0TargetDef->addPass(Ogre::PASS_SCENE));
  passScene->mShadowNode = this->dataPtr->kShadowNodeName;
  passScene->mIncludeOverlays = false;
  passScene->mFirstRQ = 0u;
  passScene->mLastRQ = 2u;
  if (!validBackground)
  {
    passScene->setAllLoadActions(Ogre::LoadAction::Clear);
    passScene->setAllClearColours(this->ogreBackgroundColor);
  }
}

// render background, e.g. sky, after opaque stuff
if (validBackground)
{
  // quad pass
  Ogre::CompositorPassQuadDef *passQuad =
      static_cast<Ogre::CompositorPassQuadDef *>(
      rt0TargetDef->addPass(Ogre::PASS_QUAD));
  passQuad->mMaterialName = this->dataPtr->kSkyboxMaterialName + "_"
      + this->Name();
  passQuad->mFrustumCorners =
      Ogre::CompositorPassQuadDef::CAMERA_DIRECTION;

  passQuad->setAllLoadActions(Ogre::LoadAction::Clear);
  passQuad->setAllClearColours(this->ogreBackgroundColor);
}

// scene pass - transparent stuff
{
  Ogre::CompositorPassSceneDef *passScene =
      static_cast<Ogre::CompositorPassSceneDef *>(
      rt0TargetDef->addPass(Ogre::PASS_SCENE));
  passScene->mIncludeOverlays = true;
  passScene->mShadowNode = this->dataPtr->kShadowNodeName;
  passScene->mFirstRQ = 2u;
}
```

In plain english, the compositor intends to write opaque objects, then sky, then transparents. Quite normal stuff. However this setup is already wrong, and what happens in reality is even weirder:

1. If the sky is rendered, all opaques rendered so far will be deleted
    - This happens because `passQuad->setAllLoadActions(Ogre::LoadAction::Clear);` is literally saying "clear everything we've drawn so far"
2. This bug is apparently going undetected because it seem all or most objects are not using RenderQueue IDs 0 and 1 (ergo, everything is 'transparent') or if they are, they didn't activate the sky

### Therefore there's two bugs:

 - Ignition considers RQ 0 and 1 as transparents, but it's not using them. Either the RQ range needs to change (i.e. expanded), or the Items need to be explicitly set to these RQs
 - The sky should not clear what's done so far (this is a one liner change)

darksylinc, 2022-02-27:
This patch will fix the incorrect compositor setup:

```diff
diff --git a/ogre2/src/Ogre2RenderTarget.cc b/ogre2/src/Ogre2RenderTarget.cc
index efd0d061..5f72c86a 100644
--- a/ogre2/src/Ogre2RenderTarget.cc
+++ b/ogre2/src/Ogre2RenderTarget.cc
@@ -198,7 +198,13 @@ void Ogre2RenderTarget::BuildCompositor()
         passScene->mIncludeOverlays = false;
         passScene->mFirstRQ = 0u;
         passScene->mLastRQ = 2u;
-        if (!validBackground)
+        if (validBackground)
+        {
+          passScene->setAllLoadActions(Ogre::LoadAction::DontCare);
+          passScene->mLoadActionDepth = Ogre::LoadAction::Clear;
+          passScene->mLoadActionStencil = Ogre::LoadAction::Clear;
+        }
+        else
         {
           passScene->setAllLoadActions(Ogre::LoadAction::Clear);
           passScene->setAllClearColours(this->ogreBackgroundColor);
@@ -216,9 +222,6 @@ void Ogre2RenderTarget::BuildCompositor()
             + this->Name();
         passQuad->mFrustumCorners =
             Ogre::CompositorPassQuadDef::CAMERA_DIRECTION;
-
-        passQuad->setAllLoadActions(Ogre::LoadAction::Clear);
-        passQuad->setAllClearColours(this->ogreBackgroundColor);
       }
 
       // scene pass - transparent stuff
```

iche033, 2022-03-01:
Thanks for the patch! I created the PR for it, https://github.com/ignitionrobotics/ign-rendering/pull/571

closed 2022-03-01 by iche033