gazebosim/gz-rendering #587: libEGL warning failed to create dri screen

quarkytale, 2022-03-17:
## Environment
* OS Version: Ubuntu 20.04
* Source build: Ignition Garden v7.0.0~pre1
* Rendering plugin: ogre2
    - System info:
      - ```
        00:02.0 VGA compatible controller [0300]: Intel Corporation Device [8086:9a60] (rev 01)
        01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:2520] (rev a1)
          ```
       -  ```
          OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 510.54
          OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
          ```
      - ```
        X.Org X Server 1.20.13
        ```
     
## Description
Getting this warning message on a dual GPU machine as well as in Docker.
```
libEGL warning: DRI2: failed to create dri screen
```

[See this related community thread](https://answers.gazebosim.org/question/27597/ignition-crashes-directly-after-start/): forcing ignition to use ogre1 removes that warning message.

Could be ignored but helpful if its removed to avoid confusion.

## Steps to reproduce
Launch the simulator `ign gazebo`

darksylinc, 2022-03-19:
Does this happen on native? The community post you link to referred to the problem happening in VMWare.

Is this just a warning or the app crashes?

Given how recent your GPUs are, it _should_ work on native. It sounds like there is a driver misconfiguration somewhere, specially since it should be using DRI3 not DRI2.

> OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 510.54
> OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

Is this the full contents of running `glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'`?

If so there is a driver problem, because NVIDIA drivers would only be offering OpenGL ES but not OpenGL (which your card definitely supports).

quarkytale, 2022-03-20:
Its just a warning, app doesn't crash.
Full contents of `glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'`:
```
direct rendering: Yes
OpenGL vendor string: NVIDIA Corporation
OpenGL renderer string: NVIDIA GeForce RTX 3060 Laptop GPU/PCIe/SSE2
OpenGL core profile version string: 4.6.0 NVIDIA 510.54
OpenGL core profile shading language version string: 4.60 NVIDIA
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL version string: 4.6.0 NVIDIA 510.54
OpenGL shading language version string: 4.60 NVIDIA
OpenGL context flags: (none)
OpenGL profile mask: (none)
OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 510.54
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

```

darksylinc, 2022-03-26:
I kept thinking about this. I couldn't reply sooner because I was ill.

So everything works just fine, it's just that there is a warning that causes unnecessary concern.

I suspect this could be caused by `GlSwitchableSupport` in Ogre-Next. However I don't have yet a solution.

Switchable support allows Ignition to switch between `X11` & `EGL Headless` rendering (i.e. for VMs without X11).

Ogre-Next needs to initialize both backends, and `EGL Headless` will start enumerating all devices on the system so that Ignition can chose between them (in case Ignition wants to use EGL Headless).

It's quite possible the warning is coming from one of these devices which doesn't end up being used. There can be many devices in a system, because it's not just your NVIDIA + Intel GPUs (2 devices). Sometimes drivers expose more variations (i.e. Compute-only, no graphics) of the same physical HW, or SW emulation devices.

If this theory is correct, it's not easy to fix. Communicating to Ogre-Next from Ignition that it doesn't plan to use Headless needs to happen extremely early and many things haven't been initialized yet.

One way to test this theory could be to [compile Ignition](https://ignitionrobotics.org/docs/garden/install_ubuntu_src) from scratch including [Ogre-Next 2.2 from source](https://github.com/ignition-forks/ogre-2.2-release) but force disabling `OGRE_GLSUPPORT_USE_EGL_HEADLESS`.

iche033, 2022-03-28:
I also see this message native without specifying to run headless. I think it could be harmless as I'm able to continue using ign-rendering / gazebo.

I see these msgs in my ogre2.log that look relevant (these are printed without specifying to use EGL):

<details><summary>ogre2.log exceptions</summary>

```
15:14:05: Found Num EGL Devices: 3
15:14:05: EGL Device: EGL_NV_device_cuda EGL_EXT_device_drm EGL_EXT_device_drm_render_node EGL_EXT_device_persistent_id #0 /dev/dri/card0
15:14:05: Trying to init device: EGL_NV_device_cuda EGL_EXT_device_drm EGL_EXT_device_drm_render_node EGL_EXT_device_persistent_id #0 /dev/dri/card0...
15:14:05: Created GL 4.5 context for device EGL_NV_device_cuda EGL_EXT_device_drm EGL_EXT_device_drm_render_node EGL_EXT_device_persistent_id #0 /dev/dri/card0
15:14:05: Destroying device: EGL_NV_device_cuda EGL_EXT_device_drm EGL_EXT_device_drm_render_node EGL_EXT_device_persistent_id #0 /dev/dri/card0...
15:14:05: EGL Device: EGL_EXT_device_drm #1 /dev/dri/card0
15:14:05: Trying to init device: EGL_EXT_device_drm #1 /dev/dri/card0...
15:14:05: OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device EGL_EXT_device_drm #1 /dev/dri/card0 in EGLSupport::getGLDisplay at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 320)
15:14:05: OGRE EXCEPTION(3:RenderingAPIException): eglInitialize failed for device EGL_EXT_device_drm #1 /dev/dri/card0 in EGLSupport::getGLDisplay at /var/lib/jenkins/workspace/ogre-2.2-debbuilder/repo/RenderSystems/GL3Plus/src/windowing/EGL/PBuffer/OgreEglPBufferSupport.cpp (line 320)
15:14:05: Destroying device: EGL_EXT_device_drm #1 /dev/dri/card0...
15:14:05: EGL Device: EGL_MESA_device_software #2
15:14:05: Trying to init device: EGL_MESA_device_software #2...
15:14:05: Created GL 4.5 context for device EGL_MESA_device_software #2
15:14:05: Destroying device: EGL_MESA_device_software #2...

```
</details>

darksylinc, 2022-03-28:
> I also see this message native without specifying to run headless.

That's what I said: Ogre doesn't know yet if ignition will use headless but it needs to present information to ignition in case it wants to, thus Ogre goes ahead and queries all devices (some of which may fail), regardless of ignition's intentions.

quarkytale, 2022-04-08:
Sorry it took me time to get to this, but I tried your suggestion @darksylinc and you're right.
I installed [orge-2.2-release](https://github.com/ignition-forks/ogre-2.2-release) from source with `-DOGRE_GLSUPPORT_USE_EGL_HEADLESS=OFF`. Built Gazebo Garden on top of it, confirmed ign-rendering package uses the manually installed ogre, from {WS}/build/ignition-rendering7/CMakeCache.txt:
`pkgcfg_lib_OGRE2_OgreMain:FILEPATH=/home/dharini/ogre_ws/install/lib/libOgreMain.so
`. Got rid of the warning!

Also thanks a lot @iche033 for helping out!